roadrunner2/macbook12-spi-driver #1: Race condition for shown state in TouchBar

Dunedan, 2017-05-29:
As already mentioned in https://github.com/Dunedan/mbp-2016-linux/pull/5#issuecomment-300258862 I noticed a race condition in `appletb`, causing it to show the wrong keys in the TouchBar. I just want to ensure this issue is still on the radar.

As it happens during regular use it's quite enjoying, because the TouchBar suddenly shows the wrong state. E.g. with `fnmode = 2` pressing `fn` should get me from function keys displayed to special keys displayed. Once that bug occurs the special keys stay displayed on the TouchBar even if no key is pressed anymore. Only pressing `fn` a few times brings it back in such a case. That correlates with the TouchBar sometimes being unresponsive in such cases.

Aside from happening during regular usage it's also possible to force this issue by frequent key presses (e.g. `fn` and left `ctrl` together). Forcing the issue can be a bit tricky, but I'm able to reliably reproduce it.

My best guess is that it's somehow a race condition between the state the TouchBar currently is in and the state it should be in causing one command to change the state not being submitted to the TouchBar.

roadrunner2, 2017-05-29:
Thanks for opening this here. I had tested frequent fn-key-presses (as fast as I could) originally, and I tried to reproduce this recently, even playing with adding extra delays, but could not. However, a couple days ago I started noticing that every now and then there is an error trying to send the usb control command - can you check your kernel logs (dmesg)? In my case I get an EPIPE (the mode varies between 1 and 2):
```
appletb: Failed to set touchbar mode to 1 (-32)
```
Can you see if you are getting similar (or other) errors when things get stuck?

Dunedan, 2017-05-30:
Yes, as it turns out I actually have dozens of these messages.

roadrunner2, 2017-05-30:
Investigating some more, the EPIPE's are coming from the iBridge device itself (urb status). Furthermore they appear to be related to "switching" between the interfaces: they most consistently show up when sending a control message to interface 2 (used for mode switching) after having sent one to interface 3 (used for dimming and display off).

I'll push some mitigations shortly.

roadrunner2, 2017-05-30:
I've pushed some updates that reduce the chance of this happening, as well as retry when it does, but it's far from perfect - I still don't have a reliable way handle the EPIPE's. But hopefully this make most of your errors go away.

Dunedan, 2017-05-30:
Still getting the EPIPE's, but haven't being able to get the TouchBar in such an odd state again after your changes. :+1:

roadrunner2, 2017-05-31:
Good to hear it hear it mostly resolves the issues you've been seeing. I'm going to leave this open, however, since the underlying issue (-EPIPE) is not fixed.

Dunedan, 2018-11-30:
I haven't seen such -EPIPE errors for a long time. Is that perhaps solved and we can close this issue?

roadrunner2, 2018-12-30:
Yeah, this appears to be fixed. Thanks for verifying and following up. Closing.

closed 2018-12-30 by roadrunner2