roadrunner2/macbook12-spi-driver #2: "spi_pxa2xx_platform"

jthiepler, 2017-06-16:
First, thank you and everyone else a lot for your work on this driver!

I installed the driver on my MBP13,3 runnig Kali (4.9.0). While the Touchbar works perfectly fine, neither the keyboard or the touchpad work. I nailed it down to the missing "spi_pxa2xx_platform" module, but I am lost on how to add it since this part is never mentioned anywhere on the web. I am aware that this is probably a newby question but I sincerely can not solve this on my own and would be very glad if someone could give me the right tip!

roadrunner2, 2017-06-16:
Have you tried Debian's [Module Assistant](https://wiki.debian.org/ModuleAssistant)? Otherwise you probably need to follow these instructions to [recompile the Kali kernel](http://docs.kali.org/development/recompiling-the-kali-linux-kernel) - make sure you enable `SPI_PXA2XX` in the config.

jthiepler, 2017-06-16:
Thank you so much, I recompiled the kernel (my first time) and it works like a charm now!

roadrunner2, 2017-06-17:
Glad to hear you got it working.

closed 2017-06-17 by roadrunner2