roadrunner2/macbook12-spi-driver #3: Touchbar does not awaken after resume

roadrunner2, 2017-07-12:
After suspending and resuming, the touchbar stays off and inactive; however, at the USB level everything looks fine (the responses to the various commands all indicate success). My best guess right now is that some additional command/initialization is needed at suspend and/or resume time.

MacOS uses a much more elaborate protocol to handle the touchbar, whereas this driver is based off of the simpler one used by the Windows driver. So if somebody could run Windows and capture the usb traffic during suspend and resume that would be great.

roadrunner2, 2017-07-23:
@Dunedan I took a look at your traces from cb22/macbook12-spi-driver#30 and played around a bit. Unfortunately those DRLC bulk requests are not relevant for us currently, because the endpoint they are sent to doesn't even exist. I.e. we would have to switch into the full MacOS mode to even be able to send those requests.

Dunedan, 2017-07-23:
Ok, fair enough, as the Touch Bar of course runs in "full mode" under MacOS. Unfortunately when running Windows in a VM suspend is disabled there (probably a driver issue). But let me have another look if I can get that to work somehow.

Dunedan, 2017-07-23:
So. While I was not able to get suspend working in a Windows VM running under Virtualbox (as that doesn't seem to be possible at all), I thought that the driver probably sends the correct commands when the operating system is being shut down as well.
So here are traces from Windows shutting down/restarting: [touchbar-restart-and-shutdown.zip](https://github.com/roadrunner2/macbook12-spi-driver/files/1168350/touchbar-restart-and-shutdown.zip)

roadrunner2, 2017-07-24:
@Dunedan Thanks. That's an interesting idea. I just took a quick look at the traces, and there appear to be a few candidates there - I'll try things out in a few days.

puellanivis, 2017-12-06:
As a random mention here, having switched back to OSX, if my laptop is kept in sleep mode long enough, it will switch into hibernation mode. This requires a reboot to restart the system, where upon… the touchbar is black and entirely unresponsive.

I just thought it might be interesting to note that apparently even OSX cannot get it to work just right all the time either.

Just catching up on things, to see where things are at, because the “Smarch” Bug (“Month 13 is out of bounds”) hit my OSX… and it’s causing me a lot of headaches.

roadrunner2, 2017-12-10:
Interesting tidbit, thanks.

As an update, I did look into the traces, and tried a few things, but couldn't find anything that helped/worked. The full mode that MacOS uses seems to work differently enough from the simple mode that this driver (and the Windows driver) users that there must be some other mechanism when using it.

So, an appeal for help: can somebody who can also boot Windows get a usb trace of the touchbar while suspending and resuming? Apparently something called `Logman` can be used to capture such traces.

ClashTheBunny, 2018-02-11:
I also get the dead touch bar often in MacOS.  It's usually when I run my battery to the percent where it will shut off by itself.  When it unhybernates, I get something similar with a dead touchid, but working touchbar.  So there are obviously levels of breakage...

drrtuy, 2018-04-18:
Could you suggest the way to disable iBridge in MacOS to pass it through to a VirtualBox VM to get USB URB traffic dump?

drrtuy, 2018-04-19:
Since it was late at night, I'd totally forgotten to mention the source of data. 
The dumps was taken using MBP14.3 with MacOS High Sierra. tcpdump collected the data, whilst I closed the laptop's lid.

roadrunner2, 2018-04-20:
@drrtuy Thanks for those traces. But something is odd with them: they appear to contain packets for only a single direction, namely from iBridge to host, and no packets in the reverse direction.

According to the traces, the CLEAR FEATURE is a command sent by the iBridge to the host, so I don't think that's useful here.

Also, the thing is this: the touchbar can run in one of two modes, which I call "simple" and "full" - the former is used by Windows and my `appletb` driver here, and just allows for showing either a fixed list of special keys or the function keys (or turning the touchbar off); the "full" mode is what MacOS uses, where (apparently) arbitrary keys can be defined etc. The traces from the "full" mode aren't really useful in figuring out issues in the "simple" mode, as the protocol is quite different (and indeed there are even additional usb interfaces/endpoints that don't exist in the "simple" mode). Hence what I really need are traces from Windows.

Lastly, a general update here. Playing around with suspend/resume and the xhci driver, I'm starting to think that this might actually be an xhci driver issue: if the driver is not touched during suspend/resume, then upon resume no iBridge devices are visible! The xhci driver has to be unbound from all USB controllers on suspend, and on resume (re)bound to them, for the iBridge devices to appear after resume. One thing I have yet to try is building the xhci driver as a module and removing and re-adding the module across suspend/resume.

drrtuy, 2018-04-20:
I wonder which drivers must I install to get TB work under windows? Should I use bootcamp drivers pack?

roadrunner2, 2018-04-24:
@drrtuy Thank you very much for those traces! From a quick look it looks like there are several new requests in there I haven't seen before. I'll need to play with this some more.

A question, though: in the suspend trace, where did the suspend end and the resume start? Looking at the timestamps, there's a 20 second gap between 11:17:14 and 11:17:34 - was that when the laptop was suspended? So no control messages were sent upon suspend, only upon resume? In the hibernate case I see a gap from 11:20:43 to 11:21:55, so assume that's when it was hibernating?

drrtuy, 2018-04-24:
The answer is yes for both questions.
I saw, that it takes about 30 seconds to get MacOS suspended, so I should
dump for a little longer. But I was a bit excited and waited less then 30
seconds. Message me, if you need additional dumps.

drrtuy, 2018-05-01:
Any updates with the issue?

roadrunner2, 2018-05-08:
Sorry, I haven't been able to spend much time on this recently. Furthermore, for some reason the USB traces are not correct: they show data being sent to the device as coming from the device... Anyway, after further digging into the USB specs I'm fairly confident I have extracted the correct info from the traces. Unfortunately, even with these new control messages I haven't been able to get the touchbar to turn on :disappointed: Still looking at things, though.

drrtuy, 2018-05-08:
> messages I haven't been able to get the touchbar to turn on 😞 Still

I'm glad that you spend your time on the issue anyway.
If the dump didn't get us to the point, then we need more dumps to offer to
the God of dumps :) - I'll take dumps using a couple of different USB
sniffers. Hope they will be more accurate and informative.
I think the sniffer could loose the traffic, since I'm not aware about the
order of taking devices down/up. Is there a concept of promiscuous mode for
the port/device in USB world?

roadrunner2, 2018-09-25:
Thanks to everyone who helped here, in particular @drrtuy - in the end the trigger to wake up the device was an ACPI method, not any USB operation, hence why the traces didn't reveal anything useful.

closed 2018-09-25 by roadrunner2

drrtuy, 2018-09-25:
Wow, great news indeed!

Many thanks for the reasearch and looking forward for the update.

commit 107e6b68 (2018-10-06): appletb: Wake up iBridge device after resume.

The iBridge device ("ASOC" in the ACPI tree) has a power management
method "SOCW" that needs to be invoked to wake the device after resume.
For good measure (and maximum suspend time), we also now explicitly
disable the power on suspend.

This closes #3.
commit 483bbee6 (2018-10-23): appletb: Wake up iBridge device after resume.

The iBridge device ("ASOC" in the ACPI tree) has a power management
method "SOCW" that needs to be invoked to wake the device after resume.
For good measure (and maximum suspend time), we also now explicitly
disable the power on suspend.

This closes #3.
commit 77bd003c (2018-12-09): appletb: Wake up iBridge device after resume.

The iBridge device ("ASOC" in the ACPI tree) has a power management
method "SOCW" that needs to be invoked to wake the device after resume.
For good measure (and maximum suspend time), we also now explicitly
disable the power on suspend.

This closes #3.