roadrunner2/macbook12-spi-driver #4: evemu-record stops showing any input when a palm is released slowly from the touchpad

peterychuang, 2017-08-02:
This problem was first discovered when I was trying to make ```libinput``` capable of rejecting palm touches based on touch-sizes (see [this branch](https://github.com/peterychuang/libinput/tree/wip/touchpad-apple-touch-major-v2) and [this thread](https://github.com/Dunedan/mbp-2016-linux/issues/20#issuecomment-314157722) ).

When I release my palm very slowly from the touchpad, often times the machine fails to respond to the next finger input. ```evemu-record``` reveals that when a palm is released slowly, it sometimes stops registering anything in the middle of the slow release. If ```ABS_MT_TOUCH_MAJOR``` is above the touch-size threshold the instance ```evemu-record``` is stuck, the cursor will fail to move on the next finger input even though ```evemu-record``` will register the said input. The cursor will only start moving again on the next touch.

roadrunner2, 2017-08-04:
Interesting - I can confirm that behaviour exactly (I've been using libinput master too recently). However, I think what evemu-record is showing is correct: for a palm I see events start only when palm size threshold is reached (or more precisely: I see a few events, and then it stops as it recognizes that this is a palm, and doesn't start showing events until the pressure is increased to the point where the palm size threshold is reached), and on releasing they stop as soon as it goes under the threshold again. So I think the pointer-is-stuck is a bug in libinput.

roadrunner2, 2018-12-30:
@peterychuang Do you still see this? I can't reproduce this anymore with libinput 1.12.4.

peterychuang, 2018-12-30:
@roadrunner2 unfortunately I've recently sold my macbook, so I can't test it any more... sorry.

roadrunner2, 2018-12-31:
Ok, thanks. Closing this then since the bug appears to be fixed.

closed 2018-12-31 by roadrunner2