roadrunner2/macbook12-spi-driver #7: Cannot get keyboard to work on MacBook 14,3

sumdog, 2018-02-25:
I'm currently on Linux 4.16.0-rc2, trying to get my onboard keyboard and touchpad working. Currently I can boot, launch to X and connect to the network (using a USB ethernet adapter).

Using the roadrunner2/macbook12-spi-driver fork, if I modprobe the `applespi` and `appletb` modules, I do get a working touch bar. I see the escape and media keys. However, I don't get a working builtin keybord or touchpad. I have to continue to use the external ones. 

If I `rmmod` and `modprobe` the `appletb` module, I see the following in my kernel logs:

```
[[ 3323.821411] input: Apple Inc. iBridge as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.2/0003:05AC:8600.0001/input/input18
[ 3323.873099] apple-touchbar 0003:05AC:8600.0001: input,hidraw0: USB HID v1.01 Keyboard [Apple Inc. iBridge] on usb-0000:00:14.0-3/input2
[ 3323.873104] apple-touchbar 0003:05AC:8600.0001: module probe done.
[ 3323.873365] apple-touchbar 0003:05AC:8600.0002: hiddev96,hidraw1: USB HID v1.01 Device [Apple Inc. iBridge] on usb-0000:00:14.0-3/input3
[ 3323.873369] apple-touchbar 0003:05AC:8600.0002: module probe done.
[ 3323.873398] appletb: Touchbar usb device added; dev=0003:05AC:8600.0001
[ 3323.971878] appletb: Failed to set touchbar display to 1 (-32)](url)
```

I get absolutely nothing for the `applespi` module. I've tried this and the `cb22` version this was forked from.

I don't currently have these modules in my initrd. I'm using [better-initramfs](https://github.com/slashbeast/better-initramfs) for my initramfs (my primary partition is luks encrypted and has LVMs and I've found this initrd works best), but it doesn't seem to support modprobe/module support .. although I can probably hack at it and add them. Could loading `applespi` earlier in the boot process fix this issue? 

Here are my current modules:

```
lsmod
Module                  Size  Used by
appletb                20480  0
applespi               24576  0
amdgpu               2527232  2
brcmfmac              212992  0
brcmutil               16384  1 brcmfmac
mfd_core               16384  1 amdgpu
chash                  16384  1 amdgpu
gpu_sched              20480  1 amdgpu
ttm                    81920  1 amdgpu
```

And here is my `/proc/config.gz`:

[config.gz](https://github.com/roadrunner2/macbook12-spi-driver/files/1755744/config.gz)

sumdog, 2018-02-25:
Nevermind, I figured it out. I was missing `CONFIG_X86_INTEL_LPSS=y` and `CONFIG_MFD_INTEL_LPSS_PCI=y`. 

The README is a little confusing because it states

    CONFIG_X86_INTEL_LPSS=n if running a kernel before 4.14

..and then later tells you to configure it as a module anyway. I think that's what got me confused; thought I didn't need it.

roadrunner2, 2018-02-26:
Sorry if the README is confusing; but it does state, right before the part you quoted above, that that is for the 2015 MB - for all other models the next paragraph applies. I'll think about how to make this clearer.

sumdog, 2018-02-26:
No worries. I'm just glad it works! Thanks for all your development work. It's incredibly helpful!

ClashTheBunny, 2018-05-14:
This can be closed?

roadrunner2, 2018-05-14:
Sorry, yes I think so - closing it.

closed 2018-05-14 by roadrunner2