roadrunner2/macbook12-spi-driver #8: Keyboard shortscuts don't work with Touch Bar keys

Dunedan, 2018-03-04:
Following the [rollover problems with the keyboard](https://github.com/cb22/macbook12-spi-driver/issues/53), there is also an issue with key combinations involving the functions keys on the Touch Bar. I'm facing this issue since the beginning of `appletb`.
Combinations involving `Ctrl + Alt + <a Touch Bar function key>` key simply don't work.

E.g. with `fnmode = 2`:
`Alt + F1` = works
`Alt + Meta + F1` = works
`Ctrl + F1` = works
`Ctrl + Meta + F1` = works
`Ctrl + Alt + F1` = doesn't work
`Ctrl + Alt + Esc` = works
`Ctrl + Alt + Fn + Esc` = works
`Ctrl + Alt + Fn + Mute audio` = works
`Ctrl + Alt + Meta + Esc` = works
`Ctrl + Alt + Meta + F1` = doesn't work

Using `evtest` I looked into the events emitted by the Touch Bar. I noticed that all key presses emit an event of type `EV_KEY`, but the function keys also emit an `MSC_SCAN` event of type `EV_MSC` before. I'm not sure if that's relevant, but that's the only obvious difference I noticed so far.

roadrunner2, 2018-03-05:
I'm having trouble reproducing this: e.g. ctrl+meta+f1 (fnmode 1 or 2) works just fine for me, triggering the associated behaviour, as well as ctrl+alt+f1, meta+f1, alt+f1, etc. Can you elaborate on "doesn't work": does nothing happen, or is something else triggered, or what? Do you see the correct keycodes when you run `libinput-debug-events --show-keycodes`? Also, what DE are you running? I'm running Gnome 3.26.

Regarding the MSC_SCAN: that's fine: it's generated by the usbhid driver, and just means it's also sending the raw scan codes as events. However, double checking this, I noticed that the reported capabilities were way off, including many things that the touchbar doesn't actually support - I've just pushed an update which fixes that, though in practise it shouldn't change anything.

Dunedan, 2018-03-09:
Under the assumption that you're running Xorg with something like `DontVTSwitch` (something I don't) I believe I figured out what the problem is:

The whole problem is related to VT switching under Xorg. Turns out, to switch to another VT Xorg requires the Ctrl + Alt + Fx key combination to come from a single input device. As the internal keyboard and the Touch Bar are two separate input devices, it doesn't work.

I was able to reproduce this problem on a completely different computer with two external USB keyboards, which show the same behavior in that case. That also explains why it works for me when using an external USB keyboard with the MBP or when not running Xorg at all.

roadrunner2, 2018-04-02:
Sorry for the late reply. This is interesting: assuming you're using libinput under X too, I wonder if the recent [multiple-paired-keyboards](https://cgit.freedesktop.org/wayland/libinput/commit/?id=4d7592066a120b00386283ebcbb7503296720626) support can be used here (available in libinput 1.10). Looking at the code, the trick seems to be to mark the touchbar as an internal keyboard; I think something like the following in `/etc/udev/hwdb.d/61-apple-touchbar.hwdb` (or similar) should do the trick:
```
libinput:name:*Apple Inc. iBridge*:dmi:*
 LIBINPUT_ATTR_KEYBOARD_INTEGRATION=internal

```

roadrunner2, 2018-04-02:
Ok, just played with this a bit, and the above doesn't seem to work/help. The only other thing I can think of right now, short of patching Xorg, would be to merge `applespi` and `appletb` into one, but that would be fairly ugly and I'd rather not go there.

Dunedan, 2018-04-07:
Seems like somebody else already documented that problem earlier (with no solution though): https://bbs.archlinux.org/viewtopic.php?id=230399

Neither patching Xorg, nor merging the drivers sounds like the right solution for me. I did a quick test with Wayland instead of Xorg and it appears to work fine with Wayland, which leads me to believe it's actually no bug in libinput, but in Xorg instead.

roadrunner2, 2018-12-30:
Closing this as "won't fix", because I don't think there's anything we can (reasonably) do about this from the driver side. Feel free to reopen if somebody has a good idea.

closed 2018-12-30 by roadrunner2