roadrunner2/macbook12-spi-driver #9: Runaway kworker (maybe related to these drivers)?

whereswaldon, 2018-07-19:
Firstly, thanks so much for putting these resources together! It's awesome to be able to run Linux on these machines at all.

I followed the guidance in [this gist](https://gist.github.com/roadrunner2/1289542a748d9a104e7baec6a92f9cd7) to set up everything in Fedora 28 on a MacBookPro14,3, and it is largely working beautifully. My touchbar works, as does the trackpad and keyboard.

However, my machine runs very hot. When I looked into why, I discovered that one CPU is always running at about 80%, and `top` reports that the culprit is `kworker/6:3`. It seemed reasonable to me to ask whether this might be a bug in one of the `applespi` or `appletb` drivers, as I don't think that this happened before I installed them.

Some possibly helpful diagnostic information:
```
$ dmesg | grep apple
[    0.285641] apple-properties: device path parse error -19 at 0x14:
[    0.285645] apple-properties: 00000000: a6 01 00 00 04 00 00 00 02 01 0c 00 d0 41 03 0a  .............A..
[    0.285647] apple-properties: 00000010: 00 00 00 00 01 01 06 00 00 02 7f ff 04 00 1e 00  ................
[    0.285650] apple-properties: 00000020: 00 00 73 00 61 00 76 00 65 00 64 00 2d 00 63 00  ..s.a.v.e.d.-.c.
[    0.285652] apple-properties: 00000030: 6f 00 6e 00 66 00 69 00 67 00 00 00 e2 00 00 00  o.n.f.i.g.......
[    0.285654] apple-properties: 00000040: 47 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  G...............
[    0.285656] apple-properties: 00000050: 00 00 03 00 06 00 1b 59 00 6c 05 00 00 00 00 00  .......Y.l......
[    0.285658] apple-properties: 00000060: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    0.285660] apple-properties: 00000070: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    0.285663] apple-properties: 00000080: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    0.285665] apple-properties: 00000090: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    0.285667] apple-properties: 000000a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    0.285669] apple-properties: 000000b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    0.285672] apple-properties: 000000c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    0.285674] apple-properties: 000000d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    0.285676] apple-properties: 000000e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    0.285678] apple-properties: 000000f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    0.285681] apple-properties: 00000100: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    0.285683] apple-properties: 00000110: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 2c 00  ..............,.
[    0.285685] apple-properties: 00000120: 00 00 41 00 41 00 50 00 4c 00 2c 00 69 00 67 00  ..A.A.P.L.,.i.g.
[    0.285687] apple-properties: 00000130: 2d 00 70 00 6c 00 61 00 74 00 66 00 6f 00 72 00  -.p.l.a.t.f.o.r.
[    0.285690] apple-properties: 00000140: 6d 00 2d 00 69 00 64 00 00 00 08 00 00 00 06 00  m.-.i.d.........
[    0.285692] apple-properties: 00000150: 1b 59 20 00 00 00 41 00 41 00 50 00 4c 00 2c 00  .Y ...A.A.P.L.,.
[    0.285694] apple-properties: 00000160: 47 00 66 00 78 00 59 00 54 00 69 00 6c 00 65 00  G.f.x.Y.T.i.l.e.
[    0.285696] apple-properties: 00000170: 00 00 08 00 00 00 01 00 00 00 24 00 00 00 67 00  ..........$...g.
[    0.285699] apple-properties: 00000180: 72 00 61 00 70 00 68 00 69 00 63 00 2d 00 6f 00  r.a.p.h.i.c.-.o.
[    0.285701] apple-properties: 00000190: 70 00 74 00 69 00 6f 00 6e 00 73 00 00 00 08 00  p.t.i.o.n.s.....
[    0.285703] apple-properties: 000001a0: 00 00 0c 00 00 00                                ......
[    1.646903] applespi: loading out-of-tree module taints kernel.
[    1.646925] applespi: module verification failed: signature and/or required key missing - tainting kernel
[    1.767629] applespi: info 40 d0 00 00 00 00 78 00 20 10 02 00 00 02 6e 00 05 01 09 02 a1 01 09 01 a1 00 05 09 19 01 29 03
[    1.767632] applespi: info 15 00 25 01 85 02 95 03 75 01 81 02 95 01 75 05 81 01 05 01 09 30 09 31 15 81 25 7f 75 08 95 02
[    1.767634] applespi: info 81 06 95 04 75 08 81 01 c0 c0 05 0d 09 05 a1 01 06 00 ff 09 0c 15 00 26 ff 00 75 08 95 10 85 3f
[    1.767635] applespi: info 81 22 c0 06 00 ff 09 0c a1 01 06 00 ff 09 0c 15 00 26 ff 00 85 44 75 08 96 d7 06 81 00 c0 c0 fb
[    1.767637] applespi: info 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    1.767638] applespi: info 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    1.767640] applespi: info 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    1.767641] applespi: info 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 89 dc
[    1.769165] applespi: spi-device probe done: spi-APP000D:00
[    1.770894] applespi: modeswitch done.
[    1.885284] Modules linked in: amdkfd amd_iommu_v2 amdgpu(+) spi_pxa2xx_platform chash i2c_algo_bit gpu_sched drm_kms_helper ttm crct10dif_pclmul drm crc32_pclmul crc32c_intel nvme ghash_clmulni_intel nvme_core intel_lpss_pci intel_lpss applespi(OE) ax88179_178a(+) usbnet mii
[    2.597563] apple-touchbar 0003:05AC:8600.0004: input,hidraw0: USB HID v1.01 Keyboard [Apple Inc. iBridge] on usb-0000:00:14.0-3/input2
[    2.597570] appletb: Connected to keyboard input device
[    2.597573] appletb: Connected to touchpad input device
[    2.597585] apple-touchbar 0003:05AC:8600.0004: module probe done.
[    2.598866] apple-touchbar 0003:05AC:8600.0005: hiddev96,hidraw1: USB HID v1.01 Device [Apple Inc. iBridge] on usb-0000:00:14.0-3/input3
[    2.598872] appletb: Connected to keyboard input device
[    2.598875] appletb: Connected to touchpad input device
[    2.598884] apple-touchbar 0003:05AC:8600.0005: module probe done.
[    2.598939] appletb: Touchbar usb device added; dev=0003:05AC:8600.0004
[    2.700439] appletb: Failed to set touchbar display to 1 (-32)
[   11.755950] apple_gmux: Found gmux version 4.0.29 [indexed]
[   12.122269] applesmc: key=862 fan=2 temp=46 index=45 acc=0 lux=0 kbd=0
[   12.122348] applesmc applesmc.768: hwmon_device_register() is deprecated. Please convert the driver to use hwmon_device_register_with_info().
[  299.277441] appletb: Failed to set touchbar display to 1 (-32)
[  594.190798] appletb: Failed to set touchbar display to 1 (-32)
[  811.278534] appletb: Failed to set touchbar display to 1 (-32)
[ 1069.328527] appletb: Failed to set touchbar display to 1 (-32)
[ 1294.606986] appletb: Failed to set touchbar display to 1 (-32)
[ 1536.267561] appletb: Failed to set touchbar display to 1 (-32)
[ 1769.747588] appletb: Failed to set touchbar display to 1 (-32)
[... 4 lines cut ...]
```
The CPU is getting hot enough to cause throttling:
```
[ 3375.818766] CPU7: Core temperature above threshold, cpu clock throttled (total events = 911)
[ 3375.818767] CPU3: Core temperature above threshold, cpu clock throttled (total events = 911)
[ 3375.818768] CPU5: Package temperature above threshold, cpu clock throttled (total events = 1620)
[ 3375.818770] CPU3: Package temperature above threshold, cpu clock throttled (total events = 1620)
[ 3375.818774] CPU7: Package temperature above threshold, cpu clock throttled (total events = 1620)
[ 3375.818803] CPU4: Package temperature above threshold, cpu clock throttled (total events = 1620)
[ 3375.818804] CPU0: Package temperature above threshold, cpu clock throttled (total events = 1620)
[ 3375.818806] CPU6: Package temperature above threshold, cpu clock throttled (total events = 1620)
[ 3375.818807] CPU2: Package temperature above threshold, cpu clock throttled (total events = 1620)
[ 3375.818807] CPU1: Package temperature above threshold, cpu clock throttled (total events = 1620)
[ 3375.871785] CPU7: Core temperature/speed normal
[ 3375.871785] CPU3: Core temperature/speed normal
[ 3375.871787] CPU6: Package temperature/speed normal
[ 3375.871787] CPU2: Package temperature/speed normal
[ 3375.871788] CPU5: Package temperature/speed normal
[ 3375.871789] CPU1: Package temperature/speed normal
[ 3375.871790] CPU3: Package temperature/speed normal
[ 3375.871790] CPU7: Package temperature/speed normal
[ 3375.871818] CPU4: Package temperature/speed normal
[ 3375.871819] CPU0: Package temperature/speed normal
```
I unloaded and reloaded the `applespi` module and the runaway `kworker` went away (for the time being). I don't have enough empirical data to say that it actually has anything to do with the drivers in this repo, but I thought that I should report it in case others encounter similar behavior.

roadrunner2, 2018-07-20:
The `applespi` (for kernels 4.14 or newer) does not use workqueues; `appletb`, however, does. In `applespi` basically everything is run from a softirq handler, i.e. in `ksoftirqd` threads. So I'm a bit surprised that reloading `applespi` solved the issue. I'm assuming this happens consistently?

Can you show me the output from `cat /proc/<pid-of-runaway-kworker>/stack` (might want to run this a few times)? Also, this [stackexchange answer](https://unix.stackexchange.com/questions/22851/why-is-kworker-consuming-so-many-resources-on-linux-3-0-0-12-server/65270#65270) has some info on how to get more details about the kworker.

In short, I have no idea what's going on on your system, so need some more info about the errant kworker thread.

whereswaldon, 2018-07-24:
Thanks for your response! It isn't happening anymore, so I'm pretty sure it was coincidence. I really appreciate your work on these drivers. Since it isn't reproducible, I'm going to go ahead and close this.

closed 2018-07-24 by whereswaldon

whereswaldon, 2018-08-07:
In case someone else has this problem, I found an apparent fix thanks to this issue:
https://github.com/Dunedan/mbp-2016-linux/issues/70