roadrunner2/macbook12-spi-driver #10: Touchbar stay on all the time.

coolguy1004, 2018-07-22:
Hi,
I noticed the touchbar stays on all the time now. I'm running kernel: 4.17.6-200.fc28.x86_64
Below is the dmesg |grep appletb
[root@snguyenfed power]# dmesg |grep appletb
[    1.536839] appletb: Touchbar usb device added; dev=0003:05AC:8600.0001
[    1.537194] appletb: Connected to keyboard input device
[    1.537195] appletb: Connected to keyboard input device
[    1.537271] appletb: Connected to touchpad input device
[    1.537272] appletb: Connected to touchpad input device
[    1.636050] appletb: Failed to set touchbar display to 1 (-32)
[    1.658260] Modules linked in: amdkfd amd_iommu_v2 amdgpu(+) spi_pxa2xx_platform chash i2c_algo_bit gpu_sched drm_kms_helper ttm crct10dif_pclmul crc32_pclmul crc32c_intel nvme drm intel_lpss_pci nvme_core intel_lpss ghash_clmulni_intel appletb(OE) applespi(OE)
[  300.092614] appletb: Failed to set touchbar display to 1 (-32)
[  594.975370] appletb: Failed to set touchbar display to 1 (-32)
[  840.753906] appletb: Failed to set touchbar display to 1 (-32)

roadrunner2, 2018-07-27:
I can't reproduce this. The touchbar dimming is definitely broken since kernel 4.16 (those are those `Failed to set touchbar display` errors); but actually turning the touchbar off works fine for me, and I don't see any related errors in the dmesg output above either.

Sorry if this is stupid, but are you sure you waited long enough (with no keyboard or touchpad activity)? Have you tried setting the idle-timeout to a lower value - e.g. this will set it 10 seconds:
```
echo 10 | sudo tee /sys/class/hidraw/hidraw0/device/idle_timeout
```

roadrunner2, 2018-09-11:
I just pushed a fix for what was causing those "Failed to set touchbar display" errors - it might well fix this issue too.

arno01, 2018-09-17:
@roadrunner2 for me, the touchbar dimming is working just fine on ``4.18.8-041808-generic`` and probably for older kernels too. It is that I have just tested it by lowering down the idle timeout with the command you pasted above.

But I am always having the ``appletb: Failed to set touchbar display to 1 (-32)`` message, each time I press or unpress the Fn key. Any clue why? Or how could I help you to debug that?

roadrunner2, 2018-09-18:
@arno01 Thanks for testing.

Regarding the error, please pull my latest code - I fixed this issue a few days ago, though by all means let me know if you still see it.

arno01, 2018-09-18:
@roadrunner2 np, thanks for writing the code! ;-)
I've tried it out and the annoying message has gone!

Thank you!

roadrunner2, 2018-09-19:
@coolguy1004 Can you still reproduce this even with the latest code?

roadrunner2, 2018-12-30:
No response from submitter, I can't reproduce this, and several potential issues have been fixed in the mean time. So closing this. Feel free to reopen if it's still an issue.

closed 2018-12-30 by roadrunner2