roadrunner2/macbook12-spi-driver #11: Swap option/command keys

nedclimaterisk, 2018-09-20:
In the hid_apple driver, there is the [option to swap the cmd/option keys](https://wiki.archlinux.org/index.php/Apple_Keyboard#Switching_Cmd_and_Alt.2FAltGr), which is nice, because they're in an annoying place on smaller macbooks. There is also a [patched version that allows swapping of the ctrl/fn keys](https://github.com/free5lot/hid-apple-patched), and it would be great if those options could also be included.

Dunedan, 2018-09-20:
For the latter request there is already https://github.com/cb22/macbook12-spi-driver/issues/59

roadrunner2, 2018-09-20:
Right; and as I mention in cb22#59, I'm somewhat disinclined to add arbitrary key-mapping capabilities at the driver level. @nedclimaterisk, can you explain why your desktop/console/whatever's key mapping functions aren't suitable/preferrable for this? E.g. Gnome has something like 150 key mapping options to choose from. In fact, I also switch those two keys (among other mappings) with the "Alt/Win key behaviour" -> "Left Alt is swapped with Left Win" (Cmd is called Win in Gnome since that's what the key is called on PC keyboards).

nedclimaterisk, 2018-09-21:
Ah, I didn't notice that, KDE also has that option. Yes, looks like this is a pure duplicate. Sorry for the noise :)

closed 2018-09-21 by nedclimaterisk

nedclimaterisk, 2018-09-21:
Hrm.. when swapping the left option/command keys (alt, win?super?), I get the correct behaviour for the alt key as a modifier on the command key, but if I press the command key by itself, it brings up the k-menu. If they keys were truly swapped, then that behaviour should now be on the option key. I don't know if this is a driver problem, or a problem with KDE's key switching code.