roadrunner2/macbook12-spi-driver #12: function key swapped with left shift

nedclimaterisk, 2018-10-05:
I'm not sure how this happened, but my function key has become swapped with my left shift key. This was quite annoying when I booted and didn't realise, and my decryption password has a capital letter in it, so I thought I was locked out.

I have messed with a few files in /etc/modprobe.d/ when trying to swap function keys, but I have everything in apple related files commented out right now, and it's still happening:

```
$ cat /etc/modprobe.d/*apple*
# options applespi fnremap=2
# options applespi swap_fn_leftctrl=1
# options applespi swap_opt_cmd=1
# options appletb fnmode=2
# options hid_apple fnmode=2
```

I'm not even entirely sure that I'm using this driver - I'm using a macbook pro 14,1 (without the touchbar).

I think this change happened when I upgraded my kernel to the amd64 version from http://kernel.ubuntu.com/~kernel-ppa/mainline/v4.18.12/

nedclimaterisk, 2018-10-05:
ok, it's not kernel related, i just rebooted back to 4.15.0-36, and it's still happening.

roadrunner2, 2018-10-05:
> I'm not sure how this happened, but my function key has become swapped with my left shift key. This was quite annoying when I booted and didn't realise, and my decryption password has a capital [...]

This would be the one: this says to swap fn with left-shift (see also the output from `modinfo applespi`).

> ```
> # options applespi swap_fn_leftctrl=1
> # options applespi swap_opt_cmd=1

These don't exist.

> ```
> # options appletb fnmode=2

Since you don't have a touchbar, this isn't relevant either.

> ```
> # options hid_apple fnmode=2
> ```
> 
> I'm not even entirely sure that I'm using this driver - I'm using a macbook pro 14,1 (without the touchbar).

You're using the `applespi` driver for the keyboard and touchpad, but not the `appletb` driver.

Your modprobe files seem to have all options commented out, so it shouldn't be swapping the keys anymore. However, if you modified those modprobe files without rebuilding the initrd then that could explain why you're still seeing them swapped.

You can check the current value of `fnremap` with
```
cat /sys/module/applespi/parameters/fnremap
```
Assuming that gives a value of 2, check the modprobe files in your initrd (or just rebuild it). You should also be able to verify the option is not being picked up from somewhere else by just reloading the module (`sudo rmmod applespi ; sudo modprobe applespi`) and verifying `fnremap` is now 0.

roadrunner2, 2018-10-05:
Oh, and also double check you aren't setting the option on the kernel comand line (`cat /proc/cmdline`).

nedclimaterisk, 2018-10-07:
Sweet, I think it was the update-initramfs that fixed it. I updated the modeprobe.d files, and now have fnmode=2 and fnremap=1 working on boot, as I originally wanted. Thanks for your very helpful reply!

commit a7a73f0d (2017-05-14): add option iso_layout

fixes #12
closed 2018-10-07 by nedclimaterisk