roadrunner2/macbook12-spi-driver #13: Won't compile on kernel 4.15.0

double16, 2018-10-08:
The hid_report struct has changed so this won't compile on 4.15.0.

DKMS make.log for applespi-0.1 for kernel 4.15.0-36-generic (x86_64)
Mon Oct  8 07:19:46 CDT 2018
make -C /lib/modules/4.15.0-36-generic/build M=/var/lib/dkms/applespi/0.1/build modules
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-36-generic'
  CC [M]  /var/lib/dkms/applespi/0.1/build/applespi.o
  CC [M]  /var/lib/dkms/applespi/0.1/build/appletb.o
/var/lib/dkms/applespi/0.1/build/appletb.c: In function ‘appletb_find_hid_field’:
/var/lib/dkms/applespi/0.1/build/appletb.c:980:14: error: ‘struct hid_report’ has no member named ‘application’
    if (report->application != application)
              ^~
scripts/Makefile.build:339: recipe for target '/var/lib/dkms/applespi/0.1/build/appletb.o' failed
make[2]: *** [/var/lib/dkms/applespi/0.1/build/appletb.o] Error 1
make[2]: *** Waiting for unfinished jobs....
Makefile:1551: recipe for target '_module_/var/lib/dkms/applespi/0.1/build' failed
make[1]: *** [_module_/var/lib/dkms/applespi/0.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-36-generic'
Makefile:12: recipe for target 'all' failed
make: *** [all] Error 2

robotdna, 2018-10-08:
+1 for a mid 2017 Macbook Pro with touchbar running kernel 4.15.0. Cannot compile.

roadrunner2, 2018-10-09:
Apologies for screwup. Fixed now.

commit 476d2730 (2018-10-23): appletb: Fix compilation kernels prior to 4.18.

The hid_report->application field wasn't added till 4.18-rc1. For older
kernels we can just ignore the application usage check, as it's really
just an additional constraint that isn't strictly necessary (in our
cases).

This fixes #13.
commit bc283fb9 (2018-12-09): appletb: Fix compilation kernels prior to 4.18.

The hid_report->application field wasn't added till 4.18-rc1. For older
kernels we can just ignore the application usage check, as it's really
just an additional constraint that isn't strictly necessary (in our
cases).

This fixes #13.
closed 2018-10-09 by roadrunner2