roadrunner2/macbook12-spi-driver #14: Right click does not work on MBP14,2 (branch touchbar-driver-hid-driver)

vyruss, 2018-11-21:
Hi, I can't right click (two finger click) on the latest touchbar-driver-hid-driver on a MBP14,2.
Interestingly, two-finger scroll works.

roadrunner2, 2018-11-27:
Sorry for the slow reply. I don't think this is a kernel driver issue, as the driver just reports raw touches - it's up to user-space libraries (such as libinput or the Xorg input drivers) to interpret those touches and turn them into various finger clicks, scroll events, etc.

I don't know what your setup is, but assuming you're using libinput try running something like
```
libinput debug-events --enable-tap
```
and try tapping with 1, 2, or 3 fingers and you should see `BTN_LEFT`, `BTN_RIGHT`, etc events (though interestingly, clicking instead of tapping with two fingers still shows `BTN_LEFT` but does work correctly (e.g. popping up the context menu)).

vyruss, 2018-11-27:
No worries, thanks for your reply. It does indeed behave as you say, two-finger _click_ reports `BTN_LEFT` but two-finger _tap_ says 'BTN_RIGHT'. I guess that's why my Ubuntu18.04/Unity/Xorg combo is treating it as a left-click. For some reason in the above environment, my Touchpad options are restricted and missing things such as "tap to click".

vyruss, 2018-11-27:
Solved - you were right @roadrunner2, it was an Xorg input driver issue, as soon as I installed the following, right click started working again and the "missing" Synaptics touchpad options reappeared:
```bash
sudo apt install xserver-xorg-input-synaptics
```
Closing the issue but leaving the solution here in case someone comes across in the future. Thanks for your help.

closed 2018-11-27 by vyruss