roadrunner2/macbook12-spi-driver #16: hid_field_extract() called with n (64) > 32!

Dunedan, 2018-12-08:
Since the split of the drivers I regularly see messages like the following one in the logs:
```
[ 2391.090155] apple-ibridge-hid 0003:05AC:8600.0002: hid_field_extract() called with n (64) > 32! (swapper/0)
```

I suppose that's not intentional?

roadrunner2, 2018-12-09:
Well, no, it's not "intentional", but a limitation in the hid report parsing code: the author seemed to think there would never be reports with fields wider than 32 bits (and the relevant functions all return a `u32`), but a couple of the T1's reports do in fact have 64-bit fields; and one of them is an input report that the device sends every now and then, triggering the above warning.

The meaning of that report is currently unknown to me, so it's just ignored, and therefore the limitation and warning are harmless at this point in time.ut

Btw., are you sure it didn't show up before the split? I thought it was related to the ALS stuff, but could certainly be wrong.

Dunedan, 2018-12-09:
Not entirely sure, but I didn't notice such messages before.

Dunedan, 2018-12-11:
Those commands work for me to be able to unforcefully remove the ALS module. :+1:

Back to the original topic: If we can ignore these warnings I guess we can close this issue as well? Or is there a point in trying to get that changed upstream?

roadrunner2, 2018-12-30:
> Back to the original topic: If we can ignore these warnings I guess we can close this issue as well? Or is there a point in trying to get that changed upstream?

I took a look again, and from my reading of [the spec](https://www.usb.org/sites/default/files/documents/hid1_11.pdf) it looks like 64-bit fields aren't actually allowed, so I've added some patching of the reports to replace these with 32-bit fields, which avoids the warnings in the logs.

commit 64c457a5 (2018-12-30): appletb: Turn 64-bit hid report fields into 32-bit ones.

The spec says fields can't be more than 4 bytes, and hid_extract_field()
complains when encountering such fields. Furthermore, because one of
them is in an input report that is received every 40 minutes, we get a
number of ugly error messages in the kernel log. Turning the fields into
32-bit ones avoids this.

This fixes #16.
closed 2018-12-30 by roadrunner2