roadrunner2/macbook12-spi-driver #21: [Question] Advanced touchbar support?

piranna, 2019-03-24:
I'm curious about what's the current status and what would need to be done to have advanced functionality of the Touchbar so it can be used for example as an (extra) gnome-panel or similar. I'm thinking about what support would need to be added to have something like https://vas3k.com/blog/touchbar/ available on Linux.

Disclaimer: not a current user of a Touchbar Mac, but was offered at work some time ago and got interested on using it with Linux.

roadrunner2, 2019-03-27:
There is currently no support whatsoever for the advanced functionality, and AFAIK nobody has looked into it either. The work basically involves reverse engineering the protocol (usb hid reports) used to provide that functionality, and then defining and implementing API for userspace to interact with it. While not in the least opposed to somebody doing it, so far I personally haven't seen much point in spending any effort on this, because the chances any Linux apps would add support for the touchbar are pretty much nil. However, your link above does point out an interesting use-case, namely a (semi) fixed custom touchbar layout (as opposed to a per-app one), with an app to set up this custom layout - this seems like something useful on Linux.

In short, if somebody wants to reverse engineer the protocol and provide some proof-of-concept code to configure the touchbar, I'd be happy to help take it from there and expand the touchbar driver to support it.

piranna, 2019-03-27:
Yeah, I supposed lack of support was due to being so much specific device, just wanted to know if there was some improvement in this area. Definitely I think that link would be just the only valid use case for the touchbar, just only if it would not remove the function keys that would be great and maybe others manufactures would mimic it...

joax, 2020-06-15:
Hello @roadrunner2 , I would love to connect and help with this. I have checked the code and could work together with you to provide support for the two keys pending (Dashboard and Scale). Let me know.