roadrunner2/macbook12-spi-driver #22: Manjaro Deepin MacBook 12, 2017

marbetschar, 2019-04-01:
Trying to build a custom iso using Manjaro Deepin and its offical `manjaro-tools`. From there I'm [installing your AUR package](https://aur.archlinux.org/packages/macbook12-spi-driver-dkms/). But it seems the driver does not load at boot. At least keyboard and trackpad are unresponsive. Any chance you can point me into the right direction?

For details see:
https://forum.manjaro.org/t/macbook-12-2017-how-to-create-iso-with-aur-driver/80824/5

And here's what I did, step by step:
https://github.com/marbetschar/Manjaro-Linux-on-MacBook-12

marbetschar, 2019-04-01:
@roadrunner2 maybe I was able to track this down to one single question: Does the driver from the AUR package work with Kernel v4.14?

roadrunner2, 2019-04-02:
@marbetschar Yes, the `applespi` keyboard/touchpad driver should work fine under v4.14. Have you checked that the `spi_pxa2xx_platform` and `intel_lpss_pci` modules are built and loaded? And what is the output of `dmesg | grep applespi`?

marbetschar, 2019-04-02:
@roadrunner2 thanks for your response!!

The error on my end was caused by a corrupted configuration file which was responsible for building the *.iso; therefore your driver was never loaded.

That said, your driver work as expected - thank you so much for this awesome piece of work!!

Only things I‘m tinkering with now is touchpad sensitivity - and audio. Two known things, from which audio probably is not solvable right now.

closed 2019-04-02 by marbetschar

marbetschar, 2019-04-02:
@roadrunner2 thanks to your driver I was able to install Linux on my MacBook 12, 2017. Now I have two last questions:

1. `/etc/modules-load.d/macbook12.conf` necessary?
I'm wondering if it is necessary to add a `macbook12.conf` file to `/etc/modules-load.d` like described [here](https://github.com/marbetschar/Manjaro-Linux-on-MacBook-12#make-sure-the-macbook12-drivers-are-loaded-at-boot), if I'm installing the https://aur.archlinux.org/macbook12-spi-driver-dkms package?

2. Is it safe to update the kernel?
When the AUR package is installed, is it safe to update the Linux Kernel in Manjaro (based on Arch)...? Or do I need to rebuild and reinstall the package each time I'm updating the Kernel?

roadrunner2, 2019-04-02:
@marbetschar glad to hear things are working now.

> Only things I‘m tinkering with now is touchpad sensitivity - and audio. Two known things, from which audio probably is not solvable right now.

For me the main issue with the touchpad is that palm rejection in libinput is awful for this touchpad. So far I've not been able to get a good algorithm for it, however. The general sensitivity seems to be fine, though I do use something like the following in my `/etc/libinput/local-overrides.quirks`:
```ini
[MacBook(Pro) SPI Touchpads]
MatchName=*Apple SPI Touchpad*
ModelAppleTouchpad=1
AttrKeyboardIntegration=internal
AttrTouchSizeRange=200:150
AttrPalmSizeThreshold=1100
```

Regarding audio, I'm not sure anybody has started looking at it for MB12 (it's possibly similar to earlier MB's, which is tracked at https://bugzilla.kernel.org/show_bug.cgi?id=110561 , but I'm fairly sure it's noticeably different from recent MBP's).

> @roadrunner2 thanks to your driver I was able to install Linux on my MacBook 12, 2017. Now I have two last questions:
>
>     1. `/etc/modules-load.d/macbook12.conf` necessary?
>        I'm wondering if [...]

Probably not - the applespi driver should get loaded automatically these days, and I presume the other two too, but you'll have to test on your platform. I'd add that I still need to explicitly add the modules to the initrd (in my case via `/etc/dracut.conf.d/...`), but I have no idea how that is done on your distro.

>     2. Is it safe to update the kernel?
>        When the AUR package is installed, is it safe to update the Linux Kernel in Manjaro (based on Arch)...? Or do I need to rebuild and reinstall the package each time I'm updating the Kernel?

Sorry, I've never used Manjaro or Arch, so I really have no idea here either. When using `dkms` the module is automatically rebuilt when a newer kernel is installed, but I don't know what AUR does.

marbetschar, 2019-04-03:
@roadrunner2 awesome!! Thank you so much for those insights and your time, highly appreciate!

marbetschar, 2019-04-03:
For the sake of completeness: 

1. `/etc/modules-load.d/macbook12.conf` is not necessary
2. Simply updating the Kernel is not safe. This breaks Keyboard and Trackpad support (tested in Manjaro KDE)

roadrunner2, 2019-04-04:
In that case it might be better to build an AUR dkms package instead of one providing pre-compiled modules.

marbetschar, 2019-04-04:
What do you mean by "build an AUR dkms package instead" ...?

I'm basically shipping a built AUR dkms package, as described [here](https://github.com/marbetschar/Manjaro-Linux-on-MacBook-12#build-the-macbook12-spi-driver-dkms-package) - unfortunately this is needed, otherwise I'm not able to use my laptop at all ;)

But I need to figure out how to update/replace it once the machine is installed, that's true.

roadrunner2, 2019-04-05:
Apologies, I didn't look at the package closely enough, but since you said updating the kernel wasn't working I just assumed you were building a package with pre-built modules. So, since it does look like the package being built is a dkms package, then it should automatically rebuild the modules whenever a new kernel is installed (even on Arch, according to https://wiki.archlinux.org/index.php/Dynamic_Kernel_Module_Support). I guess you want to see if the modules get rebuilt when a new kernel is installed (`dkms status`) and wether they get installed correctly and the initrd is (re)built with them (at least with the `applespi` driver).