roadrunner2/macbook12-spi-driver #23: intel-lpss-pci kernel bug

meeuw, 2019-04-11:
Hi, I hope you can help me; the macbook12-spi-driver seems to be working fine but when I load intel-lpss-pci modprobe get stuck (even without the macbook12-spi-drivers loaded).

I have blacklisted the module because it really slows down booting so I can load it manually.

Do you know what I can do? I'm using refind without spoof_osx_version. I'm using a macbook pro 13,3 (2016).

I'll attach the output from a debug kernel of dmesg.

meeuw, 2019-04-11:
```[   54.293351] intel-lpss 0000:00:15.0: enabling device (0000 -> 0002)
[   54.304968] idma64 idma64.0: Found Intel integrated DMA 64-bit
[   54.326390] intel-lpss 0000:00:19.0: enabling device (0000 -> 0002)
[   54.330549] dw-apb-uart.1: ttyS4 at MMIO 0x8272a000 (irq = 21, base_baud = 115200) is a 16550A
[   54.331303] serial serial0: tty port ttyS4 registered
[   54.344357] intel-lpss 0000:00:1e.0: enabling device (0000 -> 0002)
[   54.346233] idma64 idma64.2: Found Intel integrated DMA 64-bit
[   54.347084] dw-apb-uart.2: ttyS5 at MMIO 0x8272b000 (irq = 20, base_baud = 3000000) is a 16550A
[   54.347690] serial serial1: tty port ttyS5 registered
[   54.348205] hci_uart_bcm serial1-0: Unexpected ACPI gpio_int_idx: -1
[   54.348298] hci_uart_bcm serial1-0: Unexpected number of ACPI GPIOs: 0
[   54.348314] hci_uart_bcm serial1-0: No reset resource, using default baud rate
[   54.359337] intel-lpss 0000:00:1e.1: enabling device (0000 -> 0002)
[   54.361346] idma64 idma64.3: Found Intel integrated DMA 64-bit
[   54.362954] dw-apb-uart.3: ttyS6 at MMIO 0x8272c000 (irq = 21, base_baud = 115200) is a 16550A
[   54.363803] serial serial2: tty port ttyS6 registered
[   54.376464] intel-lpss 0000:00:1e.2: enabling device (0000 -> 0002)
[   54.377626] idma64 idma64.4: Found Intel integrated DMA 64-bit
[   54.383826] dw-apb-uart dw-apb-uart.2: Looking for DMA channel "rx" at index 1...
[   54.383906] idma64 idma64.0: private_candidate: dma0chan0 filter said false
[   54.383916] idma64 idma64.0: private_candidate: dma0chan1 filter said false
[   54.383979] dmaengine: __dma_request_channel: success (dma1chan0)
[   54.383982] dw-apb-uart dw-apb-uart.2: Looking for DMA channel "tx" at index 0...
[   54.384009] idma64 idma64.0: private_candidate: dma0chan0 filter said false
[   54.384011] idma64 idma64.0: private_candidate: dma0chan1 filter said false
[   54.384013] idma64 idma64.2: private_candidate: dma1chan0 busy
[   54.384016] dmaengine: __dma_request_channel: success (dma1chan1)
[   54.384399] BUG: unable to handle kernel NULL pointer dereference at 0000000000000090
[   54.384403] #PF error: [normal kernel read fault]
[   54.384405] PGD 0 P4D 0 
[   54.384408] Oops: 0000 [#1] SMP PTI
[   54.384411] CPU: 4 PID: 66 Comm: kworker/u16:1 Tainted: G        W         5.1.0-0.rc3.git1.1.fc31.x86_64 #1
[   54.384413] Hardware name: Apple Inc. MacBookPro13,3/Mac-A5C67F7XXXXXXXXX, BIOS 251.0.0.0.0 10/30/2018
[   54.384417] Workqueue: events_unbound async_run_entry_fn
[   54.384422] RIP: 0010:regulator_enable+0x11/0xc0
[   54.384424] Code: 43 0c e9 e2 fe ff ff e9 f3 21 00 00 66 66 2e 0f 1f 84 00 00 00 00 00 66 90 0f 1f 44 00 00 41 55 41 54 55 48 89 fd 48 83 ec 70 <4c> 8b af 90 00 00 00 65 48 8b 04 25 28 00 00 00 48 89 44 24 68 31
[   54.384426] RSP: 0018:ffffb0ec81b9fd80 EFLAGS: 00010282
[   54.384428] RAX: ffffffffb31159a0 RBX: ffff948d22f936c8 RCX: 0000000000000002
[   54.384430] RDX: ffffffffb368a690 RSI: 000000000000000a RDI: 0000000000000000
[   54.384432] RBP: 0000000000000000 R08: 0000000000000000 R09: 0000000000000000
[   54.384434] R10: 0000000000000001 R11: 0000000000000001 R12: ffff948d2d812000
[   54.384436] R13: ffff948d2d869200 R14: 0000000000000000 R15: ffff948d28e5fda8
[   54.384438] FS:  0000000000000000(0000) GS:ffff948d2e400000(0000) knlGS:0000000000000000
[   54.384439] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   54.384441] CR2: 0000000000000090 CR3: 0000000429f02004 CR4: 00000000003606e0
[   54.384443] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[   54.384445] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[   54.384446] Call Trace:
[   54.384454]  ? process_one_work+0x1e9/0x5e0
[   54.384457]  regulator_bulk_enable_async+0x12/0x20
[   54.384460]  async_run_entry_fn+0x39/0x160
[   54.384464]  process_one_work+0x272/0x5e0
[   54.384469]  worker_thread+0x50/0x3b0
[   54.384474]  kthread+0x108/0x140
[   54.384477]  ? process_one_work+0x5e0/0x5e0
[   54.384479]  ? kthread_park+0x90/0x90
[   54.384483]  ret_from_fork+0x3a/0x50
[   54.384490] Modules linked in: idma64 intel_lpss_pci(+) intel_lpss fuse bnep xt_CHECKSUM ipt_MASQUERADE tun bridge stp llc nf_conntrack_netbios_ns nf_conntrack_broadcast xt_CT ip6t_rpfilter ip6t_REJECT nf_reject_ipv6 ipt_REJECT nf_reject_ipv4 xt_conntrack ebtable_nat ip6table_nat ip6table_mangle ip6table_raw ip6table_security iptable_nat nf_nat iptable_mangle iptable_raw iptable_security nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c ip_set nfnetlink ebtable_filter ebtables ip6table_filter ip6_tables iptable_filter ip_tables sunrpc arc4 intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm irqbypass iTCO_wdt iTCO_vendor_support mei_hdcp crct10dif_pclmul crc32_pclmul mt76x0u mt76x0_common mt76x02_usb mt76_usb mt76x02_lib mt76 ghash_clmulni_intel applesmc input_polldev intel_cstate mac80211 intel_uncore intel_rapl_perf cfg80211 snd_hda_codec_generic uvcvideo snd_hda_codec_hdmi ledtrig_audio joydev videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common
[   54.384521]  snd_hda_intel videodev snd_hda_codec snd_hda_core media snd_hwdep snd_seq snd_seq_device snd_pcm i2c_i801 hci_uart snd_timer snd thunderbolt btqca btrtl mei_me btbcm soundcore btintel mei bluetooth sbs ecdh_generic rfkill sbshc apple_gmux pcc_cpufreq apple_bl hid_lenovo amdgpu chash amd_iommu_v2 gpu_sched i2c_algo_bit ttm drm_kms_helper nvme drm crc32c_intel nvme_core video ax88179_178a usbnet mii uas usb_storage
[... cut ...]
[   54.401519] idma64 idma64.0: private_candidate: dma0chan1 filter said false
[   54.401520] idma64 idma64.2: private_candidate: dma1chan0 busy
[   54.401522] idma64 idma64.2: private_candidate: dma1chan1 busy
[   54.401523] idma64 idma64.3: private_candidate: dma2chan0 filter said false
[   54.401525] idma64 idma64.3: private_candidate: dma2chan1 filter said false
[   54.401526] idma64 idma64.4: private_candidate: dma3chan0 busy
[   54.401528] dmaengine: __dma_request_channel: success (dma3chan1)```

roadrunner2, 2019-04-12:
Looks like you're running a 5.0 kernel and hence running into the bluetooth driver regression - see Dunedan/mbp-2016-linux#97 - near the end is a link to a patch.

closed 2019-04-16 by meeuw