roadrunner2/macbook12-spi-driver #24: Not working on 12-inch MacBook Early 2015

ComradeSparklePony, 2019-04-24:
I recently tried to use this driver to get support for the built-in keyboard and touchpad on my 12-inch MacBook Early 2015. I have tried both the version available in the Arch User Repository (installed with `yay`), as well as cb22's version (installed manually). Neither have been successful at getting the keyboard and touchpad to respond.

I'm pretty new to Linux, so there is a fair chance that I did something wrong.

I don't know if this is helpful, but `xinput list` does display both the keyboard and the touchpad.

Thank you.

roadrunner2, 2019-04-24:
So this would be the MacBook8,1(`cat /sys/class/dmi/id/product_name`)? Is the `applespi` module loaded at all (`lsmod | grep applespi`)? And are the required pxa2xx modules loaded (`lsmod | grep spi_pxa2xx`)? What does `dmesg | grep applespi` show? And does `libinput list-devices` show any "Apple SPI ..." devices?

ComradeSparklePony, 2019-04-24:
Yes, this is the Macbook8,1. The `applespi` and pxa2xx modules are loaded. The output of `dmesg | grep applespi` is:

    [    2.740531] applespi: loading out-of-tree module taints kernel.
    [    2.797013] applespi: spi=device probe done: spi-APP000D:00

`libinput list-devices` shows `Apple SPI Keyboard` and `Apple SPI Touchpad`.

roadrunner2, 2019-04-24:
Interesting: what appears to be missing is the "applespi: modeswitch done" log message, which seems to indicate the SPI messages aren't being delivered properly. This sounds a lot like cb22/macbook12-spi-driver#49, though that one so far has only appeared after doing a suspend/resume. But just make sure we're not running into a variant of cb22/macbook12-spi-driver#71, can you add `irqfixup` to your kernel command line (either edit `/boot/..../grub.cfg` before rebooting, or edit the command line at the grub prompt during boot) and see if that helps?

Assuming the `irqfixup` doesn't help any, can you next try the following: reload the `applespi` module using
```sh
sudo rmmod applespi ; sudo modprobe applespi debug=0xd07
```
Then click some keys on the (laptop) keyboard, and finally show me the output of `dmesg | applespi` again.

ComradeSparklePony, 2019-04-24:
`irqfixup` did not work. Apparently it was not found(?).

`dmesg | grep applespi`:

    [    3.043395] applespi: loading out-of-tree module taints kernel.
    [    3.100271] applespi: spi=device probe done: spi-APP000D:00
    [  361.263544]  applespi_remove+0xcf/0x160 [applespi]

When running `sudo rmmod applespi` there was an error: `rmmod: ERROR: Module applespi is in use`.  Trying to remove the modules that it depended on (`led_class` and `crc16`) would lead to a recursive "in use by"-- applespi is in use by led_class, led_class is in use by applespi. I used `dkms` to uninstall and reinstall applespi after `rmmod` and `modprobe` did not work.

Some other information that may or may not be helpful:

- I use Manjaro Linux running the Manjaro version of kernel 4.14
- The builtin keyboard on the laptop works fine in GRUB.

ComradeSparklePony, 2019-04-24:
Wait, I just gave it another try. I used `irqfixup` incorrectly (I tried executing it as a command instead of using it as a boot parameter). `irqfixup` did the trick.

Thank you very much for the help!

closed 2019-04-24 by ComradeSparklePony