roadrunner2/macbook12-spi-driver #26: compile on linux-5.1.10 -- iio_triggered_buffer  unknown symbol

ohquait, 2019-06-16:
DKMS make.log for applespi-0.1-1pclos2019 for kernel 5.1.10-pclos1 (x86_64)
Sun 16 Jun 2019 03:30:35 PM UTC
make -C /lib/modules/5.1.10-pclos1/build M=/var/lib/dkms/applespi/0.1-1pclos2019/build clean
make[1]: Entering directory '/usr/src/kernel-devel-5.1.10-pclos1'
  CLEAN   /var/lib/dkms/applespi/0.1-1pclos2019/build/Module.symvers
make[1]: Leaving directory '/usr/src/kernel-devel-5.1.10-pclos1'
make -C /lib/modules/5.1.10-pclos1/build M=/var/lib/dkms/applespi/0.1-1pclos2019/build modules
make[1]: Entering directory '/usr/src/kernel-devel-5.1.10-pclos1'
  CC [M]  /var/lib/dkms/applespi/0.1-1pclos2019/build/applespi.o
  CC [M]  /var/lib/dkms/applespi/0.1-1pclos2019/build/apple-ibridge.o
  CC [M]  /var/lib/dkms/applespi/0.1-1pclos2019/build/apple-ib-tb.o
  CC [M]  /var/lib/dkms/applespi/0.1-1pclos2019/build/apple-ib-als.o
  Building modules, stage 2.
  MODPOST 4 modules
**WARNING: "iio_triggered_buffer_setup" [/var/lib/dkms/applespi/0.1-1pclos2019/build/apple-ib-als.ko] undefined!
WARNING: "iio_triggered_buffer_cleanup" [/var/lib/dkms/applespi/0.1-1pclos2019/build/apple-ib-als.ko] undefined!**
  CC      /var/lib/dkms/applespi/0.1-1pclos2019/build/apple-ib-als.mod.o
  LD [M]  /var/lib/dkms/applespi/0.1-1pclos2019/build/apple-ib-als.ko
  CC      /var/lib/dkms/applespi/0.1-1pclos2019/build/apple-ib-tb.mod.o
  LD [M]  /var/lib/dkms/applespi/0.1-1pclos2019/build/apple-ib-tb.ko
  CC      /var/lib/dkms/applespi/0.1-1pclos2019/build/apple-ibridge.mod.o
  LD [M]  /var/lib/dkms/applespi/0.1-1pclos2019/build/apple-ibridge.ko
  CC      /var/lib/dkms/applespi/0.1-1pclos2019/build/applespi.mod.o
  LD [M]  /var/lib/dkms/applespi/0.1-1pclos2019/build/applespi.ko
make[1]: Leaving directory '/usr/src/kernel-devel-5.1.10-pclos1'
mkdir -p /lib/modules/5.1.10-pclos1/kernel/3rdparty/applespi
xz -f /lib/modules/5.1.10-pclos1/kernel/3rdparty/applespi/*.ko
/sbin/depmod -a 5.1.10-pclos1

ohquait, 2019-07-15:
with next test on linux-5.2.1::::
DKMS make.log for applespi-0.1-2pclos2019 for kernel 5.2.1-pclos1 (x86_64)
Mon 15 Jul 2019 01:30:33 PM UTC
make -C /lib/modules/5.2.1-pclos1/build M=/var/lib/dkms/applespi/0.1-2pclos2019/build  modules
make[1]: Entering directory '/usr/src/kernel-devel-5.2.1-pclos1'
  CC [M]  /var/lib/dkms/applespi/0.1-2pclos2019/build/applespi.o
  CC [M]  /var/lib/dkms/applespi/0.1-2pclos2019/build/apple-ibridge.o
  CC [M]  /var/lib/dkms/applespi/0.1-2pclos2019/build/apple-ib-tb.o
  CC [M]  /var/lib/dkms/applespi/0.1-2pclos2019/build/apple-ib-als.o
  Building modules, stage 2.
  MODPOST 4 modules
ERROR: "iio_triggered_buffer_setup" [/var/lib/dkms/applespi/0.1-2pclos2019/build/apple-ib-als.ko] undefined!
ERROR: "iio_triggered_buffer_cleanup" [/var/lib/dkms/applespi/0.1-2pclos2019/build/apple-ib-als.ko] undefined!
make[2]: *** [scripts/Makefile.modpost:91: __modpost] Error 1
make[1]: *** [Makefile:1593: modules] Error 2
make[1]: Leaving directory '/usr/src/kernel-devel-5.2.1-pclos1'
make: *** [Makefile:20: modules] Error 2
???

roadrunner2, 2019-07-19:
Sorry for the delay: looks like you don't have `CONFIG_IIO_TRIGGERED_BUFFER` set in your kernel config.

ohquait, 2019-07-22:
exact, that is it, hmm...
Now I just have to convince the boss to include this in the kernel config.

roadrunner2, 2019-07-22:
If you don't need/want the ambient-light-sensor, then you can also just comment out the als driver in the `Makefile` and `dkms.conf` and just build the rest.

ohquait, 2019-07-22:
> If you don't need/want the ambient-light-sensor, then you can also just comment out the als driver in the `Makefile` and `dkms.conf` and just build the rest.

that works, 
`# apt-get install dkms-applespi-0.1-3pclos2019*rpm
Reading Package Lists... Done
Building Dependency Tree... Done
Selecting dkms-applespi for 'dkms-applespi-0.1-3pclos2019.x86_64.rpm'
The following packages will be upgraded
   dkms-applespi (0.1-1pclos2019 => 0.1-3pclos2019)
1 upgraded, 0 newly installed, 0 removed and 0 not upgraded.
Need to get 0B/47.1kB of archives.
After unpacking 24.2kB of additional disk space will be used.
Committing changes...
Preparing                                ############################## [100%]
Updating / installing
  dkms-applespi-0.1-3pclos2019.x86_64    ############################## [100%]

Creating symlink /var/lib/dkms/applespi/0.1-3pclos2019/source ->
                 /usr/src/applespi-0.1-3pclos2019

DKMS: add Completed.

Preparing kernel 5.2.2-pclos1 for module build:
(This is not compiling a kernel, just preparing kernel symbols)
Storing current .config to be restored when complete
Running Generic preparation routine
make clean.....
using /proc/config.gz
make oldconfig.....
make prepare.....

Building module:
cleaning build area....
'make' SKIP_STACK_VALIDATION=1 all.....
cleaning build area....
cleaning kernel tree (make clean).....

DKMS: build Completed.

applespi.ko.xz:
 - Installation
   - Installing to /lib/modules/5.2.2-pclos1/kernel/3rdparty/applespi/

apple-ibridge.ko.xz:
 - Installation
   - Installing to /lib/modules/5.2.2-pclos1/kernel/3rdparty/applespi/

apple-ib-tb.ko.xz:
 - Installation
   - Installing to /lib/modules/5.2.2-pclos1/kernel/3rdparty/applespi/

depmod......

DKMS: install Completed.

------------------------------
Deleting module version: 0.1-1pclos2019
completely from the DKMS tree.
------------------------------
Done.
` 
Thanks!

closed 2019-07-23 by roadrunner2