roadrunner2/macbook12-spi-driver #27: suggestion / question

ohquait, 2019-07-24:
it is possible to include the corresponding .c, .h and kernelconf and make files from linux / driver / iio ... (... iio_triggered_buffer ...) for apple-ib-as in the source, so that these together with apple-ib-as compiled, regardless of the user's kernel-config?

roadrunner2, 2019-07-24:
Hmm, why just iio_triggered_buffer? Where do we stop? The drivers depend on a bunch of other drivers (at the very least all the modules and code enabled by SPI_PXA2XX_PCI, MFD_INTEL_LPSS_PCI, IIO, IIO_BUFFER, IIO_TRIGGER, IIO_TRIGGERED_BUFFER, and more). What distro are you running that doesn't have these enabled?

ohquait, 2019-07-24:
I use PCLinuxOS and Quait(this is for testbuilding before give back drivers and xfce to PCLinuxOS)
but Tex was so lovely and have enabled spi_pxa2xx_platform, SPI_PXA2XX_PCI, MFD_INTEL_LPSS_PCI,
just for me, also enabled in his kernel-config:
CONFIG_IIO=m
CONFIG_IIO_BUFFER=y
CONFIG_IIO_BUFFER_CB=m
CONFIG_IIO_BUFFER_HW_CONSUMER is not set
CONFIG_IIO_KFIFO_BUF=m
CONFIG_IIO_CONFIGFS is not set
CONFIG_IIO_TRIGGER=y
CONFIG_IIO_CONSUMERS_PER_TRIGGER=2

so i not found IIO_TRIGGERED_BUFFER
But i think, that's needed for apple-ib-als....?