roadrunner2/macbook12-spi-driver #28: Disable Touchpad While Typing: No synaptics properties on device

marc-git, 2019-07-27:
Just installed fresh version of Debian 10 on MBP 14,3. I have to say you have all done fantastic work on this. Four lines to get keyboard, touchpad and touchbar working! This is great. 

Has there ever been support for synaptics on the touchpad? I thought in the previous version I had installed (one file) that I had turned off the touchpad while typing, but that option is not available anymore (xfce).  

When I try to manually do it with the daemon:

`$syndaemon -k -i 1 -d
No synaptics properties on device 'Apple SPI Touchpad'.
`

I get pain. 

Has anyone looked in to this? Has it already been done and then dropped? I'm not so good with C but I could try and haul it back if it is laying around.

roadrunner2, 2019-07-28:
Are you sure you're using the synaptics Xorg driver and not libinput? The latter is the default these days, and from the error it sounds like you may actually be using that. If you are using libinput, then there are two things involved for dwt:
1. marking the keyboard as an internal keyboard
2. configuring dwt

For the first point, see the [Keyboard/Touchpad/Touchbar](https://gist.github.com/roadrunner2/1289542a748d9a104e7baec6a92f9cd7#keyboardtouchpadtouchbar) section of my gist, in particular the `61-libinput-local.hwdb` and `local-overrides.quirks` part.

Regarding the second point, dwt is usually enabled by default, but you can check it by running `libinput list-devices` and looking for the Touchpad device. As far as changing that value in xfce, I have no idea, sorry - in Gnome you can use the gnome-tweaks app for this.

However, if you really are (and want to keep) using the synaptics driver, then sorry, I haven't tried to use it, so not sure what the issue is, but it's likely not related to the `applespi` kernel driver (but if it is, I'm certainly open to fixing it if necessary).

marc-git, 2019-07-28:
I didn't land on that page at all, but instead on this one: https://github.com/roadrunner2/macbook12-spi-driver 
Maybe put a link there to that gist? You were right I was not using synaptic, just thought I must be.  I guess my knowledge is constantly becoming obsolete.  

Anyway you have solved my problem! thanks!  and also answered my second question about the touchpad speed brilliant. I may no longer need to bother booting MacOS soon. 

sidenote about txpower: I set mine to txpower 1. I had understood that this was actually increasing the power because the number used to be negative (dBs above threshold?) Like that I have a workable wifi even if not brilliant. 

Available for testing on my model if needed. Would like to give *something* back for this even if I'm not a good programmer.

marc-git, 2019-07-28:
I guess close the issue btw...

closed 2019-07-28 by marc-git

roadrunner2, 2019-07-29:
> I didn't land on that page at all, but instead on this one: https://github.com/roadrunner2/macbook12-spi-driver
> Maybe put a link there to that gist?

Done.

> sidenote about txpower: I set mine to txpower 1. I had understood that this was actually increasing the power because the number used to be negative (dBs above threshold?) Like that I have a workable wifi even if not brilliant.

0 dBm is 1mw; transmit power is usually higher than that, so the dBm values are positive - signal levels (i.e. receive power) are much lower, and hence generally have negative dBm values.

> Available for testing on my model if needed. Would like to give _something_ back for this even if I'm not a good programmer.

Thanks - more things to test will come up. But if you want to help, I'd suggest following the [mbp-2016-linux](/Dunedan/mbp-2016-linux) repo as that is where most of the discussions around these MBP's take place, and calls for testing are typically posted there too - this repo is limited to just the keyboard, trackpad, and touchbar driver.