roadrunner2/macbook12-spi-driver #31: Support for Retina MacBook Air

fl4co, 2019-10-21:
Hi, I have a 2019 Retina MacBook Air. Keyboard and trackpad never worked on Linux, so I was excited to try kernel 5.3 as I read about new MacBook drivers being merged.

I tried booting a live distro (Tails 4.0 rc1, kernel 5.3.2) but unfortunately keyboard and trackpad don't work.

I see that this driver does not explicitly support the Air, but maybe newest Airs can use an adapted version of the Pros' driver. I have no technical skill on the matter but I'm available for testing if necessary.

Thank you.

fl4co, 2019-10-21:
Ok, I've being doing some digging and it seems that newest Macs do not use the SPI interface for keyboard and trackpad, looks like the T2 chip manages them. I guess that's the explanation for this driver not working.

RED29, 2019-12-24:
@fl4co could you shine some light on your research? I mean I also have some problems with a macbook (pro) mid 2017 and how to know if the keyboard and touchpad are directed by the SPI or T2 chip. If I know it is the T2 chip I will stop spending hours. Thanks!

roadrunner2, 2019-12-24:
The 2o16 and 2017 models have an SPI based keyboard and trackpad; the T2 chip was introduced in the 2018 models.

roadrunner2, 2020-02-24:
I'm closing this as I think the question has been answered.

@fl4co you probably want take a look at the work being done for the [MacBookPro 15,* machines](https://github.com/Dunedan/mbp-2016-linux/issues/71) as I think those are closest to your machine.

@RED29 If you're having issues with getting the keyboard and trackpad working on your 2017 MBP (so presumably one of the 14,* models) then please open a new issue with more info.

closed 2020-02-24 by roadrunner2