roadrunner2/macbook12-spi-driver #32: Can't compile on 5.4.1

AnthonyRonning, 2019-12-02:
```
root@linux:/usr/src/applespi-0.1# sudo dkms install applespi/0.1

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.4.1-050401-generic...(bad exit status: 2)
ERROR (dkms apport): binary package for applespi: 0.1 not found
Error! Bad return status for module build on kernel: 5.4.1-050401-generic (x86_64)
Consult /var/lib/dkms/applespi/0.1/build/make.log for more information.
```

From the makelog: 

```
...
./include/asm-generic/bug.h:155:3: note: in expansion of macro ‘WARN’
   WARN(1, format);    \
   ^~~~
/var/lib/dkms/applespi/0.1/build/applespi.c:557:3: note: in expansion of macro ‘WARN_ONCE’
   WARN_ONCE(1, "Unknown msg type %d", type);
   ^~~~~~~~~
scripts/Makefile.build:265: recipe for target '/var/lib/dkms/applespi/0.1/build/applespi.o' failed
make[2]: *** [/var/lib/dkms/applespi/0.1/build/applespi.o] Error 1
Makefile:1652: recipe for target '/var/lib/dkms/applespi/0.1/build' failed
make[1]: *** [/var/lib/dkms/applespi/0.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.1-050401-generic'
Makefile:16: recipe for target 'all' failed
make: *** [all] Error 2
```

Makefile shows a lot, mostly all similar to the top 2 lines there. Let me know if a full log is needed/useful. I really just need touchbar so if there's an easier way then that'll be nice.

roadrunner2, 2019-12-02:
This is odd, because it compiles fine as part of the kernel with that WARN_ONCE in it. What exactly is the error (you've snipped it away)? Also, does manually running `make KERNELRELEASE=5.4.1-050401-generic` in `/usr/src/applespi-0.1/` also fail?

Since the `applespi` module is part of the kernel now, you can also just remove that entry from the `dkms.conf` file (and renumber the other entries).

AnthonyRonning, 2019-12-02:
I'll give that a try later, and with a full log - it was very long and the above part looked repetitive. 

I did try with 5.0.0 since I still had that installed and it worked. I couldn't find `spi_pxa2xx_platform` in my 5.4.1 kernel so it could be that? I was trying to recompile with that setting in the config last night, working through compiling errors. 

I'll also try to remove `applespi` too. Will report back, thanks!