roadrunner2/macbook12-spi-driver #33: Linux distro with kernel 5.3.9

RED29, 2019-12-24:
Macbook pro mid 2017 still wont work with a linux distro that uses a linux kernel 5.3.9. Are there extra steps? If yes, that maybe the readme should be adjusted because there is now that it work “out of box”.

roadrunner2, 2019-12-24:
Please elaborate on "won't work": what exactly isn't working? The keyboard? The touchbar? Something else? A lot of information, btw, can be found at https://github.com/Dunedan/mbp-2016-linux/issues/ .

Regarding the keyboard/trackpad driver here, yes, it's part of kernel 5.3, but not all distros are enabling it yet, so you may still need to build it yourself if your distro doesn't.

RED29, 2019-12-25:
Ok clear. Thanks for explaining. I found one Linux distro with kernel 5.4.6 that is working.

yoeldovidcohen, 2019-12-25:
What distro?

roadrunner2, 2020-01-10:
Btw., Fedora has it enabled as of kernel 5.3.10 (see https://bugzilla.redhat.com/show_bug.cgi?id=1769465), and Ubuntu 19.10 appears to have it enabled in their 5.3 kernels.