roadrunner2/macbook12-spi-driver #34: Can't build mpb15 branch on 5.4.8

TRPB, 2020-01-08:
The main branch works fine but if I try to compile the mbp15 branch I get the following error:

```
==> dkms install macbook12-spi-driver/0+git.303 -k 5.4.8-1-mbp
Error!  Build of applespi.ko failed for: 5.4.8-1-mbp (x86_64)
Consult the make.log in the build directory
/var/lib/dkms/macbook12-spi-driver/0+git.303/build/ for more information.
```

The log file it mentions doesn't show anything obviously wrong:

```
DKMS make.log for macbook12-spi-driver-0+git.303 for kernel 5.4.8-1-mbp (x86_64)
Wed  8 Jan 11:04:34 GMT 2020
make -C /usr/lib/modules/5.4.8-1-mbp/build M=/var/lib/dkms/macbook12-spi-driver/0+git.303/build modules
make[1]: Entering directory '/usr/lib/modules/5.4.8-1-mbp/build'
  CC [M]  /var/lib/dkms/macbook12-spi-driver/0+git.303/build/apple-ibridge.o
  CC [M]  /var/lib/dkms/macbook12-spi-driver/0+git.303/build/apple-ib-tb.o
  CC [M]  /var/lib/dkms/macbook12-spi-driver/0+git.303/build/apple-ib-als.o
  Building modules, stage 2.
  MODPOST 3 modules
  CC [M]  /var/lib/dkms/macbook12-spi-driver/0+git.303/build/apple-ib-als.mod.o
  CC [M]  /var/lib/dkms/macbook12-spi-driver/0+git.303/build/apple-ib-tb.mod.o
  CC [M]  /var/lib/dkms/macbook12-spi-driver/0+git.303/build/apple-ibridge.mod.o
  LD [M]  /var/lib/dkms/macbook12-spi-driver/0+git.303/build/apple-ib-als.ko
  LD [M]  /var/lib/dkms/macbook12-spi-driver/0+git.303/build/apple-ib-tb.ko
  LD [M]  /var/lib/dkms/macbook12-spi-driver/0+git.303/build/apple-ibridge.ko
make[1]: Leaving directory '/usr/lib/modules/5.4.8-1-mbp/build'

```

roadrunner2, 2020-01-08:
> The main branch works fine but if I try to compile the mbp15 branch I get the following error:
>
> ```
> ==> dkms install macbook12-spi-driver/0+git.303 -k 5.4.8-1-mbp
> Error!  Build of applespi.ko failed [...]

Something's odd on your system: it appears to want to build `applespi.ko`, which doesn't exist in this branch. I don't know where that is coming from, but since you seem to have previously built the main branch I'm wondering if there are some leftovers from that build?

mikeeq, 2020-01-08:
It's building fine for me on 5.4.8 but switching from media keys to F-unction keys (F1..F12) stopped working when pressing FN. 

https://travis-ci.com/mikeeq/mbp-fedora/jobs/273480312#L6965
```
make: Entering directory '/usr/src/kernels/5.4.8-200.mbp.fc31.x86_64'
  CC [M]  /opt/drivers/touchbar/apple-ibridge.o
  CC [M]  /opt/drivers/touchbar/apple-ib-tb.o
  CC [M]  /opt/drivers/touchbar/apple-ib-als.o
  Building modules, stage 2.
  MODPOST 3 modules
  CC [M]  /opt/drivers/touchbar/apple-ib-als.mod.o
Wed Jan  8 12:24:27 UTC 2020
  LD [M]  /opt/drivers/touchbar/apple-ib-als.ko
  CC [M]  /opt/drivers/touchbar/apple-ib-tb.mod.o
  LD [M]  /opt/drivers/touchbar/apple-ib-tb.ko
  CC [M]  /opt/drivers/touchbar/apple-ibridge.mod.o
  LD [M]  /opt/drivers/touchbar/apple-ibridge.ko
make: Leaving directory '/usr/src/kernels/5.4.8-200.mbp.fc31.x86_64'
```

TRPB, 2020-01-08:
> Something's odd on your system: it appears to want to build applespi.ko, which doesn't exist in this branch. I don't know where that is coming from, but since you seem to have previously built the main branch I'm wondering if there are some leftovers from that build?

Strange. I just tried trizen instead of yay (which presumably uses a different cache directory) and it gives the same issue. As does removing the package and reinstalling it.

TRPB, 2020-01-08:
Ah, I've found the issue. dkms.conf from the main branch is packaged in AUR along with the PKGBUILD.

roadrunner2, 2020-01-10:
Glad to hear you got it sorted out.

closed 2020-01-10 by roadrunner2