roadrunner2/macbook12-spi-driver #35: Received corrupted packet (crc mismatch)

leifliddy, 2020-01-08:
I'm running Fedora 31 on a macbook9,1 with kernel 5.4.7
The ```applespi``` was included in kernel 5.3.x, which worked great. 
But...since I upgraded to ```kernel 5.4.x```, I'm been experiencing these ```crc mismatch```  pretty frequently. 
ie
```
[52054.190342] applespi spi-APP000D:00: Received corrupted packet (crc mismatch)
```
When this error occurs, the keyboard does completely dark (no backlight) and I'm not able to adjust the keyboard backlight level at all. Also, the green led for 'Caps lock' no longer works. 
However, the keyboard and trackpad continue to function. So, this error seems to just affect the keyboard lighting functions. 

I never experienced this issue before. I never had any issues with kernel 5.3 (when the module was included in the kernel). Or before that, when I was compiling the module via dkms. 

**one other item of note on my system is that I'm using LUKS to encrypt my root/home/swap partition. Therefore, I've added these module to my initramfs image
```applespi spi_pxa2xx_platform intel_lpss_pci```

Is anyone else experiencing this issue w/ kernel 5.4?

roadrunner2, 2020-01-10:
I've occasionally seen this corrupted packet too, but very rarely. However, looking at my dmesg on 5.4 I do see more of them too, so it looks like something changed a bit in 5.4. Out of curiosity could you enable tracing of the crc errors with
```
echo 1 | sudo tee /sys/kernel/debug/tracing/events/applespi/applespi_bad_crc/enable
```
and then look at `/sys/kernel/debug/tracing/trace` after you run into some of the crc errors? I'm consistently seeing the following packet there:
```
 40 01 00 00 00 80 0a 0a 11 00 01 78 02 96 08 00...
```

Anyway, on to the hanging issue: I've pushed an update to this repo with a workaround to handle these cases (see 3a24873edfcd). You can just cherry-pick that commit on to your kernel, or you can use the driver in this repo instead. I'll send the patch upstream shortly.

leifliddy, 2020-01-17:
Sorry, I've have been busy the last few days. I'll apply that patch, and will send you the tracing data the next time I see that that error.

leifliddy, 2020-01-20:
I haven't experienced this issue since upgrading to kernel 5.4.10 a few days ago. I'll wait a few more days and see if it materializes again...

roadrunner2, 2020-01-22:
Ok, thanks for the update. Btw., you can safely enable tracing of crc errors anyway, as that has no impact in the absence of errors and negligible impact even when they do occur.

wdormann, 2020-10-21:
I have a Macbook14,3 and I'll say that I have a somewhat less than 50% chance that upon boot I'll have a working keyboard backlight.  Looking at dmesg output, the relevant difference is:

```
--- dmesg.backlight	2020-10-20 14:52:59.000000000 -0400
+++ dmesg.nobacklight	2020-10-20 14:53:26.000000000 -0400
@@ -1130,7 +1131,7 @@
 input: Apple SPI Keyboard as /devices/pci0000:00/0000:00:1e.3/pxa2xx-spi.5/spi_master/spi2/spi-APP000D:00/input/input5
 input: Apple SPI Touchpad as /devices/pci0000:00/0000:00:1e.3/pxa2xx-spi.5/spi_master/spi2/spi-APP000D:00/input/input6
 applespi: spi-device probe done: spi-APP000D:00
-applespi: modeswitch done.
+applespi spi-APP000D:00: Received corrupted packet (crc mismatch)
 usb 1-2: new high-speed USB device number 2 using xhci_hcd
 usb 6-1: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
 usb 6-1: New USB device found, idVendor=05ac, idProduct=100e, bcdDevice=45.28
```

The problem is that I don't seem to be able to do the `/sys/kernel/debug/tracing/events/applespi/applespi_bad_crc/enable` tracing option.  On my system (Ubuntu 20.04 w/ mainline kernel 5.8.3) my path stops at `/sys/kernel/debug/tracing/events/`.  There is no `applespi` directory in that path.

Thoughts?