roadrunner2/macbook12-spi-driver #36: Palm detection could be improved

gilescope, 2020-01-12:
Two finger scroll happens if the fingers are many inches apart and with such a wide touchpad it is very easy to have your palm trigger scrolling. I'd suggest ignoring any scrolling request that are more than two inches apart.

The flip side is which input is the one to use then for the mouse pointer? I'd suggest it's the one with more acceleration. The palm is likely to be fairly fixed when it does accidentally activate the touchpad.

I'm using the main Linux kernel - maybe you've already improved the palm detection here?

roadrunner2, 2020-01-15:
Scrolling, palm detection, acceleration, etc are all handled by libinput - the driver just provides raw touch events (well, the trackpad does seem also do some palm detection, but I haven't been able to figure out how to properly interpret the received values, so the driver does not currently provide any palm info).

To get things working properly, you need to make sure you A) have the proper hwdb entries (make sure you have `systemd`/`udev` version 242 or later, or use the overrides I provide on my [gist](https://gist.github.com/roadrunner2/1289542a748d9a104e7baec6a92f9cd7)), and B) have the proper libinput quirks (again, see my gist for details). If you're not happy with the behaviour, then the libinput quirks is where you want to start playing around.

roadrunner2, 2020-02-24:
Closing this, as I think the issue has been resolved as far as this driver is concerned.

closed 2020-02-24 by roadrunner2