roadrunner2/macbook12-spi-driver #38: Problem with a Touch Bar on macbook pro 13,3

1755, 2020-01-19:
Hi! I would like to use ubuntu 19.10 on my macbook pro 13,3. 

At first, I have tried a Live CD and everything except a sound works good. Also I have successfuly installed and run modules for touchbar while running the Live CD. But when I tried to reproduce these steps with installed ubuntu, I have not success.

I have done the following commands:
```
echo -e "\n# applespi\napplespi\nspi_pxa2xx_platform\nintel_lpss_pci" >> /etc/initramfs-tools/modules

apt install dkms
git clone https://github.com/roadrunner2/macbook12-spi-driver.git /usr/src/applespi-0.1
dkms install -m applespi -v 0.1
update-initramfs -u -k all
reboot
```

```
sergey@sergey-MacBookPro:~$ uname -a
Linux sergey-MacBookPro 5.3.0-26-generic #28-Ubuntu SMP Wed Dec 18 05:37:46 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

```
sergey@sergey-MacBookPro:~$ lsmod | grep apple
apple_ib_als           20480  0
industrialio_triggered_buffer    16384  1 apple_ib_als
industrialio           77824  3 industrialio_triggered_buffer,kfifo_buf,apple_ib_als
apple_ib_tb            24576  0
apple_ibridge          24576  2 apple_ib_als,apple_ib_tb
hid                   131072  2 apple_ib_als,apple_ibridge
applesmc               20480  0
input_polldev          20480  1 applesmc
apple_gmux             24576  0
apple_bl               20480  1 apple_gmux
applespi               40960  0
video                  49152  1 apple_gmux
```

```

It looks like the module is loaded successfuly but the touchbar does not work.

One more wired thing. When I changed parameters for the module in my initramfs-tools/modules:
```
apple_ib_tb idle_timeout=-1 dim_timeout=-1 fnmode=1
applespi
spi_pxa2xx_platform
intel_lpss_pci
```
And then updated it and rebooted I have parameters unchanged:

```
Module: apple_ib_tb
Parameter: dim_timeout --> -2
Parameter: fnmode --> 1
Parameter: idle_timeout --> 300
```

@roadrunner2 Can you help me please?

roadrunner2, 2020-01-22:
Can you look in your dmesg to see if there are any errors reported there?
```
dmesg | grep applespi
```

Also, you didn't by any chance blow away the original EFI partition on the disk did you? It's important that that partition be preserved (with its original contents) or the touchbar won't work - see also Dunedan/mbp-2016-linux#52. In general you probably want to also keep the MacOS installation in case there are firmware updates, but that's optional.

roadrunner2, 2020-02-24:
@1755: Any update on this?

xinhangliu, 2020-08-16:
I also encountered this problem.

```
$ uname -a
Linux xxxx 5.8.1-arch1-1 #1 SMP PREEMPT Wed, 12 Aug 2020 18:50:43 +0000 x86_64 GNU/Linux
```

```
$ lsmod | grep apple
applesmc               24576  0
apple_ib_als           20480  0
industrialio_triggered_buffer    16384  1 apple_ib_als
apple_mfi_fastcharge    20480  0
industrialio           90112  3 industrialio_triggered_buffer,kfifo_buf,apple_ib_als
apple_gmux             20480  0
apple_bl               20480  1 apple_gmux
apple_ib_tb            28672  0
apple_ibridge          28672  2 apple_ib_als,apple_ib_tb
hid                   143360  3 usbhid,apple_ib_als,apple_ibridge
applespi               49152  0
crc16                  16384  3 bluetooth,ext4,applespi
```

```
$ dmesg | grep apple
[    0.698957] applespi: loading out-of-tree module taints kernel.
[    0.698978] applespi: module verification failed: signature and/or required key missing - tainting kernel
[    0.928022] applespi spi-APP000D:00: modeswitch done.
[    2.137527] apple_gmux: Found gmux version 4.0.29 [indexed]
[    2.227887] usbcore: registered new device driver apple-mfi-fastcharge
[    3.022427] applesmc: key=902 fan=2 temp=46 index=45 acc=0 lux=0 kbd=0
[    3.022570] applesmc applesmc.768: hwmon_device_register() is deprecated. Please convert the driver to use hwmon_device_register_with_info().
```

harryiliffe, 2020-08-31:
I'm having the same issue
```
$ uname -a
Linux xxxxx 5.4.0-42-generic #46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

```
$ lsmod | grep apple
apple_ib_als           20480  0
industrialio_triggered_buffer    16384  1 apple_ib_als
industrialio           73728  3 industrialio_triggered_buffer,kfifo_buf,apple_ib_als
apple_ib_tb            24576  0
applesmc               20480  0
input_polldev          20480  1 applesmc
apple_ibridge          24576  2 apple_ib_als,apple_ib_tb
apple_gmux             24576  0
apple_bl               20480  1 apple_gmux
hid                   131072  4 usbhid,apple_ib_als,hid_generic,apple_ibridge
applespi               45056  0
video                  49152  1 apple_gmux
```
```
$ dmesg | grep applespi
[    0.916456] applespi: loading out-of-tree module taints kernel.
[    0.916488] applespi: module verification failed: signature and/or required key missing - tainting kernel
[    1.103312] applespi spi-APP000D:00: modeswitch done.
```

meichthys, 2023-02-22:
Also same here. I did have it working when I was dual-booting MacOS + Ubuntu, but now that i'm trying to wipe everything to install only Ubuntu, i can't seem to get this to work.

pedroresende, 2024-10-14:
@roadrunner2 I'm also getting the same erro message using Ubuntu 24.04, any ideas ?