roadrunner2/macbook12-spi-driver #39: DKMS package error

ivanderjmw, 2020-01-19:
I am quite unsure of this error, but I keep getting it on a fresh installation of Ubuntu 16.04 on a Macbook pro 14,2.

` dkms install -m applespi -v 0.1

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
make KERNELRELEASE=4.15.0-45-generic....(bad exit status: 2)
ER
[make-log.txt](https://github.com/roadrunner2/macbook12-spi-driver/files/4083300/make-log.txt)

ROR (dkms apport): binary package for applespi: 0.1 not found
Error! Bad return status for module build on kernel: 4.15.0-45-generic (x86_64)
Consult /var/lib/dkms/applespi/0.1/build/make.log for more information.
`
The log file mentioned is as attached.
Will really appreciate your help!

ivanderjmw, 2020-01-20:
SOLVED the issue by targeting a kernel, discovered it from this website [https://nixaid.com/linux-on-macbookpro/](url)
`dkms install -m applespi -v 0.1 -k 4.13.0-36-generic`

Somehow this worked.

closed 2020-01-20 by ivanderjmw

roadrunner2, 2020-01-22:
Sorry for the slow reply. I've just pushed a fix to allow the module(s) to be compiled with kernels at least as far back as 4.14 now. Though I'm surprised it compiled under 4.13 for you! Anyway, thanks for reporting this.