roadrunner2/macbook12-spi-driver #41: [Feature Request] delete key on the touchbar?

treymerkley, 2020-02-21:
Is there any way you could add a Windows-like delete key to go with the existing "backspace" delete key? Everything else about my keyboard setup with your touchbar is perfect, all I would like is a delete key. 

That also may not be how this works, which is fine. I just wanted to ask.

mikeeq, 2020-02-21:
Do you know that you can use FN+Backspace instead?

treymerkley, 2020-02-21:
I did not, and that's close enough. Thanks!

closed 2020-02-21 by treymerkley

roadrunner2, 2020-02-21:
Just FYI, there are also some other fn+key combos:
- fn+return = insert
- fn+up-arrow = page-up
- fn+down-arrow = page-down
- fn+left-arrow = home
- fn+right-arrow = end

Also, these are not specific to this driver, but are present on all Apple keyboards under Linux.