roadrunner2/macbook12-spi-driver #42: Manjaro Stable update 3-14-2020 breaks touchbar

treymerkley, 2020-03-16:
some reason, when I updated, this isn't working anymore. I uninstalled and reinstalled using both git and AUR, but neither gets it to load. I was on 5.4, but neither 4.19 or 5.5 work either. Neither does rebuilding with DKMS. 

Thanks!

roadrunner2, 2020-03-16:
Thanks for the info. Unfortunately, the dmesg output you gave above is for the keyboard driver, which is separate from the touchbar driver. So can you show me the output of the following commands:
```sh
dmesg | egrep -i 'apple-ib|ibridge'
lsusb -t
ls -l /sys/bus/hid/devices/*/driver
```

treymerkley, 2020-03-17:
Sure, thank you so much for helping out.

```
[trey@citron ~]$ sudo dmesg | sudo egrep -i 'apple-ib|ibridge'
[sudo] password for trey: 
[    1.971431] usb 1-3: Product: iBridge
```
```
[trey@citron ~]$ lsusb -t
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/8p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
```
```
[trey@citron ~]$ ls -l /sys/bus/hid/devices/*/driver
ls: cannot access '/sys/bus/hid/devices/*/driver': No such file or directory
```

roadrunner2, 2020-03-17:
Hmm, that `lsusb -t` output is very odd: there should be another 3 devices listed as children of the last item. For reference, this is what I see on my system:
```
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/8p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 3: Dev 2, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 3: Dev 2, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 3: Dev 2, If 2, Class=Human Interface Device, Driver=usbhid, 480M
    |__ Port 3: Dev 2, If 3, Class=Human Interface Device, Driver=usbhid, 480M
```

Also, the fact there are no hid devices is problematic.

All in all, this looks like some USB issue on your system, but it's hard to tell what. The fact that the iBridge device does seem to be detected at one point according to the kernel logs, but then later `lsusb -t` isn't showing it (and there are no hid devices), makes it sound like there may be some other error that is causing the iBridge device to go away. Can you look through your kernel log and see if there's anything weird related to USB in there?

roadrunner2, 2020-03-17:
Oh, two more things: what machine/model are you on? And what is the output from
```sh
ls -l /sys/bus/usb/devices/
```

treymerkley, 2020-03-17:
That is pretty weird. Does this help at all?

```
[trey@citron ~]$ ls -l /sys/bus/usb/devices
total 0
lrwxrwxrwx 1 root root 0 Mar 17 08:45 1-0:1.0 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-0:1.0
lrwxrwxrwx 1 root root 0 Mar 17 08:45 1-3 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-3
lrwxrwxrwx 1 root root 0 Mar 17 08:45 2-0:1.0 -> ../../../devices/pci0000:00/0000:00:14.0/usb2/2-0:1.0
lrwxrwxrwx 1 root root 0 Mar 17 08:45 3-0:1.0 -> ../../../devices/pci0000:00/0000:00:01.1/0000:04:00.0/0000:05:02.0/0000:07:00.0/usb3/3-0:1.0
lrwxrwxrwx 1 root root 0 Mar 17 08:45 4-0:1.0 -> ../../../devices/pci0000:00/0000:00:01.1/0000:04:00.0/0000:05:02.0/0000:07:00.0/usb4/4-0:1.0
lrwxrwxrwx 1 root root 0 Mar 17 08:45 5-0:1.0 -> ../../../devices/pci0000:00/0000:00:01.2/0000:7a:00.0/0000:7b:02.0/0000:7d:00.0/usb5/5-0:1.0
lrwxrwxrwx 1 root root 0 Mar 17 08:45 6-0:1.0 -> ../../../devices/pci0000:00/0000:00:01.2/0000:7a:00.0/0000:7b:02.0/0000:7d:00.0/usb6/6-0:1.0
lrwxrwxrwx 1 root root 0 Mar 17 08:45 usb1 -> ../../../devices/pci0000:00/0000:00:14.0/usb1
lrwxrwxrwx 1 root root 0 Mar 17 08:45 usb2 -> ../../../devices/pci0000:00/0000:00:14.0/usb2
lrwxrwxrwx 1 root root 0 Mar 17 08:45 usb3 -> ../../../devices/pci0000:00/0000:00:01.1/0000:04:00.0/0000:05:02.0/0000:07:00.0/usb3
lrwxrwxrwx 1 root root 0 Mar 17 08:45 usb4 -> ../../../devices/pci0000:00/0000:00:01.1/0000:04:00.0/0000:05:02.0/0000:07:00.0/usb4
lrwxrwxrwx 1 root root 0 Mar 17 08:45 usb5 -> ../../../devices/pci0000:00/0000:00:01.2/0000:7a:00.0/0000:7b:02.0/0000:7d:00.0/usb5
lrwxrwxrwx 1 root root 0 Mar 17 08:45 usb6 -> ../../../devices/pci0000:00/0000:00:01.2/0000:7a:00.0/0000:7b:02.0/0000:7d:00.0/usb6
```

```
[trey@citron ~]$ sudo dmesg | grep usb
[    0.628249] usbcore: registered new interface driver usbfs
[    0.628253] usbcore: registered new interface driver hub
[    0.628273] usbcore: registered new device driver usb
[    0.786585] usbcore: registered new interface driver usbserial_generic
[    0.786588] usbserial: USB Serial support registered for generic
[    1.652378] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.05
[    1.652379] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.652379] usb usb1: Product: xHCI Host Controller
[    1.652380] usb usb1: Manufacturer: Linux 5.5.8-1-MANJARO xhci-hcd
[    1.652381] usb usb1: SerialNumber: 0000:00:14.0
[    1.652912] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 5.05
[    1.652913] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.652914] usb usb2: Product: xHCI Host Controller
[    1.652914] usb usb2: Manufacturer: Linux 5.5.8-1-MANJARO xhci-hcd
[    1.652915] usb usb2: SerialNumber: 0000:00:14.0
[    1.654576] usb usb3: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.05
[    1.654578] usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.654579] usb usb3: Product: xHCI Host Controller
[    1.654580] usb usb3: Manufacturer: Linux 5.5.8-1-MANJARO xhci-hcd
[    1.654581] usb usb3: SerialNumber: 0000:07:00.0
[    1.654855] usb usb4: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 5.05
[    1.654856] usb usb4: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.654857] usb usb4: Product: xHCI Host Controller
[    1.654858] usb usb4: Manufacturer: Linux 5.5.8-1-MANJARO xhci-hcd
[    1.654858] usb usb4: SerialNumber: 0000:07:00.0
[    1.656355] usb usb5: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.05
[    1.656356] usb usb5: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.656357] usb usb5: Product: xHCI Host Controller
[    1.656359] usb usb5: Manufacturer: Linux 5.5.8-1-MANJARO xhci-hcd
[    1.656359] usb usb5: SerialNumber: 0000:7d:00.0
[    1.656641] usb usb6: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 5.05
[    1.656643] usb usb6: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.656644] usb usb6: Product: xHCI Host Controller
[    1.656645] usb usb6: Manufacturer: Linux 5.5.8-1-MANJARO xhci-hcd
[    1.656646] usb usb6: SerialNumber: 0000:7d:00.0
[    1.984848] usb 1-3: new high-speed USB device number 2 using xhci_hcd
[    2.005808] usb 1-3: New USB device found, idVendor=05ac, idProduct=8600, bcdDevice= 1.01
[    2.005813] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.005816] usb 1-3: Product: iBridge
[    2.005819] usb 1-3: Manufacturer: Apple Inc.
[    2.935347] usbcore: registered new interface driver brcmfmac
[    3.887263] usbcore: registered new interface driver usbhid
[    3.887266] usbhid: USB HID core driver
[    3.994856] usbcore: registered new interface driver uvcvideo
```

treymerkley, 2020-03-17:
Quick additional comment: I tried to connect to my laptop via bluetooth as well, but it says that no bluetooth device was found. I know that's mostly unrelated, but it might help to know.

roadrunner2, 2020-03-18:
> That is pretty weird. Does this help at all? I'm on a Apple Macbook Pro 15.4” Touch Bar A1707 - Core i7 - 3.1Ghz.

Ok, that's the MacBookPro14,3 then.

> ```
> [trey@citron ~]$ ls -l /sys/bus/usb/devices
> total 0
> lrwxrwxrwx 1 root root 0 Mar 17 08:45 1-0:1.0 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-0:1.0
> lrwxrwxrwx 1 root root 0 Mar 17 08:45 1-3 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-3
> lrwxrwxrwx 1 root root 0 Mar 17 08:45 2-0:1.0 -> ../../../devices/pci0000:00/0000:00:14.0/usb2/2-0:1.0
> [snip]
> ```

And this confirms the iBridge USB devices are truly missing (there should be four `1-3:1.x` devices there).

Since you say this happens on different kernels, I'm starting to suspect something is weird with the T1/iBridge chip itself. Dumb question, but have you completely powered off the laptop since this problem started?

roadrunner2, 2020-03-18:
> Quick additional comment: I tried to connect to my laptop via bluetooth as well, but it says that no bluetooth device was found. I know that's mostly unrelated, but it might help to know.

Assuming you're talking about kernel 5.5, see the bottom of Dunedan/mbp-2016-linux#29.

treymerkley, 2020-03-18:
Yeah, I've completely power cycled it, too. It shouldn't be a hardware thing; I dual boot, so I checked the Mac side before posting and both work there. Bluetooth is broken on all of the kernels for me, but if there's a fix I can wait for 5.5-9.

treymerkley, 2020-03-18:
Does this help? I was looking through `dmesg`.

```
[    1.922130] usb 1-3: new high-speed USB device number 2 using xhci_hcd
[    1.941678] usb 1-3: New USB device found, idVendor=05ac, idProduct=8600, bcdDevice= 1.01
[    1.941680] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    1.941681] usb 1-3: Product: iBridge
[    1.941682] usb 1-3: Manufacturer: Apple Inc.
```

mflor35, 2020-03-19:
The Touchpad just flickers and turns off. 

I am on `Macbook Pro 13,3` running Manjaro with `kernel  5.5.9` + patches from @aunali1 

https://github.com/aunali1/linux-mbp-arch

``` 
> lsusb -t                                                                                                                                                  13:26:57
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/8p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 2: Dev 2, If 0, Class=Vendor Specific Class, Driver=rt2800usb, 480M
    |__ Port 7: Dev 4, If 0, Class=Vendor Specific Class, Driver=, 480M
```
```
chac ) sudo dmesg | sudo egrep -i 'apple-ib|ibridge'                                                                                                                 13:32:40
[sudo] password for chac: 
[    2.302744] usb 1-3: Product: iBridge
[    3.639293] input: Apple Inc. iBridge as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.2/0003:05AC:8600.0001/input/input4
[    3.695553] apple-ibridge-hid 0003:05AC:8600.0001: input,hidraw0: USB HID v1.01 Keyboard [Apple Inc. iBridge] on usb-0000:00:14.0-3/input2
[    3.695766] apple-ibridge-hid 0003:05AC:8600.0002: hiddev0,hidraw1: USB HID v1.01 Device [Apple Inc. iBridge] on usb-0000:00:14.0-3/input3
[    3.955564] input: Apple Inc. iBridge as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.2/0003:05AC:8600.0001/input/input12
[    4.016738] apple-ibridge-hid 0003:05AC:8600.0001: input,hidraw0: USB HID v1.01 Keyboard [Apple Inc. iBridge] on usb-0000:00:14.0-3/input2
[    4.022169] apple-ibridge-hid 0003:05AC:8600.0002: hiddev0,hidraw1: USB HID v1.01 Device [Apple Inc. iBridge] on usb-0000:00:14.0-3/input3
[    4.155846] input: Apple Inc. iBridge as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.2/0003:05AC:8600.0001/input/input15
[    4.214060] apple-ibridge-hid 0003:05AC:8600.0001: input,hidraw0: USB HID v1.01 Keyboard [Apple Inc. iBridge] on usb-0000:00:14.0-3/input2
[    4.214748] apple-ibridge-hid 0003:05AC:8600.0002: hiddev0,hidraw1: USB HID v1.01 Device [Apple Inc. iBridge] on usb-0000:00:14.0-3/input3
[    4.449663] apple-ibridge APP7777:00: tb: Touchbar deactivated
```

treymerkley, 2020-03-19:
@mflor35 add it to the update thread on Manjaro, too. Someone who works on it might have insight if everyone with the problem mentions there.

roadrunner2, 2020-03-20:
@mflor35 Thanks for your info. First of all, this furthermore appears to confirm it's truly a Manjaro (or least Linux) issue, not a one-off h/w problem or something. Second, your dmesg output and your comment that it flickers (I presume you meant "touchbar", not "trackpad") are very interesting: it looks like something is actively causing the iBridge USB device to be removed/disabled.

Can one of you show me the output of
```
sudo grep . /sys/bus/usb/devices/usb1/1-3/*
sudo grep . /sys/bus/usb/devices/usb1/1-3/power/*
sudo grep . /sys/bus/usb/devices/usb1/1-3/port/power/*
ls -l /sys/bus/usb/devices/usb1/1-3/driver
```

Also, I agree, getting some Manjaro devs involved would be good, as it appears to be something recent and particular to that distro that is causing the problem (something related to USB).

treymerkley, 2020-03-20:
```
[trey@citron ~]$ sudo grep . /sys/bus/usb/devices/usb1/1-3/*
[sudo] password for trey: 
/sys/bus/usb/devices/usb1/1-3/authorized:1
/sys/bus/usb/devices/usb1/1-3/avoid_reset_quirk:0
/sys/bus/usb/devices/usb1/1-3/bcdDevice:0101
/sys/bus/usb/devices/usb1/1-3/bDeviceClass:ef
/sys/bus/usb/devices/usb1/1-3/bDeviceProtocol:01
/sys/bus/usb/devices/usb1/1-3/bDeviceSubClass:02
/sys/bus/usb/devices/usb1/1-3/bMaxPacketSize0:64
/sys/bus/usb/devices/usb1/1-3/bNumConfigurations:3
/sys/bus/usb/devices/usb1/1-3/busnum:1
Binary file /sys/bus/usb/devices/usb1/1-3/descriptors matches
/sys/bus/usb/devices/usb1/1-3/dev:189:1
/sys/bus/usb/devices/usb1/1-3/devnum:2
/sys/bus/usb/devices/usb1/1-3/devpath:3
/sys/bus/usb/devices/usb1/1-3/devspec:(null)
grep: /sys/bus/usb/devices/usb1/1-3/driver: Is a directory
grep: /sys/bus/usb/devices/usb1/1-3/ep_00: Is a directory
grep: /sys/bus/usb/devices/usb1/1-3/firmware_node: Is a directory
/sys/bus/usb/devices/usb1/1-3/idProduct:8600
/sys/bus/usb/devices/usb1/1-3/idVendor:05ac
/sys/bus/usb/devices/usb1/1-3/ltm_capable:no
/sys/bus/usb/devices/usb1/1-3/manufacturer:Apple Inc.
/sys/bus/usb/devices/usb1/1-3/maxchild:0
grep: /sys/bus/usb/devices/usb1/1-3/port: Is a directory
grep: /sys/bus/usb/devices/usb1/1-3/power: Is a directory
/sys/bus/usb/devices/usb1/1-3/product:iBridge
/sys/bus/usb/devices/usb1/1-3/quirks:0x0
/sys/bus/usb/devices/usb1/1-3/removable:fixed
grep: /sys/bus/usb/devices/usb1/1-3/remove: Permission denied
/sys/bus/usb/devices/usb1/1-3/rx_lanes:1
/sys/bus/usb/devices/usb1/1-3/speed:480
grep: /sys/bus/usb/devices/usb1/1-3/subsystem: Is a directory
/sys/bus/usb/devices/usb1/1-3/tx_lanes:1
/sys/bus/usb/devices/usb1/1-3/uevent:MAJOR=189
/sys/bus/usb/devices/usb1/1-3/uevent:MINOR=1
/sys/bus/usb/devices/usb1/1-3/uevent:DEVNAME=bus/usb/001/002
/sys/bus/usb/devices/usb1/1-3/uevent:DEVTYPE=usb_device
/sys/bus/usb/devices/usb1/1-3/uevent:DRIVER=usb
/sys/bus/usb/devices/usb1/1-3/uevent:PRODUCT=5ac/8600/101
/sys/bus/usb/devices/usb1/1-3/uevent:TYPE=239/2/1
/sys/bus/usb/devices/usb1/1-3/uevent:BUSNUM=001
/sys/bus/usb/devices/usb1/1-3/uevent:DEVNUM=002
/sys/bus/usb/devices/usb1/1-3/urbnum:15
/sys/bus/usb/devices/usb1/1-3/version: 2.00
```

```
[trey@citron ~]$ sudo grep . /sys/bus/usb/devices/usb1/1-3/power/*
/sys/bus/usb/devices/usb1/1-3/power/active_duration:3307
/sys/bus/usb/devices/usb1/1-3/power/async:enabled
/sys/bus/usb/devices/usb1/1-3/power/autosuspend:2
/sys/bus/usb/devices/usb1/1-3/power/autosuspend_delay_ms:2000
/sys/bus/usb/devices/usb1/1-3/power/connected_duration:223357
/sys/bus/usb/devices/usb1/1-3/power/control:auto
/sys/bus/usb/devices/usb1/1-3/power/level:auto
/sys/bus/usb/devices/usb1/1-3/power/persist:1
/sys/bus/usb/devices/usb1/1-3/power/runtime_active_kids:0
/sys/bus/usb/devices/usb1/1-3/power/runtime_active_time:3169
/sys/bus/usb/devices/usb1/1-3/power/runtime_enabled:enabled
/sys/bus/usb/devices/usb1/1-3/power/runtime_status:suspended
/sys/bus/usb/devices/usb1/1-3/power/runtime_suspended_time:220049
/sys/bus/usb/devices/usb1/1-3/power/runtime_usage:0
```

```
[trey@citron ~]$ sudo grep . /sys/bus/usb/devices/usb1/1-3/port/power/*
/sys/bus/usb/devices/usb1/1-3/port/power/async:enabled
grep: /sys/bus/usb/devices/usb1/1-3/port/power/autosuspend_delay_ms: Input/output error
/sys/bus/usb/devices/usb1/1-3/port/power/control:auto
/sys/bus/usb/devices/usb1/1-3/port/power/runtime_active_kids:0
/sys/bus/usb/devices/usb1/1-3/port/power/runtime_active_time:231529
/sys/bus/usb/devices/usb1/1-3/port/power/runtime_enabled:enabled
/sys/bus/usb/devices/usb1/1-3/port/power/runtime_status:active
/sys/bus/usb/devices/usb1/1-3/port/power/runtime_suspended_time:0
/sys/bus/usb/devices/usb1/1-3/port/power/runtime_usage:0
```

```
[trey@citron ~]$ ls -l /sys/bus/usb/devices/usb1/1-3/driver
lrwxrwxrwx 1 root root 0 Mar 20 08:27 /sys/bus/usb/devices/usb1/1-3/driver -> ../../../../../bus/usb/drivers/usb
```

mflor35, 2020-03-20:
@roadrunner2 yes! I meant touchbar.

@treymerkley I'll add my info to the manjaro forums

roadrunner2, 2020-03-22:
Thanks for the outputs. So I see two things, compared to mine (I'm running a 5.5 kernel):
1. `/sys/bus/usb/devices/usb1/1-3/` is a missing a `bConfigurationValue` and a `configuration` entry, as well as a `bNumInterfaces` (this is probably because of the former two missing)
2. `/sys/bus/usb/devices/usb1/1-3/power/runtime_status` is showing `suspended` instead of `active`

What happens if you manually unbind and then rebind the iBridge USB device?

treymerkley, 2020-03-22:
Hey it worked! It starts back automatically. However, it doesn't survive reboot, which makes sense, since it's a temporary binding.

roadrunner2, 2020-03-23:
So this pretty conclusively demonstrates that it's something in the latest Manjaro that is messing with the iBridge usb device, and it's neither a kernel nor a driver issue. At this point I'm afraid I can't help you any further in figuring out what that could be, though.

I'll leave this ticket open for now - please report back here if/when you find out what is causing it.

treymerkley, 2020-03-23:
That's totally fine, and thank you so much for helping diagnose the issue. I'll keep updating this ticket, but here's a link to the forum thread I created over on Manjaro:

mflor35, 2020-03-24:
So I decided to install `Ubuntu 20.04 LTS`  with kernel `5.5.11-050511-generic` and the touch bar works as expected.

I think @roadrunner2 is right. I looked to see if people running Fedora were having an issue but I couldn't find anything. Looks like It's a Manjaro specific issue.

treymerkley, 2020-03-24:
Can you reply to that forum thread with this info? The usual response to Mac hardware from penguins is generally "Linux generally doesn't support Mac very well that's why you should have gotten a different machine, " so your test will help legitimize the problem.

roadrunner2, 2020-03-28:
A thought occurred to me today: you might try the following patch to try and figure out what is causing the iBridge's HID devices to disappear again during boot. As you can see, the patch will cause the touchbar module to just dump a listing of all processes on the system into `/tmp/hid-remove.ps` when the HID devices are removed. Maybe that will give some indication of what is triggering the issue.

```patch
--- a/apple-ib-tb.c
+++ b/apple-ib-tb.c
@@ -1097,6 +1097,11 @@ static void appletb_remove(struct hid_device *hdev)
                appletb_mark_active(tb_dev, false);
 
                dev_info(tb_dev->log_dev, "Touchbar deactivated\n");
+
+               {
+                       char *args[] = { "/bin/sh", "-c", "ps -ef > /tmp/hid-remove.ps", NULL };
+                       call_usermodehelper(args[0], args, NULL, UMH_WAIT_PROC);
+               }
        }
 
        report_info = appletb_get_report_info(tb_dev, hdev);
```

mflor35, 2020-04-04:
@treymerkley I followed up on your[ post ](https://forum.manjaro.org/t/manjaro-stable-update-3-14-2020-breaks-usb-ibridge-interface-on-macbook-pro/131106/3)at the Majaro forum.

@roadrunner2 I applied the patch but I do not see anything under `/tmp/hid-remove.ps` 
I removed the module with
```
sudo dkms remove -m applespi -v 0.1 --all
```
Reboot my machine and reinstalled the module
```
sudo dkms install -m applespi/0.1
```
Am I missing something?

unbinding and rebinding the iBridge USB device works for now. I appreciate the help!

renatolfc, 2020-04-05:
> What happens if you manually unbind and then rebind the iBridge USB device?
> 
> ```
> echo '1-3' | sudo tee /sys/bus/usb/drivers/usb/unbind
> echo '1-3' | sudo tee /sys/bus/usb/drivers/usb/bind
> ```

I'm on Arch (not Manjaro) and was/am having the same issue. The above fixes my issues as well.

roadrunner2, 2020-04-06:
@mflor35 The point of the patch is to try and show what process might be responsible for removing the USB HID devices during *boot* - you said the touchbar flickers during boot, indicating the devices show up but then get removed again. So the way to test is to reboot your machine after having installed the module (i.e. since it's now installed, just reboot and see if the `/tmp/hid-remove.ps` file shows up now).

roadrunner2, 2020-04-06:
@renatolfc That is very interesting - this would indicate the problem is more widespread then. Did you recently upgrade anything, and if so what? Maybe that can help narrow down the culprit.

renatolfc, 2020-04-06:
@roadrunner2 I noticed the problem some 2 weeks ago and was trying to debug the boot process to try and figure what was causing the problem.

I had assumed this was being caused by the set of custom systemd units I'd written to make this MacBook 14,3 more usable, so I was debugging it with `systemd-analyze plot`/`systemd-analyze dump` but with no luck. What I noticed from `systemd-analyze dump` was that systemd found the iBridge and triggers the start of usbmuxd. Stopping/restarting the usbmuxd doesn't trigger the "bug" though...

```
-> Unit sys-devices-pci0000:00-0000:00:14.0-usb1-1\x2d3.device:
        Description: iBridge
        Instance: n/a
        Unit Load State: loaded
        Unit Active State: active
        State Change Timestamp: Mon 2020-04-06 07:20:48 -03
        Inactive Exit Timestamp: Mon 2020-04-06 07:20:48 -03
        Active Enter Timestamp: Mon 2020-04-06 07:20:48 -03
        Active Exit Timestamp: n/a
        Inactive Enter Timestamp: n/a
        May GC: no
        Need Daemon Reload: no
        Transient: no
        Perpetual: no
        Garbage Collection Mode: inactive
        Slice: n/a
        CGroup: n/a
        CGroup realized: no
        Invocation ID: 7f5d07fd38f543249ab08ba37ef9e257
        Following Set Member: dev-bus-usb-001-005.device
        Wants: usbmuxd.service (origin-udev)
        References: usbmuxd.service (origin-udev)
        StopWhenUnneeded: no
        RefuseManualStart: no
        RefuseManualStop: no
        DefaultDependencies: yes
        OnFailureJobMode: replace
        IgnoreOnIsolate: yes
        Device State: plugged
        Sysfs Path: /sys/devices/pci0000:00/0000:00:14.0/usb1/1-3
        Found: found-udev
        udev SYSTEMD_WANTS: usbmuxd.service
```

renatolfc, 2020-04-06:
Well, it turns out it *was* usbmuxd's fault. At least in my system.

I just did a `pacman -Rdd usbmuxd` (essentially removing usbmuxd without doing dependency resolution), removed the commands mentioned in https://github.com/roadrunner2/macbook12-spi-driver/issues/42#issuecomment-602160740 from my custom systemd units and issued a reboot. No deactivation of the TouchBar was detected.

I'm assuming [this commit](https://github.com/libimobiledevice/usbmuxd/commit/8a69f1a78a58476f77b66916091d2405d0cd815f) is the culprit... :thinking:

renatolfc, 2020-04-06:
A quick and dirty fix (@mflor35) would be to write and activate a unit file such as the one below. The actual fix would be to work with the usbmuxd developers to update udev rules so that it doesn't bind to the iBridge. usbmuxd is a dependency to upower and libimobiledevice, I don't think uninstalling it would be an option.

```ini
[Unit]
Description=Re-enable MacBook 14,3 TouchBar
Before=display-manager.service
After=usbmuxd.service

[Service]
Type=oneshot
ExecStart=/bin/sh -c "echo '1-3' > /sys/bus/usb/drivers/usb/unbind"
ExecStart=/bin/sh -c "echo '1-3' > /sys/bus/usb/drivers/usb/bind"
RemainAfterExit=yes
TimeoutSec=0

[Install]
WantedBy=multi-user.target
```

roadrunner2, 2020-04-07:
@renatolfc Good catch! Not sure why usbmux is binding to the iBridge/T1-chip (the commit comment is very sparse... and wrong AFAICT, as product-id 0x8600 is the T1 chip,  not the T2 chip). So definitely need to take it up with them. To that end I've filed bug libimobiledevice/usbmuxd#138 with them.

rickmark, 2020-04-07:
Hey There,

So the T2 as PID_8600 is how it appears when booted from ramdisk in restore mode.  (A phase of recovery).  These changes were to support restoring a DFU mode T2 on another MacBook, but I may not have taken into account people running linux on T2 based Macs.  From the LIMD perspective, that computer has two T2's attached (one internal on the synthetic USB bus) and the other on an external port.  Clearly, we cannot takeover the T2 on the internal, but also we are in a bind because not controlling the T2 on the external breaks idevicerestore.

R

roadrunner2, 2020-04-08:
@rickmark Here's a thought: do the remote T1's show up as removable? Because on a MBP itself the internal one shows up as fixed:
```
% grep . /sys/bus/usb/devices/1-3/{idProduct,product,removable}
/sys/bus/usb/devices/1-3/idProduct:8600
/sys/bus/usb/devices/1-3/product:Apple T1 Controller
/sys/bus/usb/devices/1-3/removable:fixed
```
If they do show up as removable, then adding a `ATTR{removable}=="removable"` or similar to the udev rules might do the trick.

rickmark, 2020-04-08:
I've spent some time today trying to boot into recovery (but my computer is evil and cursed).  I think you may be on to something because IIRC it's the hub's port that confers that attribute.  I don't have a T2 Mac with Linux installed at the moment.

R

treymerkley, 2020-04-12:
Hey, sorry I disappeared for a while, I didn't have network access. 

@rickmark is there anything you want me to try on my machine?

rickmark, 2020-04-12:
@treymerkley - I don't currently have a machine I can test this on at the moment, but I need to test and see if there's any details that differ when the T2 vs the T1 in the USB descriptor.  I can probably get to this after some work I'm finishing on checkra1n and pongoOS.  (If you're not doing anything with the T2 you can probably use the removable trick as a workaround until I get that data)

treymerkley, 2020-04-13:
@renatolfc I'm having a hard time finding info on how to set this up. Can you link something?

renatolfc, 2020-04-15:
@treymerkley Sure. Take a look at [the Arch Wiki](https://wiki.archlinux.org/index.php/Systemd#Writing_unit_files).

Essentially, what you need is to write a file at `/etc/systemd/system` and add the contents of https://github.com/roadrunner2/macbook12-spi-driver/issues/42#issuecomment-609749027 to it. (For example, I've named mine `/etc/systemd/system/macbook-quirks.service`.)

Then, enable your unit with `systemctl enable macbook-quirks.service`. This should hopefully fix your issues.

treymerkley, 2020-05-07:
@renatolfc it fixed my issues right until now. Now it's giving me the error: 
```
[trey@citron ~]$ systemctl status macbook-quirks.service
● macbook-quirks.service - Re-enable MacBook 14,3 TouchBar
     Loaded: loaded (/etc/systemd/system/macbook-quirks.service; enabled; vendor preset: disabled)
     Active: failed (Result: exit-code) since Thu 2020-05-07 07:57:47 MDT; 33min ago
    Process: 758 ExecStart=/bin/sh -c echo '1-3' > /sys/bus/usb/drivers/usb/unbind (code=exited, status=1/FAILURE)
   Main PID: 758 (code=exited, status=1/FAILURE)

May 07 07:57:47 citron systemd[1]: Starting Re-enable MacBook 14,3 TouchBar...
May 07 07:57:47 citron sh[758]: /bin/sh: line 0: echo: write error: No such device
May 07 07:57:47 citron systemd[1]: macbook-quirks.service: Main process exited, code=exited, status=1/FAILURE
May 07 07:57:47 citron systemd[1]: macbook-quirks.service: Failed with result 'exit-code'.
May 07 07:57:47 citron systemd[1]: Failed to start Re-enable MacBook 14,3 TouchBar.
```

renatolfc, 2020-05-08:
Taking a look at Manjaro's updated package list, it seems there's been quite a few kernel updates in the last few days. Perhaps a kernel update broke it?

What happens if you follow the instructions in https://github.com/roadrunner2/macbook12-spi-driver/issues/42#issuecomment-601557029? Any changes from https://github.com/roadrunner2/macbook12-spi-driver/issues/42#issuecomment-602160740?

Unfortunately, I accidentally destroyed my Arch install this weekend, so I won't be able to debug this. 😬

treymerkley, 2020-05-19:
No biggie, fixed itself. Right now it's not as reliable as a real fix, but it's pretty great for now. Thanks again!

divad42, 2020-07-29:
I am having the same issue on Debian sid (kernel version 5.7.0).  The workaround works for me.  Does this suggest that it is not an Arch/Manjaro-specific issue, but rather a kernel-related one?  Arch/Manjaro users and Debian sid users are on a later version of the kernel than most people, after all.

renatolfc, 2020-07-29:
@divad42 Neither. [usbmuxd](https://github.com/libimobiledevice/usbmuxd/issues/138) is the [culprit](https://github.com/roadrunner2/macbook12-spi-driver/issues/42#issuecomment-609735938).

fannullone, 2021-01-02:
I had the same problem of touchbar not working on my mbp13,3 fedora core 33 and kernel 5.9
I could get it back by the two actions below:
- apply patch to applespi module because dkms was fail compile: https://github.com/roadrunner2/macbook12-spi-driver/pull/55/commits/f406bb28dd77ef2cad327fc9e3da2fe68416dffa
- rebind the usb as described above:
echo '1-3' | sudo tee /sys/bus/usb/drivers/usb/unbind
echo '1-3' | sudo tee /sys/bus/usb/drivers/usb/bind

I know this is not a proper fix but at least I could get the touchbar back :-)
Thanks everybody for help!

ikonopistsev, 2021-04-10:
> I had the same problem of touchbar not working on my mbp13,3 fedora core 33 and kernel 5.9
> I could get it back by the two actions below:
> 
>     * apply patch to applespi module because dkms was fail compile: [f406bb2](https://github.com/roadrunner2/macbook12-spi-driver/commit/f406bb28dd77ef2cad327fc9e3da2fe68416dffa)
> 
>     * rebind the usb as described above:
>       echo '1-3' | sudo tee /sys/bus/usb/drivers/usb/unbind
>       echo '1-3' | sudo tee /sys/bus/usb/drivers/usb/bind
> 
> 
> I know this is not a proper fix but at least I could get the touchbar back :-)
> Thanks everybody for help!

You my hero!)

danielealbano, 2021-05-14:
Found this today, I was facing the same exact problem on Ubuntu 20.10, the unbind / bind fixed my problem as well and finally I can unmap the 3-fingers-swipe-down gesture I had mapped to the ESC key :joy: 

Thanks!!!

Cliffback, 2022-05-21:
> @treymerkley Sure. Take a look at [the Arch Wiki](https://wiki.archlinux.org/index.php/Systemd#Writing_unit_files).
> 
> Essentially, what you need is to write a file at `/etc/systemd/system` and add the contents of [#42 (comment)](https://github.com/roadrunner2/macbook12-spi-driver/issues/42#issuecomment-609749027) to it. (For example, I've named mine `/etc/systemd/system/macbook-quirks.service`.)
> 
> Then, enable your unit with `systemctl enable macbook-quirks.service` (assuming you named the file `macbook-quirks.service`). This should hopefully fix your issues.

This did it for me as well! I have spent hours trying to get the Touch Bar to work, on the latest Pop Os (kernel 5.17) on my MacbookPro13,3 and the touchbar just appeared for two seconds at boot. This did the trick. Thank you so much for figuring it out!

scobiej, 2022-07-30:
> @renatolfc it fixed my issues right until now. Now it's giving me the error:
> 
> ```
> [trey@citron ~]$ systemctl status macbook-quirks.service
> ● macbook-quirks.service - Re-enable MacBook 14,3 TouchBar
>      Loaded: loaded (/etc/systemd/system/macbook-quirks.service; enabled; vendor preset: disabled)
>      Active: failed (Result: exit-code) since Thu 2020-05-07 07:57:47 MDT; 33min ago
>     Process: 758 ExecStart=/bin/sh -c echo '1-3' > /sys/bus/usb/drivers/usb/unbind (code=exited, status=1/FAILURE)
>    Main PID: 758 (code=exited, status=1/FAILURE)
> 
> May 07 07:57:47 citron systemd[1]: Starting Re-enable MacBook 14,3 TouchBar...
> May 07 07:57:47 citron sh[758]: /bin/sh: line 0: echo: write error: No such device
> May 07 07:57:47 citron systemd[1]: macbook-quirks.service: Main process exited, code=exited, status=1/FAILURE
> May 07 07:57:47 citron systemd[1]: macbook-quirks.service: Failed with result 'exit-code'.
> May 07 07:57:47 citron systemd[1]: Failed to start Re-enable MacBook 14,3 TouchBar.
> ```
> 
> Have you ever seen this or have any advice? Thanks!

Seems like the devices have not been created when it runs. I've got the same issue on Ubuntu 22.04. Solved by sleeping the service for 2 seconds before it was attempted and all worked but there must be a much better way.

`
[Unit]
Description=Re-enable MacBook 14,3 TouchBar
Before=display-manager.service

[Service]
Type=oneshot
ExecStartPre=/bin/sleep 2
ExecStart=/bin/sh -c "echo '1-3' > /sys/bus/usb/drivers/usb/unbind"
ExecStart=/bin/sh -c "echo '1-3' > /sys/bus/usb/drivers/usb/bind"
RemainAfterExit=yes
TimeoutSec=0

[Install]
WantedBy=multi-user.target
`

rob-hills, 2022-10-12:
I'm installing Ubuntu 22.04 on a MBP 14,3 (Linux 15.5.0-50) and had issues with getting my touchbar to work.  The original workaround posted by @roadrunner2 in his [2020 comment](https://github.com/roadrunner2/macbook12-spi-driver/issues/42#issuecomment-602160740):

> What happens if you manually unbind and then rebind the iBridge USB device?
> 
> ```
> echo '1-3' | sudo tee /sys/bus/usb/drivers/usb/unbind
> echo '1-3' | sudo tee /sys/bus/usb/drivers/usb/bind
> ```

resolved my problem.  Now, thanks to @scobiej 's [suggested service configuration above](https://github.com/roadrunner2/macbook12-spi-driver/issues/42#issuecomment-1200289798) I have it working after reboot.

Hopefully the underlying issue (apparently with usbmuxd) will be resolved sooner rather than later!

TheOnly3aq, 2024-11-13:
For me, on my 2017 MBP running Kali Linux everything installed normally, yet touchbar was still not working. I ran the workaround by binding and unbinding the device, yet I seem to not have that device at all. I'm getting the following error:

```
1-3
tee: /sys/bus/usb/drivers/usb/unbind: No such device

```

Anyone got any ideas? I suspected it might be because I am not running dual boot, just the full 1TB for kali.