roadrunner2/macbook12-spi-driver #43: Problem on 13,3 (15") with Ubuntu 18.04.4

nndei, 2020-03-30:
Hello,

I am a very big noob who doesn't understand most of the guide.
However, I did the following steps but can't make this touchbar work.

1) checked my kernel with command `uname-a` and got `5.3.0-42-generic`
2) therefore jumped to load the two `spi_pxa2xx_platform` and `spi_pxa2xx_pci` with the command `modprobe`
3) `sudo -i` + password to become root
4) copied and pasted code: 
```
echo -e "\n# applespi\napplespi\nspi_pxa2xx_platform\nintel_lpss_pci" >> /etc/initramfs-tools/modules

apt install dkms
git clone https://github.com/roadrunner2/macbook12-spi-driver.git /usr/src/applespi-0.1
dkms install -m applespi -v 0.1
```
5) `sudo reboot`

Nothing has changed, how do I complete the steps and have a touchbar working? I am sorry, but I have not found any help on checking the initramfs/initrd for modules, also what if they're not there?

Best regards, Neri

roadrunner2, 2020-04-01:
Apologies for the slow reply. First of all, that change to `/etc/initramfs-tools/modules` isn't necessary, as those modules are for the keyboard/trackpad driver which appears to be already working fine (and besides, you just added a comment to that file, i.e. it's a no-op anyway). And similarly, the modprobe isn't necessary either.

Now, let's see why the touchbar isn't being activated. Here are some commands to run to gather some more info:
```sh
lsusb -t
sudo dkms status applespi
lsmod | grep apple
dmesg | grep apple
```
The first command should give output similar to this:
```
/:  Bus 06.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
/:  Bus 05.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 10000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/2p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/8p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/16p, 480M
    |__ Port 3: Dev 2, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 3: Dev 2, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 3: Dev 2, If 2, Class=Human Interface Device, Driver=usbhid, 480M
    |__ Port 3: Dev 2, If 3, Class=Human Interface Device, Driver=usbhid, 480M
```
The last four lines are the iBridge interfaces.

Next, the `dkms` output should show the `applespi` module as being installed for your kernel. If it isn't, try installing it again.

Then, the `lsmod` output should contain at least `apple_bridge` and `apple_ib_tb`. If not, try loading them explicitly (`sudo modprobe apple_ibridge`).

And lastly, the kernel messages should show some activity relating to the ibridge and touchbar drivers being activated.

nndei, 2020-04-01:
Hello, thank you for the reply.

I can't relate to what you wrote, so I think it's easier to copy past the outputs.

(1) The first command gives exactly your output, with just one more line after Bus 06, which is:
` |__ Port 2: Dev 2, If 0, Class=Vendor Specific Class, Driver=r8152, 5000M
`

(2) `dkms` output is:
```
Warning: I do not know how to handle applespi.
Usage: /usr/sbin/dkms [action] [options]
  [action]  = { add | remove | build | install | uninstall | match | autoinstall
               | mkdriverdisk | mktarball | ldtarball | mkrpm | mkkmp | mkdeb | mkbmdeb | status }
  [options] = [-m module] [-v module-version] [-k kernel-version] [-a arch]
              [-d distro] [-c dkms.conf-location] [-q] [--force] [--force-version-override] [--all]
              [--templatekernel=kernel] [--directive='cli-directive=cli-value']
              [--config=kernel-.config-location] [--archive=tarball-location]
              [--kernelsourcedir=source-location] [--no-prepare-kernel] [--no-initrd]
              [--binaries-only] [--source-only] [-r release (SuSE)] [--verbose]
              [--size] [--spec=specfile] [--media=floppy|iso|tar] [--legacy-postinst=0|1]
              [--no-depmod]
              [-j number]
Error! No action was specified.
```
(3) `lsmod` is:
```

applesmc               20480  0
input_polldev          20480  1 applesmc
apple_gmux             20480  0
apple_bl               20480  1 apple_gmux
applespi               40960  0
video                  49152  1 apple_gmux
```
(4) And finally `dmesg`:
```
[    0.283416] apple-properties: device path parse error -19 at 0x14:
[    0.283418] apple-properties: 00000000: a6 01 00 00 04 00 00 00 02 01 0c 00 d0 41 03 0a  .............A..
[    0.283419] apple-properties: 00000010: 00 00 00 00 01 01 06 00 00 02 7f ff 04 00 1e 00  ................
[    0.283420] apple-properties: 00000020: 00 00 73 00 61 00 76 00 65 00 64 00 2d 00 63 00  ..s.a.v.e.d.-.c.
[    0.283421] apple-properties: 00000030: 6f 00 6e 00 66 00 69 00 67 00 00 00 e2 00 00 00  o.n.f.i.g.......
[    0.283422] apple-properties: 00000040: 47 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  G...............
[    0.283423] apple-properties: 00000050: 00 00 03 00 06 00 1b 19 00 6c 05 00 00 00 00 00  .........l......
[    0.283423] apple-properties: 00000060: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    0.283424] apple-properties: 00000070: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    0.283425] apple-properties: 00000080: 00 00 00 00 00 00 00 01 01 01 01 00 06 10 30 a0  ..............0.
[    0.283426] apple-properties: 00000090: 02 03 40 00 00 00 00 00 00 00 00 02 c0 eb 9a 13  ..@.............
[    0.283427] apple-properties: 000000a0: 40 0b 08 07 50 00 00 00 34 00 00 00 08 00 20 00  @...P...4..... .
[    0.283427] apple-properties: 000000b0: 26 00 08 00 00 0f 60 09 78 00 00 00 00 00 00 00  &.....`.x.......
[    0.283428] apple-properties: 000000c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    0.283429] apple-properties: 000000d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    0.283430] apple-properties: 000000e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    0.283430] apple-properties: 000000f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    0.283431] apple-properties: 00000100: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    0.283432] apple-properties: 00000110: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 2c 00  ..............,.
[    0.283433] apple-properties: 00000120: 00 00 41 00 41 00 50 00 4c 00 2c 00 69 00 67 00  ..A.A.P.L.,.i.g.
[    0.283434] apple-properties: 00000130: 2d 00 70 00 6c 00 61 00 74 00 66 00 6f 00 72 00  -.p.l.a.t.f.o.r.
[    0.283434] apple-properties: 00000140: 6d 00 2d 00 69 00 64 00 00 00 08 00 00 00 06 00  m.-.i.d.........
[    0.283435] apple-properties: 00000150: 1b 19 20 00 00 00 41 00 41 00 50 00 4c 00 2c 00  .. ...A.A.P.L.,.
[    0.283436] apple-properties: 00000160: 47 00 66 00 78 00 59 00 54 00 69 00 6c 00 65 00  G.f.x.Y.T.i.l.e.
[    0.283437] apple-properties: 00000170: 00 00 08 00 00 00 01 00 00 00 24 00 00 00 67 00  ..........$...g.
[    0.283437] apple-properties: 00000180: 72 00 61 00 70 00 68 00 69 00 63 00 2d 00 6f 00  r.a.p.h.i.c.-.o.
[    0.283438] apple-properties: 00000190: 70 00 74 00 69 00 6f 00 6e 00 73 00 00 00 08 00  p.t.i.o.n.s.....
[    0.283439] apple-properties: 000001a0: 00 00 0c 00 00 00                                ......
[    1.296151] applespi spi-APP000D:00: Received corrupted packet (crc mismatch)
[    1.298458] applespi spi-APP000D:00: modeswitch done.
[    5.256424] apple_gmux: Found gmux version 4.0.29 [indexed]
[    5.545994] applesmc: key=858 fan=2 temp=46 index=45 acc=0 lux=0 kbd=0
[    5.546098] applesmc applesmc.768: hwmon_device_register() is deprecated. Please convert the driver to use hwmon_device_register_with_info().
```

nndei, 2020-04-01:
Hello, thank you for the reply.

Firstly, the `lsusb -t` command outputs your exact output, with an added line between the Bus 06 and Bus 05 lines:
`|__ Port 2: Dev 2, If 0, Class=Vendor Specific Class, Driver=r8152, 5000M`

`sudo dkms status applespi` gives no output.

`lsmod`:
`applesmc               20480  0
input_polldev          20480  1 applesmc
apple_gmux             20480  0
apple_bl               20480  1 apple_gmux
applespi               40960  0
video                  49152  1 apple_gmux
`
And lastly, `dmesg` outputs:

`[    0.283416] apple-properties: device path parse error -19 at 0x14:
[    0.283418] apple-properties: 00000000: a6 01 00 00 04 00 00 00 02 01 0c 00 d0 41 03 0a  .............A..
[    0.283419] apple-properties: 00000010: 00 00 00 00 01 01 06 00 00 02 7f ff 04 00 1e 00  ................
[    0.283420] apple-properties: 00000020: 00 00 73 00 61 00 76 00 65 00 64 00 2d 00 63 00  ..s.a.v.e.d.-.c.
[    0.283421] apple-properties: 00000030: 6f 00 6e 00 66 00 69 00 67 00 00 00 e2 00 00 00  o.n.f.i.g.......
[    0.283422] apple-properties: 00000040: 47 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  G...............
[    0.283423] apple-properties: 00000050: 00 00 03 00 06 00 1b 19 00 6c 05 00 00 00 00 00  .........l......
[    0.283423] apple-properties: 00000060: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    0.283424] apple-properties: 00000070: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    0.283425] apple-properties: 00000080: 00 00 00 00 00 00 00 01 01 01 01 00 06 10 30 a0  ..............0.
[    0.283426] apple-properties: 00000090: 02 03 40 00 00 00 00 00 00 00 00 02 c0 eb 9a 13  ..@.............
[    0.283427] apple-properties: 000000a0: 40 0b 08 07 50 00 00 00 34 00 00 00 08 00 20 00  @...P...4..... .
[    0.283427] apple-properties: 000000b0: 26 00 08 00 00 0f 60 09 78 00 00 00 00 00 00 00  &.....`.x.......
[    0.283428] apple-properties: 000000c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    0.283429] apple-properties: 000000d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    0.283430] apple-properties: 000000e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    0.283430] apple-properties: 000000f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    0.283431] apple-properties: 00000100: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[    0.283432] apple-properties: 00000110: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 2c 00  ..............,.
[    0.283433] apple-properties: 00000120: 00 00 41 00 41 00 50 00 4c 00 2c 00 69 00 67 00  ..A.A.P.L.,.i.g.
[    0.283434] apple-properties: 00000130: 2d 00 70 00 6c 00 61 00 74 00 66 00 6f 00 72 00  -.p.l.a.t.f.o.r.
[    0.283434] apple-properties: 00000140: 6d 00 2d 00 69 00 64 00 00 00 08 00 00 00 06 00  m.-.i.d.........
[    0.283435] apple-properties: 00000150: 1b 19 20 00 00 00 41 00 41 00 50 00 4c 00 2c 00  .. ...A.A.P.L.,.
[    0.283436] apple-properties: 00000160: 47 00 66 00 78 00 59 00 54 00 69 00 6c 00 65 00  G.f.x.Y.T.i.l.e.
[    0.283437] apple-properties: 00000170: 00 00 08 00 00 00 01 00 00 00 24 00 00 00 67 00  ..........$...g.
[    0.283437] apple-properties: 00000180: 72 00 61 00 70 00 68 00 69 00 63 00 2d 00 6f 00  r.a.p.h.i.c.-.o.
[    0.283438] apple-properties: 00000190: 70 00 74 00 69 00 6f 00 6e 00 73 00 00 00 08 00  p.t.i.o.n.s.....
[    0.283439] apple-properties: 000001a0: 00 00 0c 00 00 00                                ......
[    1.296151] applespi spi-APP000D:00: Received corrupted packet (crc mismatch)
[    1.298458] applespi spi-APP000D:00: modeswitch done.
[    5.256424] apple_gmux: Found gmux version 4.0.29 [indexed]
[    5.545994] applesmc: key=858 fan=2 temp=46 index=45 acc=0 lux=0 kbd=0
[    5.546098] applesmc applesmc.768: hwmon_device_register() is deprecated. Please convert the driver to use hwmon_device_register_with_info().
`

roadrunner2, 2020-04-01:
> Firstly, the `lsusb -t` command outputs your exact output, with an added line between the Bus 06 and Bus 05 lines:
> `|__ Port 2: Dev 2, If 0, Class=Vendor Specific Class, Driver=r8152, 5000M`

That's fine - this is a USB ethernet adapter you've apparently plugged in.

> `sudo dkms status applespi` gives no output.

Ah, this is our problem then - for some reason the dkms module did not get built and installed. Can you rerun `sudo dkms install -m applespi -v 0.1` and show its output?

nndei, 2020-04-01:
Man you're a wizard for that adapter thing!

Output:
`Error! Could not find module source directory.
Directory: /usr/src/applespi-0.1 does not exist.`

roadrunner2, 2020-04-01:
So it looks like you never ran the `git clone` command, or if you did that it failed. Please run that again (prefix it with `sudo` if you aren't already root), and then run the dkms install.

nndei, 2020-04-02:
Hello, thank you for the reply. 
I tried again installing the git, but then I still get the same Error as before. 

If I try to run the git again, I get 
`fatal: destination path 'macbook12-spi-driver' already exists and is not an empty directory.`

roadrunner2, 2020-04-02:
What is the exact command you're running? It looks like you're missing the last argument, causing it to clone into the `macbook12-spi-driver` directory instead of into `/usr/src/applespi-0.1`. To repeat, the full command should be:
```sh
sudo git clone https://github.com/roadrunner2/macbook12-spi-driver.git /usr/src/applespi-0.1
```

nndei, 2020-04-02:
All runs smoothly! Sorry for my inexperience. 

Thank you!

closed 2020-04-02 by nndei