roadrunner2/macbook12-spi-driver #45: Touchpad does not work for MacbookPro 14,3 with Ubuntu 20.04

jayachandranm, 2020-05-04:
I have tried Ubuntu 19.10 on the MacbookPro-14,3 model. Most of it worked out of the box, including the Touchpad and WiFi. Bluetooth and sound didn't work.

With Ubuntu 20.04 release, first I tried to upgrade, which failed with some errors. Then I tried reinstalling with 20.04. After installation, the Touchpad, Bluetooth, and sound didn't work. And the machine will freeze from time to time. The keyboard and WiFi worked without problems.

I checked the kernel drivers. 'applespi', 'spi_pxa2xx_platform', and 'intel_lpss_pci' were already loaded. With Ubuntu 19.10, with 'xinput --list', I could see the Touchpad listed, but for Ubuntu 20.04 the Touchpad was not listed.

Currently, I switched back to 19.10. Is there anything I can try further to get the machine working with Ubuntu 20.04? Or to provide any detailed logs?

roadrunner2, 2020-05-10:
Sorry for the slow reply. If the keyboard is working, then all the right drivers are loaded, so that's not the issue. Instead, I suspect there's a problem initializing the touchpad. First of all, were there any errors in the kernel log? (`dmesg | grep applespi`).

Second, I recently fixed another timing issue that came to my attention, and that fix is in the repo here but not yet upstreamed (need to do that soon). So can you try building a dkms module from the latest code here and see if that fixes the issue?

jayachandranm, 2020-05-11:
Thanks for replying. You are doing amazing work here to support Linux on Mac platform. I have an older MacBookPro that runs Ubuntu without any hiccups. But the new ones are real trouble. 

There were plenty of errors/warnings in kernel log. It was more like corrupted message for applespi. I can reinstall the machine with 20.04 and provide the exact messages. 

I will also try the latest code from the repository and update you on the status.

jayachandranm, 2020-05-12:
@roadrunner2 I installed Ubuntu 20.04 again on my Macbook Pro and installed the latest driver from the repository. And wow, my Touchpad now works again :) Thank you! 

Initially the Touchpad was not working. And the dmesg errors were,

> [    0.944697] applespi spi-APP000D:00: Received corrupted packet (crc mismatch)
> [   22.140886] applespi spi-APP000D:00: Received corrupted packet (invalid message length 8 - num-fingers 0, tp-len 48)
> ... and many repetitions of above message

After installing the new driver the dmesg showed,

> [    0.840347] applespi: loading out-of-tree module taints kernel.
> [    0.840378] applespi: module verification failed: signature and/or required key missing - tainting kernel
> [    1.004376] applespi spi-APP000D:00: modeswitch done.