roadrunner2/macbook12-spi-driver #46: mbp15 branch - Touch bar works doesn't visually change the fn-mode when FN key is pressed (since kernel 5.6.7)

mfvianna, 2020-05-13:
I'm using the mbp15 branch.
Until kernel 5.6.6, the module works well (except that sometimes it doesn't refresh the touchbar visually when the fn key is pressed or released - I opened another issue for that #44).

Since kernel 5.6.7 (tested 5.6.7, 5.6.8 and 5.6.11), the module compiles well, does enable the touch bar but it never visually changes the touch bar (although the meaning of the keys does change when the FN key is pressed or released)
In this case, from kernels >=5.6.7, an error message is generated in dmesg:

[  261.659012] apple-ib-touchbar 0003:05AC:8302.0007: tb: Failed to set touch bar mode to 2 (-110)

berrfred, 2020-05-25:
Same behaviour here on my mbp15 but it is still working properly with 5.6.7, then I updated to 5.6.13 and the described visual problem "appeared"... got it to work with 5.6.13 only once a couple of days ago and at the next boot the problem was there again.

```
apple-ib-touchbar` 0003:05AC:8302.0007: tb: Failed to set touch bar mode to 2 (-110)
```

mfvianna, 2020-05-30:
@roadrunner2 , I'd like to help debug and fix this issue, that affects this branches for all kernel versions since 5.6.7, but I don't know where to start

mfvianna, 2020-06-21:
I could not find the root cause so far, but could find a workaround.
Some discoveries:
- I've recompiled 5.6.6 and everything works, meaning the issue was not caused by difference on the compilation environment (gcc version, headers, etc.);
- After running 5.6.6 and - removing the module before booting - it does make it to work on 5.6.18 on the next boot;
- Removing the module on 5.6.18 before booting may or may not work on the next boot:  If this is in a working state and we remove the module before booting this will work on the next boot, if this is not in a working state, removing the module doesn't help the next boot;
- Booting MacOS doesn't seem to help or change the "persistent across boots" working/not working state
- Turnging off desn't seem to help or change the "persistent across boots" working/not working state

@roadrunner2 , could you take a look at this issue or shed some light?

@berrfred , I set up a workaround that consists in removing the apple_ib_tb module before shutdown.  For achieving that I created a simple systemd service that runs rmmod apple_ib_tb before shutdown/reboot.  But for this to start working you will have to boot at least once in 5.6.6 so that the "persistent across boots" gets activated.  It seems that not removing the module puts the touchbar in a "locked" state.  Booting 5.6.6 cleans it as long as you remove the module before rebooting.  MacOS seems to be able to recover from this locked state but also seems to restore it before rebooting, so it doesn't serve as a state cleaner

OBS: issue #44 seems to be unrelated and is not fixed with this workaround.

berrfred, 2020-06-21:
I could reproduce your workaround with 5.6.7 that has always been working for me:
- booted into 5.6.7
- sudo modprobe -r apple-ib-tb
- either boot or restart with 5.6.15 (the latest kernel I have installed so far) and the touchbar properly works
- as long as you keep unloading the driver before switching off, it keeps working at the next boot.

mfvianna, 2020-06-21:
Since the code is not documented and a have limited free time to investigate I could not yet find where the initialization and termination codes are or understand the functions of each variable.  Although I'm an experienced C developer, I'm not much familiarized with kernel code.
I guess, though, bases on these recent discoveries that there is some action in the termination code that could be added to the initialization code that would reset the touch bar and make it to work on all situations.

SentToDevNull, 2020-06-26:
Here's some additional info for this bug. When apple-ib-tb isn't unloaded before reboot, triggering the bug, all keys on the touchpad work EXCEPT the XF86XK_Audio* keys (which aren't detected properly; tested with `xev`). It's pretty interesting behavior, so I thought I'd mention it.