roadrunner2/macbook12-spi-driver #50: Touchbar not working anymore in 14,3 On kernel 5.7

marc-git, 2020-08-11:
On debian with kernel 5.7.0.2 this has stopped working for me. Can troubleshoot but not sure where to look...

divad42, 2020-08-11:
Can you try the steps in [this comment](https://github.com/roadrunner2/macbook12-spi-driver/issues/42#issuecomment-602160740)?  If it works, this is a duplicate of #42.

marc-git, 2020-08-12:
Hi, that didn't work. In case you might want to look at it, here is the output of:

`sudo grep . /sys/bus/usb/devices/usb1/1-3/*`
```
/sys/bus/usb/devices/usb1/1-3/authorized:1
/sys/bus/usb/devices/usb1/1-3/avoid_reset_quirk:0
/sys/bus/usb/devices/usb1/1-3/bcdDevice:0101
/sys/bus/usb/devices/usb1/1-3/bDeviceClass:ef
/sys/bus/usb/devices/usb1/1-3/bDeviceProtocol:01
/sys/bus/usb/devices/usb1/1-3/bDeviceSubClass:02
/sys/bus/usb/devices/usb1/1-3/bMaxPacketSize0:64
/sys/bus/usb/devices/usb1/1-3/bNumConfigurations:3
/sys/bus/usb/devices/usb1/1-3/busnum:1
Binary file /sys/bus/usb/devices/usb1/1-3/descriptors matches
/sys/bus/usb/devices/usb1/1-3/dev:189:1
/sys/bus/usb/devices/usb1/1-3/devnum:2
/sys/bus/usb/devices/usb1/1-3/devpath:3
/sys/bus/usb/devices/usb1/1-3/idProduct:8600
/sys/bus/usb/devices/usb1/1-3/idVendor:05ac
/sys/bus/usb/devices/usb1/1-3/ltm_capable:no
/sys/bus/usb/devices/usb1/1-3/manufacturer:Apple Inc.
/sys/bus/usb/devices/usb1/1-3/maxchild:0
/sys/bus/usb/devices/usb1/1-3/product:iBridge
/sys/bus/usb/devices/usb1/1-3/quirks:0x0
/sys/bus/usb/devices/usb1/1-3/removable:fixed
/sys/bus/usb/devices/usb1/1-3/rx_lanes:1
/sys/bus/usb/devices/usb1/1-3/speed:480
/sys/bus/usb/devices/usb1/1-3/tx_lanes:1
/sys/bus/usb/devices/usb1/1-3/uevent:MAJOR=189
/sys/bus/usb/devices/usb1/1-3/uevent:MINOR=1
/sys/bus/usb/devices/usb1/1-3/uevent:DEVNAME=bus/usb/001/002
/sys/bus/usb/devices/usb1/1-3/uevent:DEVTYPE=usb_device
/sys/bus/usb/devices/usb1/1-3/uevent:PRODUCT=5ac/8600/101
/sys/bus/usb/devices/usb1/1-3/uevent:TYPE=239/2/1
/sys/bus/usb/devices/usb1/1-3/uevent:BUSNUM=001
/sys/bus/usb/devices/usb1/1-3/uevent:DEVNUM=002
/sys/bus/usb/devices/usb1/1-3/urbnum:21
/sys/bus/usb/devices/usb1/1-3/version: 2.00
```

`sudo grep . /sys/bus/usb/devices/usb1/1-3/power/*`
```/sys/bus/usb/devices/usb1/1-3/power/active_duration:21176060
/sys/bus/usb/devices/usb1/1-3/power/async:enabled
/sys/bus/usb/devices/usb1/1-3/power/autosuspend:2
/sys/bus/usb/devices/usb1/1-3/power/autosuspend_delay_ms:2000
/sys/bus/usb/devices/usb1/1-3/power/connected_duration:21176060
/sys/bus/usb/devices/usb1/1-3/power/control:on
/sys/bus/usb/devices/usb1/1-3/power/level:on
/sys/bus/usb/devices/usb1/1-3/power/persist:1
/sys/bus/usb/devices/usb1/1-3/power/runtime_active_kids:0
/sys/bus/usb/devices/usb1/1-3/power/runtime_active_time:21175677
/sys/bus/usb/devices/usb1/1-3/power/runtime_enabled:forbidden
/sys/bus/usb/devices/usb1/1-3/power/runtime_status:active
/sys/bus/usb/devices/usb1/1-3/power/runtime_suspended_time:0
/sys/bus/usb/devices/usb1/1-3/power/runtime_usage:1
```

`sudo grep . /sys/bus/usb/devices/usb1/1-3/port/power/*`
```
/sys/bus/usb/devices/usb1/1-3/port/power/async:enabled
/sys/bus/usb/devices/usb1/1-3/port/power/control:auto
/sys/bus/usb/devices/usb1/1-3/port/power/runtime_active_kids:0
/sys/bus/usb/devices/usb1/1-3/port/power/runtime_active_time:21191563
/sys/bus/usb/devices/usb1/1-3/port/power/runtime_enabled:enabled
/sys/bus/usb/devices/usb1/1-3/port/power/runtime_status:active
/sys/bus/usb/devices/usb1/1-3/port/power/runtime_suspended_time:0
/sys/bus/usb/devices/usb1/1-3/port/power/runtime_usage:1```

there is no driver directory

treymerkley, 2020-08-30:
I think I'm experiencing the same issue on 5.8. When using the usual steps, I get:
```
$ ls /sys/bus/usb/drivers/usb
bind  uevent  unbind  usb1  usb2  usb3  usb4  usb5  usb6
$ echo '1-3' | sudo tee /sys/bus/usb/drivers/usb/unbind
1-3
tee: /sys/bus/usb/drivers/usb/unbind: No such device
```

wnuke, 2020-09-03:
I have the same issue with on mbp 14.3 on arch with kernel 5.8.5, the touchbar works normally until something happens, last time a full reinstall of arch fixed it, I have not pinpointed what change I make causes this, it usually happens after I have setup a window manager and installed spotify/other apps. I have tried the steps in #42 but I get the following:
```
% echo '1-3' | sudo tee /sys/bus/usb/drivers/usb/unbind
1-3
tee: /sys/bus/usb/drivers/usb/unbind: No such device
% echo '1-3' | sudo tee /sys/bus/usb/drivers/usb/bind
1-3
tee: /sys/bus/usb/drivers/usb/bind: No such device
```

marc-git, 2020-09-09:
ok so for me this has been fixed in debian `5.8.0-1-amd64` so closing this issue

closed 2020-09-09 by marc-git