roadrunner2/macbook12-spi-driver #54: Compile error in 5.9.0-1 too few arguments to function ‘iio_device_alloc’

marc-git, 2020-11-11:
I get the following error log. Is this a naming conflict? 

```/var/lib/dkms/applespi/0.1/build/apple-ib-als.c: In function ‘appleals_config_iio’:
/var/lib/dkms/applespi/0.1/build/apple-ib-als.c:463:29: warning: passing argument 1 of ‘iio_device_alloc’ makes pointer from integer without a cast [-Wint-conversion]
  463 |  iio_dev = iio_device_alloc(sizeof(als_dev));
      |                             ^~~~~~~~~~~~~~~
      |                             |
      |                             long unsigned int
In file included from /usr/src/linux-headers-5.9.0-1-common/include/linux/iio/buffer.h:10,
                 from /var/lib/dkms/applespi/0.1/build/apple-ib-als.c:32:
/usr/src/linux-headers-5.9.0-1-common/include/linux/iio/iio.h:684:49: note: expected ‘struct device *’ but argument is of type ‘long unsigned int’
  684 | struct iio_dev *iio_device_alloc(struct device *parent, int sizeof_priv);
      |                                  ~~~~~~~~~~~~~~~^~~~~~
/var/lib/dkms/applespi/0.1/build/apple-ib-als.c:463:12: error: too few arguments to function ‘iio_device_alloc’
  463 |  iio_dev = iio_device_alloc(sizeof(als_dev));
      |            ^~~~~~~~~~~~~~~~
In file included from /usr/src/linux-headers-5.9.0-1-common/include/linux/iio/buffer.h:10,
                 from /var/lib/dkms/applespi/0.1/build/apple-ib-als.c:32:
/usr/src/linux-headers-5.9.0-1-common/include/linux/iio/iio.h:684:17: note: declared here
  684 | struct iio_dev *iio_device_alloc(struct device *parent, int sizeof_priv);
      |                 ^~~~~~~~~~~~~~~~
make[3]: *** [/usr/src/linux-headers-5.9.0-1-common/scripts/Makefile.build:288: /var/lib/dkms/applespi/0.1/build/apple-ib-als.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [/usr/src/linux-headers-5.9.0-1-common/Makefile:1796: /var/lib/dkms/applespi/0.1/build] Error 2
make[1]: *** [/usr/src/linux-headers-5.9.0-1-common/Makefile:185: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.9.0-1-amd64'
make: *** [Makefile:16: all] Error 2

```

0neGuyDev, 2020-11-21:
A temporary fix is to comment the line out, that being line `463`  in `apple-ib-als.c` that'll allow you to compile it, not sure what side effects this could have for Mac's with an iBridge/T2/TouchBar, I haven't been able to test that.

This only really and intermediary until someone either PR's a fix or @roadrunner2 makes a fix.

marc-git, 2020-11-21:
> A temporary fix is to comment the line out, that being line `463` in `apple-ib-als.c` that'll allow you to compile it, not sure what side effects this could have for Mac's with an [...]

Did you test if it works on macs without the T2 ? I'm interesting in getting the touchbar on my MBP 14,3 working again. Tried the current pull request too but that didn't work (compiled though).

0neGuyDev, 2020-11-21:
Unfortunately no. 

I haven't been able to use the original fork nor roadrunner's they both get the same error messages (`dmesg | grep applespi`) and it doesn't seem to work even when reloading the module...

Can't for the life of me figure out why...

fannullone, 2021-01-02:
I got the touchbar back working: https://github.com/roadrunner2/macbook12-spi-driver/issues/42#issuecomment-753445371

PatrickVerner, 2021-01-04:
I forked it here with rado0x54's fix: https://github.com/PatrickVerner/macbook12-spi-driver

marc-git, 2021-01-04:
> I forked it here with rado0x54's fix: https://github.com/PatrickVerner/macbook12-spi-driver
cheers

marc-git, 2021-01-13:
Yes works great for me straight from your fork @PatrickVerner

closed 2021-01-13 by marc-git