roadrunner2/macbook12-spi-driver #57: 5.10 dkms install

Zizigola, 2021-04-24:
Hi first of all you are amazing thanks for fantastic work :)

does this support 5.10 kernel?I'm getting this error :

make -j8 KERNELRELEASE=5.10.32-1-lts...(bad exit status: 2)
ERROR (dkms apport): binary package for applespi: 0.1 not found
Error! Bad return status for module build on kernel: 5.10.32-1-lts (x86_64)
Consult /var/lib/dkms/applespi/0.1/build/make.log for more information.

AndreasAZiegler, 2021-04-25:
I have the same problem on Arch Linux.

sweah, 2021-07-14:
Hi, 
I have the same error message with Kernel 5.11 on Ubuntu 21.04:

make -j4 KERNELRELEASE=5.11.0-24-generic...(bad exit status: 2)
ERROR (dkms apport): binary package for applespi: 0.1 not found
Error! Bad return status for module build on kernel: 5.11.0-24-generic (x86_64)
Consult /var/lib/dkms/applespi/0.1/build/make.log for more information.

Is there a way to solve this problem?
Thanks xou in advance and thanks for all the fantastic work.

rob-hills, 2022-09-20:
I am getting a similar error message while following the instructions at https://nixaid.com/linux-on-macbookpro/

When I get to the point where I need to build and install applespi, I get the following error message:
```
make -j8 KERNELRELEASE=5.15.0-47-generic...(bad exit status: 2)
ERROR (dkms apport): binary package for applespi: 0.1 not found
Error! Bad return status for module build on kernel: 5.15.0-47-generic (x86_64)
```

I note that there is a pull request to fix this for kernels up to 5.18.  It would be fabulous if that could be merged into trunk. But it appears this project isn't being maintained?

Rade0nFighter, 2023-09-30:
Same

All good until the last line here:

```
echo -e "\n# applespi\napplespi\nspi_pxa2xx_platform\nintel_lpss_pci" >> /etc/initramfs-tools/modules

apt install dkms
git clone https://github.com/cb22/macbook12-spi-driver.git /usr/src/applespi-0.1
dkms install -m applespi -v 0.1
```

```
Building module:
Cleaning build area...
make -j8 KERNELRELEASE=6.2.0-33-generic...(bad exit status: 2)
ERROR (dkms apport): binary package for applespi: 0.1 not found
Error! Bad return status for module build on kernel: 6.2.0-33-generic (x86_64)
Consult /var/lib/dkms/applespi/0.1/build/make.log for more information.
```

Contents of make.log:

```
KMS make.log for applespi-0.1 for kernel 6.2.0-33-generic (x86_64)
<date>
make -C /lib/modules/6.2.0-33-generic/build M=/var/lib/dkms/applespi/0.1/build modules
make[1]: Entering directory '/usr/src/linux-headers-6.2.0-33-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~23.04) 12.3.0
  You are using:           gcc-12 (Ubuntu 12.3.0-1ubuntu1~23.04) 12.3.0
  CC [M]  /var/lib/dkms/applespi/0.1/build/applespi.o
  CC [M]  /var/lib/dkms/applespi/0.1/build/apple-ibridge.o
  CC [M]  /var/lib/dkms/applespi/0.1/build/apple-ib-tb.o
  CC [M]  /var/lib/dkms/applespi/0.1/build/apple-ib-als.o
/var/lib/dkms/applespi/0.1/build/apple-ib-tb.c: In function ‘appletb_fill_report_info’:
/var/lib/dkms/applespi/0.1/build/apple-ib-tb.c:949:9: warning: enumeration value ‘HID_REPORT_TYPES’ not handled in switch [-Wswitch]
  949 |         switch (field->report->type) {
      |         ^~~~~~
/var/lib/dkms/applespi/0.1/build/apple-ib-als.c: In function ‘appleals_config_iio’:
/var/lib/dkms/applespi/0.1/build/apple-ib-als.c:463:36: warning: passing argument 1 of ‘iio_device_alloc’ makes pointer from integer without a cast [-Wint-conversion]
  463 |         iio_dev = iio_device_alloc(sizeof(als_dev));
      |                                    ^~~~~~~~~~~~~~~
      |                                    |
      |                                    long unsigned int
In file included from ./include/linux/iio/buffer.h:10,
                 from /var/lib/dkms/applespi/0.1/build/apple-ib-als.c:32:
./include/linux/iio/iio.h:721:49: note: expected ‘struct device *’ but argument is of type ‘long unsigned int’
  721 | struct iio_dev *iio_device_alloc(struct device *parent, int sizeof_priv);
      |                                  ~~~~~~~~~~~~~~~^~~~~~
/var/lib/dkms/applespi/0.1/build/apple-ib-als.c:463:19: error: too few arguments to function ‘iio_device_alloc’
  463 |         iio_dev = iio_device_alloc(sizeof(als_dev));
      |                   ^~~~~~~~~~~~~~~~
./include/linux/iio/iio.h:721:17: note: declared here
  721 | struct iio_dev *iio_device_alloc(struct device *parent, int sizeof_priv);
      |                 ^~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/applespi/0.1/build/apple-ib-als.c:36:
/var/lib/dkms/applespi/0.1/build/apple-ib-als.c:485:72: error: ‘struct iio_dev’ has no member named ‘id’
  485 |         iio_trig = iio_trigger_alloc("%s-dev%d", iio_dev->name, iio_dev->id);
      |                                                                        ^~
./include/linux/iio/trigger.h:166:61: note: in definition of macro ‘iio_trigger_alloc’
  166 |         __iio_trigger_alloc((parent), THIS_MODULE, (fmt), ##__VA_ARGS__)
      |                                                             ^~~~~~~~~~~
./include/linux/iio/trigger.h:166:29: error: passing argument 1 of ‘__iio_trigger_alloc’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  166 |         __iio_trigger_alloc((parent), THIS_MODULE, (fmt), ##__VA_ARGS__)
      |                             ^~~~~~~~
      |                             |
      |                             char *
/var/lib/dkms/applespi/0.1/build/apple-ib-als.c:485:20: note: in expansion of macro ‘iio_trigger_alloc’
  485 |         iio_trig = iio_trigger_alloc("%s-dev%d", iio_dev->name, iio_dev->id);
[... 76 lines cut ...]
 1800 |         efivar_entry->var.VendorGuid = EFI_BL_LEVEL_GUID;
      |                     ^~
/var/lib/dkms/applespi/0.1/build/applespi.c:1803:15: error: implicit declaration of function ‘efivar_entry_get’; did you mean ‘efi_mokvar_entry_next’? [-Werror=implicit-function-declaration]
 1803 |         sts = efivar_entry_get(efivar_entry, NULL, &efi_data_len, &efi_data);
      |               ^~~~~~~~~~~~~~~~
      |               efi_mokvar_entry_next
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:260: /var/lib/dkms/applespi/0.1/build/apple-ibridge.o] Error 1
/var/lib/dkms/applespi/0.1/build/applespi.c: In function ‘applespi_save_bl_level’:
/var/lib/dkms/applespi/0.1/build/applespi.c:1830:15: error: implicit declaration of function ‘efivar_entry_set_safe’ [-Werror=implicit-function-declaration]
 1830 |         sts = efivar_entry_set_safe((efi_char16_t *)EFI_BL_LEVEL_NAME, efi_guid,
      |               ^~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/applespi/0.1/build/applespi.c: At top level:
/var/lib/dkms/applespi/0.1/build/applespi.c:2228:27: error: initialization of ‘void (*)(struct spi_device *)’ from incompatible pointer type ‘int (*)(struct spi_device *)’ [-Werror=incompatible-pointer-types]
 2228 |         .remove         = applespi_remove,
      |                           ^~~~~~~~~~~~~~~
/var/lib/dkms/applespi/0.1/build/applespi.c:2228:27: note: (near initialization for ‘applespi_driver.remove’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:260: /var/lib/dkms/applespi/0.1/build/applespi.o] Error 1
make[1]: *** [Makefile:2026: /var/lib/dkms/applespi/0.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.2.0-33-generic'
make: *** [Makefile:16: all] Error 2
```

Happy to help debug if I can but driver development is a bit over my head!