roadrunner2/macbook12-spi-driver #61: Complie Error

silentwitness9, 2021-12-31:
I have been trying to get this working on a 2019 MacBook Pro but get the following : 

└─# dkms install -m applespi -v 0.1                                                                                                                                              128 ⨯

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
make -j8 KERNELRELEASE=5.15.0-kali2-amd64...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.15.0-kali2-amd64 (x86_64)
Consult /var/lib/dkms/applespi/0.1/build/make.log for more information.

Make.log shows : 

DKMS make.log for applespi-0.1 for kernel 5.15.0-kali2-amd64 (x86_64)
Fri 31 Dec 10:21:05 GMT 2021
make -C /lib/modules/5.15.0-kali2-amd64/build M=/var/lib/dkms/applespi/0.1/build modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-kali2-amd64'
  CC [M]  /var/lib/dkms/applespi/0.1/build/applespi.o
/var/lib/dkms/applespi/0.1/build/applespi.c:62:10: fatal error: linux/input-polldev.h: No such file or directory
   62 | #include <linux/input-polldev.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [/usr/src/linux-headers-5.15.0-kali2-common/scripts/Makefile.build:282: /var/lib/dkms/applespi/0.1/build/applespi.o] Error 1
make[1]: *** [/usr/src/linux-headers-5.15.0-kali2-common/Makefile:1892: /var/lib/dkms/applespi/0.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-kali2-amd64'
make: *** [Makefile:11: all] Error 2

Any advice very much appreciated. Thanks

lvxingtu, 2022-10-19:
Having the same issue

lvxingtu, 2022-10-19:
Seems linux/input-polldev.h removed from linux newer kernel

Madmatt06, 2022-10-25:
I have the same issue on a 2018 MacBook Pro.