roadrunner2/macbook12-spi-driver #62: Support kernel 5.13.19

terentev, 2022-02-01:
trying to compile 5.13.19-28.31

`
DKMS make.log for applespi-0.1 for kernel 5.13.19-28.31custom (x86_64)
Tue 01 Feb 2022 08:16:38 AM MSK
make -C /lib/modules/5.13.19-28.31custom/build M=/var/lib/dkms/applespi/0.1/build modules
make[1]: Entering directory '/usr/src/linux-headers-5.13.19-28.31custom'
  CC [M]  /var/lib/dkms/applespi/0.1/build/applespi.o
  CC [M]  /var/lib/dkms/applespi/0.1/build/apple-ibridge.o
  CC [M]  /var/lib/dkms/applespi/0.1/build/apple-ib-tb.o
  CC [M]  /var/lib/dkms/applespi/0.1/build/apple-ib-als.o
/var/lib/dkms/applespi/0.1/build/apple-ib-als.c: In function ‘appleals_config_iio’:
/var/lib/dkms/applespi/0.1/build/apple-ib-als.c:491:31: error: passing argument 1 of ‘iio_trigger_alloc’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  491 |  iio_trig = iio_trigger_alloc("%s-dev%d", iio_dev->name, iio_dev->id);
      |                               ^~~~~~~~~~
      |                               |
      |                               char *
In file included from /var/lib/dkms/applespi/0.1/build/apple-ib-als.c:36:
./include/linux/iio/trigger.h:165:21: note: expected ‘struct device *’ but argument is of type ‘char *’
  165 | struct iio_trigger *iio_trigger_alloc(struct device *parent, const char *fmt, ...);
      |                     ^~~~~~~~~~~~~~~~~
/var/lib/dkms/applespi/0.1/build/applespi.c: In function ‘applespi_setup_read_txfrs’:
/var/lib/dkms/applespi/0.1/build/applespi.c:590:6: error: ‘struct spi_transfer’ has no member named ‘delay_usecs’
  590 |  dl_t->delay_usecs = applespi->spi_settings.spi_cs_delay;
      |      ^~
/var/lib/dkms/applespi/0.1/build/applespi.c: In function ‘applespi_setup_write_txfrs’:
/var/lib/dkms/applespi/0.1/build/applespi.c:619:6: error: ‘struct spi_transfer’ has no member named ‘delay_usecs’
  619 |  wt_t->delay_usecs = SPI_RW_CHG_DELAY_US;
      |      ^~
/var/lib/dkms/applespi/0.1/build/applespi.c:622:6: error: ‘struct spi_transfer’ has no member named ‘delay_usecs’
  622 |  dl_t->delay_usecs = applespi->spi_settings.spi_cs_delay;
      |      ^~
/var/lib/dkms/applespi/0.1/build/applespi.c:626:6: error: ‘struct spi_transfer’ has no member named ‘delay_usecs’
  626 |  wr_t->delay_usecs = SPI_RW_CHG_DELAY_US;
      |      ^~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/applespi/0.1/build/apple-ib-als.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/applespi/0.1/build/applespi.o] Error 1
make[1]: *** [Makefile:1879: /var/lib/dkms/applespi/0.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.13.19-28.31custom'
make: *** [Makefile:16: all] Error 2
`

CP12Z, 2022-07-10:
Was there any resolution on this? Having issues compiling on  kernel 5.18.0-1

make -C /lib/modules/5.18.0-1parrot1-amd64/build M=/var/lib/dkms/applespi/0.1/build modules
make[1]: Entering directory '/usr/src/linux-headers-5.18.0-1parrot1-amd64'
  CC [M]  /var/lib/dkms/applespi/0.1/build/applespi.o
  CC [M]  /var/lib/dkms/applespi/0.1/build/apple-ibridge.o
  CC [M]  /var/lib/dkms/applespi/0.1/build/apple-ib-tb.o
  CC [M]  /var/lib/dkms/applespi/0.1/build/apple-ib-als.o
/var/lib/dkms/applespi/0.1/build/apple-ib-als.c: In function ‘appleals_config_iio’:
/var/lib/dkms/applespi/0.1/build/apple-ib-als.c:463:29: warning: passing argument 1 of ‘iio_device_alloc’ makes pointer from i
nteger without a cast [-Wint-conversion]
  463 |  iio_dev = iio_device_alloc(sizeof(als_dev));
      |                             ^~~~~~~~~~~~~~~
      |                             |
      |                             long unsigned int
In file included from /usr/src/linux-headers-5.18.0-1parrot1-common/include/linux/iio/buffer.h:10,
                 from /var/lib/dkms/applespi/0.1/build/apple-ib-als.c:32:
/usr/src/linux-headers-5.18.0-1parrot1-common/include/linux/iio/iio.h:662:49: note: expected ‘struct device *’ but argument is
 of type ‘long unsigned int’
  662 | struct iio_dev *iio_device_alloc(struct device *parent, int sizeof_priv);
      |                                  ~~~~~~~~~~~~~~~^~~~~~
/var/lib/dkms/applespi/0.1/build/apple-ib-als.c:463:12: error: too few arguments to function ‘iio_device_alloc’
  463 |  iio_dev = iio_device_alloc(sizeof(als_dev));
      |            ^~~~~~~~~~~~~~~~
In file included from /usr/src/linux-headers-5.18.0-1parrot1-common/include/linux/iio/buffer.h:10,
                 from /var/lib/dkms/applespi/0.1/build/apple-ib-als.c:32:
/usr/src/linux-headers-5.18.0-1parrot1-common/include/linux/iio/iio.h:662:17: note: declared here
  662 | struct iio_dev *iio_device_alloc(struct device *parent, int sizeof_priv);
      |                 ^~~~~~~~~~~~~~~~
/var/lib/dkms/applespi/0.1/build/apple-ib-als.c:485:65: error: ‘struct iio_dev’ has no member named ‘id’
  485 |  iio_trig = iio_trigger_alloc("%s-dev%d", iio_dev->name, iio_dev->id);
      |                                                                 ^~
/var/lib/dkms/applespi/0.1/build/apple-ib-als.c:485:31: error: passing argument 1 of ‘iio_trigger_alloc’ from incompatible poi
nter type [-Werror=incompatible-pointer-types]
  485 |  iio_trig = iio_trigger_alloc("%s-dev%d", iio_dev->name, iio_dev->id);
      |                               ^~~~~~~~~~
      |                               |
      |                               char *
In file included from /var/lib/dkms/applespi/0.1/build/apple-ib-als.c:36:
/usr/src/linux-headers-5.18.0-1parrot1-common/include/linux/iio/trigger.h:167:54: note: expected ‘struct device *’ but argumen
t is of type ‘char *’
  167 | struct iio_trigger *iio_trigger_alloc(struct device *parent, const char *fmt, ...);
      |                                       ~~~~~~~~~~~~~~~^~~~~~
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-5.18.0-1parrot1-common/scripts/Makefile.build:293: /var/lib/dkms/applespi/0.1/build/apple
-ib-als.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/var/lib/dkms/applespi/0.1/build/applespi.c: In function ‘applespi_setup_read_txfrs’:
/var/lib/dkms/applespi/0.1/build/applespi.c:590:6: error: ‘struct spi_transfer’ has no member named ‘delay_usecs’
  590 |  dl_t->delay_usecs = applespi->spi_settings.spi_cs_delay;
      |      ^~
/var/lib/dkms/applespi/0.1/build/applespi.c: In function ‘applespi_setup_write_txfrs’:
/var/lib/dkms/applespi/0.1/build/applespi.c:619:6: error: ‘struct spi_transfer’ has no member named ‘delay_usecs’
  619 |  wt_t->delay_usecs = SPI_RW_CHG_DELAY_US;
      |      ^~
/var/lib/dkms/applespi/0.1/build/applespi.c:622:6: error: ‘struct spi_transfer’ has no member named ‘delay_usecs’
  622 |  dl_t->delay_usecs = applespi->spi_settings.spi_cs_delay;
      |      ^~
/var/lib/dkms/applespi/0.1/build/applespi.c:626:6: error: ‘struct spi_transfer’ has no member named ‘delay_usecs’
  626 |  wr_t->delay_usecs = SPI_RW_CHG_DELAY_US;
      |      ^~
/var/lib/dkms/applespi/0.1/build/applespi.c: At top level:
/var/lib/dkms/applespi/0.1/build/applespi.c:2228:13: error: initialization of ‘void (*)(struct spi_device *)’ from incompatibl
e pointer type ‘int (*)(struct spi_device *)’ [-Werror=incompatible-pointer-types]
 2228 |  .remove  = applespi_remove,
      |             ^~~~~~~~~~~~~~~
/var/lib/dkms/applespi/0.1/build/applespi.c:2228:13: note: (near initialization for ‘applespi_driver.remove’)
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-5.18.0-1parrot1-common/scripts/Makefile.build:293: /var/lib/dkms/applespi/0.1/build/apple
spi.o] Error 1
make[1]: *** [/usr/src/linux-headers-5.18.0-1parrot1-common/Makefile:1858: /var/lib/dkms/applespi/0.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.18.0-1parrot1-amd64'
make: *** [Makefile:16: all] Error 2

Perronef5, 2022-07-13:
Having the same issue on `5.13.0-30`

MaffooClock, 2022-12-08:
Same problem on 5.19.0-26

terentev, 2022-12-08:
`evg@evg:/usr/src/applespi-0.1$ diff applespi.c applespi.c.backup
590c590
< 	dl_t->delay.value = applespi->spi_settings.spi_cs_delay;
---
> 	dl_t->delay_usecs = applespi->spi_settings.spi_cs_delay;
619c619
< 	wt_t->delay.value = SPI_RW_CHG_DELAY_US;
---
> 	wt_t->delay_usecs = SPI_RW_CHG_DELAY_US;
622c622
< 	dl_t->delay.value = applespi->spi_settings.spi_cs_delay;
---
> 	dl_t->delay_usecs = applespi->spi_settings.spi_cs_delay;
626c626
< 	wr_t->delay.value = SPI_RW_CHG_DELAY_US;
---
> 	wr_t->delay_usecs = SPI_RW_CHG_DELAY_US;
`

terentev, 2022-12-08:
> Same problem on 5.19.0-26

diff applespi.c applespi.c.backup
590c590
< 	dl_t->delay.value = applespi->spi_settings.spi_cs_delay;
---
> 	dl_t->delay_usecs = applespi->spi_settings.spi_cs_delay;
619c619
< 	wt_t->delay.value = SPI_RW_CHG_DELAY_US;
---
> 	wt_t->delay_usecs = SPI_RW_CHG_DELAY_US;
622c622
< 	dl_t->delay.value = applespi->spi_settings.spi_cs_delay;
---
> 	dl_t->delay_usecs = applespi->spi_settings.spi_cs_delay;
626c626
< 	wr_t->delay.value = SPI_RW_CHG_DELAY_US;
---
> 	wr_t->delay_usecs = SPI_RW_CHG_DELAY_US;

terentev, 2022-12-08:
[image]

terentev, 2022-12-08:
[image]

terentev, 2022-12-08:
[image]

terentev, 2022-12-08:
Ubuntu 20.04 Macbook 12 touchpad is too sensitive
Create file /etc/libinput/local-overrides.quirks

[MacBook(Pro) SPI Touchpads]
MatchName=*Apple SPI Touchpad*
ModelAppleTouchpad=1
AttrKeyboardIntegration=internal
AttrTouchSizeRange=200:150
AttrPalmSizeThreshold=1200
[MacBook(Pro) SPI Keyboards]
MatchName=*Apple SPI Keyboard*
AttrKeyboardIntegration=internal