roadrunner2/macbook12-spi-driver #64: 39-usbmuxd.rules udev breaks touchbar on Ubuntu 21.10 and 22.04

davidjo, 2022-04-12:
I know this issue has been reported but googling did not list such issues so attempting to log for google.

rob-hills, 2022-10-12:
Have you seen the discussion and successful workarounds in #42 ?

davidjo, 2022-10-12:
No I hadnt - thanks for update  - maybe Ill try those fixups.
I resolved it by simply updating to the previous udev for usbmuxd which worked - the current fail is because of the addition of |5ac/8600/*  to ENV{PRODUCT} - previous ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*" , current (22.04) ENV{PRODUCT}=="5ac/12[9a][0-9a-f]/*|5ac/8600/*" - which Im guessing is the MacBook Pro 14,3 iBridge.
I dont get why this has been default enabled at all - it only seems to be needed to talk to iphones from MacBooks - given I dont have an iPhone for me its pointless.